openv/openv #453: Hallo

Wuckel01, 2019-07-21:


Wuckel01, 2019-07-21:
Hallo,
ich habe beim copelieren probleme. Habe IDE 1,6,9 mit den libraries async-mqtt-client-master,VitoWiFi-master,  und ESP8266 v.2.3.0 drauf und Node MCU 1.0 12E ausgewählt.
bekomme aber immer einen fehler.

In file included from C:\Program Files (x86)\arduino-1.6.9\libraries\async-mqtt-client-master\src/AsyncMqttClient.h:4:0,
                 from P:\Projekte\Vissmann Optoschnittstelle\VitoWiFi\VitoWiFi.ino:5:
C:\Program Files (x86)\arduino-1.6.9\libraries\async-mqtt-client-master\src/AsyncMqttClient.hpp:12:25: fatal error: ESPAsyncTCP.h: No such file or directory
 #include <ESPAsyncTCP.h>
                         ^
compilation terminated.
exit status 1
Fehler beim Kompilieren für das Board NodeMCU 1.0 (ESP-12E Module).

vieleicht habt ihr eine idee hab schone alle esp versionen durch immer das selbe.

gruss

Schnup89, 2020-01-15:
Hallo Wuckel01,

dir fehlt die ESPAsyncTCP Library Library, ich meine dass diese Lib beim async-mqtt-client dabei war.
Ansonsten einfach hier die ZIP-Datei herunterladen und unter Dokumente/Arduiono/Librarys einfügen, dann sollte der Fehler verschwinden.

https://github.com/me-no-dev/ESPAsyncTCP