openv/openv #455: Dokumentierte Adressen 209F - Vitotronic 200 - KO1B

michael581-crypto, 2019-08-12:
Hallo,
ich bin seid einigen Tagen dabei meine Ölheizung in mein Smarthome zu intigrieren.
Die Optolink Schnittstelle kann ich mit dem Raspberry Pi ansteuern. 
Leider finde ich keine Dokumentation Adressen oder fertige 
vcontrol.xml bzw. vito.xml Files die ich verwenden kann. 
Auch genaue Adressen zum ansprechen der Steuerung kann ich nicht wirklich finden.

Kannst mir da jemand helfen? oder hat jemand fertige Files ?

Bin über jede Hilfe dankbar.

Sonnengruesser, 2019-08-12:
Hier im Wiki:
https://github.com/openv/openv/wiki/Adressen
Oder meinst du was anderes?

Mario275, 2019-11-30:
Hallo, habe auch eine 209F Steuerung allerdings die KO2B. Hat jemand ausser der Adressenliste hier im Wiki eine mit mehr Datenpunkten. Bin noch blutiger Anfänger, werde sonst wohl oder übel durchprobieren was geht, habe aber etwas Angst das sich die Heizungssteuerung aufhängt.