openv/openv #456: Database of ready to use vito-xml files

martinschm, 2019-08-14:
Hi,
as far as I understood every device requires their specific configuration in the vito.xml. As I assume, many people will have the same devices, hence I am wondering if there is a sort of database with ready to use vito.xml files

Eg I am now looking to setup my Vitotronic 200 KW1 and there are probably people who have done so before.

Maybe it could be a separate subfolder here on Github of the project containing the files

MissMusic, 2020-02-01:
Nothing on this?

I have been searching like a mad man, surely this must exist somewhere? Or have the repository disappear with the old wiki? I can't imagine all this info is available here but no xml-files with basic commands added.

Another question, is there any sort of forum or community anywhere? Or is this project abandoned by everyone?

Sonnengruesser, 2020-02-01:
There is one [here](https://github.com/openv/openv/wiki/Vorschlag__Vito-Masterdateien), however it seems that there are so many differences depending on your specific installation, that making a "generic" file will end up with most of the addresses returning "0" or garbage values.
Also there were some legal concerns, so an existing database was discontinued (don't know it myself).

MissMusic, 2020-02-01:
Thanks, i found that but only a few addresses were compatible with my Vitodens 333-G. I used the data in the addresses part of the wiki to modify/create new commands in the linked XML. One datapoint i can't find is Hot Water target temp, all i can get is current HW temp.

Sonnengruesser, 2020-02-01:
what is your system ID (reply to 0x00F8)?

MissMusic, 2020-02-01:
If I'm correct this is the raw response in vcontrold:
<RECV: received 01 01 00 F8 01 20 21
20 -> V200WB2 ID=2053 Protokoll:GWG_VBEM
V200WB2 ID=2053 Protokoll:GWG_VBEM
This code in the vcontrold.xml seem to map it to the text string, I have no idea if it the string output is correct. 
`	<unit name='DeviceType'>
           <abbrev>DT</abbrev>
           <type>enum</type>
           <enum bytes='20 53 01 2B' text='V200WB2 ID=2053 Protokoll:GWG_VBEM'/>
	         <enum bytes='20 98' text='V200KW2 ID=2098 Protokoll:KW'/>
	         <enum bytes='20 C2' text='VDensHO1 ID=20C2 Protokoll:KW,P300'/>
		 <enum bytes='20 4D' text='V200-S ID=204D Protokoll:KW,P300'/>
           <enum text='UNKNOWN'/>
	</unit>`

Sonnengruesser, 2020-02-01:
I found these:
`GWG_Trinkwasser_Solltemperatur~0x005C
GWG_Trinkwasser_SolltemperaturBEM~0x0011
GWG_TrinkwassertemperaturSensor1~0x0042
GWG_TrinkwassertemperaturSensor2~0x0043
GWG_TrinkwassertemperaturZusatz~0x005D`

MissMusic, 2020-02-01:
Error on all except 0x0011 that returned 0.0000. :(

philverh, 2020-02-01:
Making a full database of addresses is next to impossible, as the same address is defined differently based on the device type (0x00F8). And even the answer to 0x00F8 is only a start. There are extension bytes in use also, to define subcategories of hardware/capabilities. 
Viessmann sells software (Vitosoft 300) to manage the different devices, but only to professionals. That SW must have the info required to get all the existing data from all the devices, but reverse engineering it to publish a database would not be legal I guess...

MissMusic, 2020-02-01:
I'm not expecting a full library, if there were a way to share device specific addesses for the most basic settings and logging i would be happy.

What i need:
get & set target room and DHW temperatures
get & set operating mode
get DHW current temp
get brine in/out temps
get heating circuit in/out temps

Basically no more advanced than whats available through the ViCare App. But i want to integrate to my home automation localy, and not "magically" using Viessmann cloud that just stops working without any way to troubleshoot.

Sonnengruesser, 2020-02-02:
What you can do: get Vitosoft
For the start, just look into ecnEventTypes.xml and ecnDataPointType.xml. First is a list of addresses with their functions and data structures. The latter is a list of heating devices, their ids and a list of the addresses they support.
I did just that and found 20 different versions of the ID=2053, which I did not compare in detail.
I also found that for my heating device there are additional addresses not listed in the xml files.
So right now it's a mess :(

MissMusic, 2020-02-02:
I don't think ID 2053 is correct for my system. That output string after sending 0x00F8 is fetched from the xml-file and not from the heat-pump.
[image]
If i understand this right, my system should be 0x0120 based on the raw response shown in vcontrold:
<RECV: received 01 01 00 F8 01 20 21

Sonnengruesser, 2020-02-02:
That's what I saw as well, but there is no 0120 in the List, so I thought maybe that's an Identification Extension...
Or my download of Vitosoft is too old and doesn't contain this system.

MissMusic, 2020-02-02:
Yea, i too realized that Vitosoft is probably too old. My heat pump model were released in the beginning of 2019. I did realize some of the codes looked familiar though, and I tried reading the same numbers as the display shows in the pumps menu and it works! 6000 get me the DHW Target temp. Next step is to change settings and so far no luck, this is whatvcontrold outputs:

[image]

Anyway, thanks a lot for your help!