openv/openv #457: Probleme beim Einrichten mit Raspberry

martinschm, 2019-08-14:
Hi,

in Schritt 4 der Anleitung (https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi) soll man die Schnittstellen prüfen. lsusb gibt bei mir das folgende aus
`Bus 001 Device 004: ID 0bda:8178 Realtek Semiconductor Corp. RTL8192CU 802.11n WLAN Adapter
Bus 001 Device 005: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 003: ID 0424:ec00 Standard Microsystems Corp. SMSC9512/9514 Fast Ethernet Adapter
Bus 001 Device 002: ID 0424:9512 Standard Microsystems Corp. SMC9512/9514 USB Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
`
Wenn ich dann den nächsten Befehl aus der Doku absetze
~/openv/vcontrold-code/vcontrold $ dmesg | grep "now attached"

Dann kommt bei mir:
-bash: /home/pi/openv/vcontrold-code/vcontrold: Ist ein Verzeichnis

Was ist da falsch in meinem Setup?

speters, 2019-08-14:
Der Befehl lautet:
```
dmesg | grep "now attached"
```

`dmesg` listet dabei Logmeldungen aus dem Kernel Ringbuffer, durch das Pipe-Symbol `|` erfolgt dann eine Weiterleitung an `grep`, welches dann nach dem Text `now attached` sucht.

martinschm, 2019-08-14:
Danke @speters . Da hab ich die Anleitung wohl nicht richtig gelesen oder sie ist so geschrieben das man das als Noob  falsch verstehen kann.

martinschm, 2019-08-14:
@speters Bei Schritt 6 steht man soll
sudo update-rc.d vcontrol start 99 2 3 4 5 . stop 99 0 1 6 . ausführen. Da kommt bei mir
update-rc.d: warning: start and stop actions are no longer supported; falling back to defaults

Sind die Befehle veraltet ?

bertmelis, 2019-08-15:
I assume you use the latest Raspbian? It will work. It is only a warning message.

You can easily test by rebooting and see if it auto started on boot.

martinschm, 2019-08-15:
I am using Buster.

If I try to connect, i get an error refused. So I assumed its not running.

I get that error, no matter if its a remote client whose IP is whitelisted
or locally via Telnet on port 3202 (if i remember) correctly

bertmelis, 2019-08-15:
run sudo systemctl status vcontrd