openv/openv #458: Steuerung / Auslesen Lüftungsanlage - Vitovent 300

SmightyCrown, 2019-08-15:
Hallo, 
ich habe die Wärmepumpe Vitocal 200-S (ID 204D). Diese kann ichbereits  per Viessmann Optolink Kabel und Raspi auslesen und steuern.  
Mittels Modbus-Kabel kann ich meine Viessmann Lüftungsanlage Vitovent 300-W mit der Vitotronic 200 Steuerung der Wärmepumpe verbinden und regeln.
Hat jemand schon Erfahrung, ob eine Steuerung der Lüftungsanlage mit openv möglich ist? Bisher habe ich noch nichts dazu gefunden.  Gibt es eine Möglichkeit die Befehle / Werte  die zur / von der  Lüftungsanlage gesendet werden,  auszulesen um sie ggf. in die vito.xml zu integrieren?

challo2018, 2023-04-26:
Der Beitrag ist sehr alt, aber @SmightyCrown : Konntest du eine Lösung finden?

EarlTommel, 2023-05-06:
@SmightyCrown  & @challo2018 
mir hat folgender Beitrag geholfen: https://github.com/openv/openv/issues/582
Über meine Anlagen ID konnte ich die richtige Adresse finden und es über ein Array steuern
Damit sind alle Betriebsmodi steuerbar.

bennatzen, 2023-09-12:
> @SmightyCrown & @challo2018 mir hat folgender Beitrag geholfen: #582 Über meine Anlagen ID konnte ich die richtige Adresse finden und es über ein Array steuern Damit sind alle Betriebsmodi steuerbar.

Hi. Kannst du vllt deine vito.xml dazu teilen?

EarlTommel, 2023-09-13:
ich bin auf einem ESP8266 unterwegs mit HomeAssistant und ESPHome.
D.h.  ich kann dir Auszüge aus meinem yaml code geben, falls dir das hilft.

bennatzen, 2023-09-14:
> ich bin auf einem ESP8266 unterwegs mit HomeAssistant und ESPHome. D.h. ich kann dir Auszüge aus meinem yaml code geben, falls dir das hilft.

Gerne. Kann ich mit arbeiten 😁

EarlTommel, 2023-09-15:
Hier der Auszug aus der ESPHome *yaml Datei - viel Erfolg.
Falls du eine Adresse findest zum Auslsesen des Status-Ferienprogram, würde ich mich über eine Rückmeldung freuen.

```
esphome:
  name: wp
  friendly_name: wp
  platform: ESP8266
  board: nodemcuv2

# Enable logging
logger:
  hardware_uart: UART1
  level: DEBUG

api:
  encryption:
    key: "xxxx"

ota:
  password: "yyyy"

wifi:
  networks:
   - ssid: !secret wifi_ssid
     password: !secret wifi_password
   - ssid: !secret wifi_ssid_third
     password: !secret wifi_password_third

  # Enable fallback hotspot (captive portal) in case wifi connection fails, without pwd, ssid = device_name
  ap: {}

captive_portal:

external_components:
  - source: github://pr#4453
    components: [ optolink ]

# Example configuration entry
optolink:
  protocol: P300
  device_info: Vitotronic 200
  #typ: 204C ausgelsen von 0x00f8 & 0x00f9
  #logger: enable

sensor:
  - platform: optolink
    name: WP Temp aussen
    address: 0x0101
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature
    accuracy_decimals: 1
  
  - platform: optolink
    name: WP WW Temp oben
    address: 0x010d
    bytes: 2
    div_ratio: 10
    unit_of_measurement: °C
    device_class: temperature
    accuracy_decimals: 1
  
  - platform: optolink
    name: HK Operation Mode
    address: 0xb020
    bytes: 1

  - platform: optolink
    name: Lueftung Operation Mode
    address: 0xb005
    bytes: 1
  
  - platform: optolink
    name: Operation Mode
    address: 0xb000
    bytes: 1
    
  - platform: optolink
    name: JAZ
    address: 0x1680
    bytes: 1
    div_ratio: 10
    accuracy_decimals: 1

  - platform: optolink
    name: JAZ Heizen
    address: 0x1681
    bytes: 1
    div_ratio: 10
    accuracy_decimals: 1

  - platform: optolink
    name: JAZ WW
    address: 0x1682
    bytes: 1
    div_ratio: 10   
    accuracy_decimals: 1

number:
  - platform: optolink
    name: WW Soll Temperatur
    unit_of_measurement: °C
    device_class: temperature
    address: 0x6000
    bytes: 2
    div_ratio: 10
    min_value: 40
    max_value: 60
    step: 1
    #mode: box
    icon: "mdi:home-thermometer"
  - platform: optolink
    name: WW Soll-2 Temperatur
    unit_of_measurement: °C
    device_class: temperature
    address: 0x600c
    bytes: 2
    div_ratio: 10
    min_value: 40
    max_value: 60
    step: 1
   # mode: box
    icon: "mdi:home-thermometer"

select:
  - platform: optolink
    name: HK Operation Mode
    address: 0xb020
    bytes: 1
    icon: "mdi:heating-coil"
    map:
      - "0 -> Normalbetrieb"
      - "1 -> manueller Betrieb"
      - "2 -> 1x Warmwasser"

  - platform: optolink
    name: Lueftung Operation Mode
    address: 0xb005
    bytes: 1
    icon: "mdi:air-filter"
    map:
      - "0 -> Abschaltbetrieb"
      - "1 -> Grundbetrieb"
      - "2 -> Lueftungsautomatik"
      - "66 -> Intensivbetrieb"
      - "130 -> Sparbetrieb"

  - platform: optolink
    name: Operation Mode
    address: 0xb000
    bytes: 1
    icon: "mdi:heat-pump-outline"
    map:
      - "0 -> Abschaltbetrieb"
      - "1 -> nur Warmwasser"
      - "2 -> Heizen u. WW"
  ```