openv/openv #459: Benoetige Hilfe bei der Konfiguration

maluramichael, 2019-09-22:
Ich werde aus der Dokumentation leider nicht ganz schlau und weiss nicht so recht wie die vito.xml und vcontold.xml konfigurieren muss.

Meine aktuelle Hardware:

- Vitoladens 300-C 19,3 kW 2-stufig
- Vitosol 200-FM 5x SV2F HU
- Eine Vitocell

Auf der Seite [Geräte](https://github.com/openv/openv/wiki/Ger%C3%A4te) finde ich den 300-C nicht.
Was am naechsten kommt waere glaub ich folgendes

Gerät | ID | Protokoll | Bemerkung
-- | -- | -- | --
VPlusHO1 | 20C8 | 300,KW | GWG2004: Öl-Brennwert-Wandgerät mit witterungsgeführter Regelung mit Schaltuhr für 1 Heizkreis mit Mischer und 1 Heizkreis ohne Mischer

Mich verunsichert nur etwas die Bemerkung "Öl-Brennwert-Wandgerät". Mein Geraet steht auf dem Boden. Gibt es zwischen dem Vitoladens 300 und dem 300-C große Unterschiede? Oder ist die Doku hier noch unvollstaendig? Kenne mich mit den Viessmann Bezeichnungen nicht wirklich aus :)

Muss ich um an die Kommandos zu kommen einfach nur in der [Liste hier](https://github.com/openv/openv/wiki/Adressen) nach VPlusHO1 suchen?

Vielen dank schonmal fuer jegliche art von Hilfe.

maluramichael, 2019-09-22:
Ok ich glaube solangsam blicke ich die Konfiguration. Hatte vergessen die Masterdaten zu nutzen. 

vcontrold und vclient koennen jetzt miteinander reden und ich bekomme auch Daten von der Anlage. 

[image]

closed 2019-09-22 by maluramichael

maluramichael, 2019-09-22:
[image]

Laeuft sehr gut. Vielen dank an alle. Wenn jemand Hilfe mit dem Vitoladens 300-C braucht einfach bescheid sagen.

Peter71131, 2022-08-19:
Ich könnte generell Hilfe gebrauchen. Allerdings weiß ich gar nicht wo ich anfangen soll.

Hast du evtl. nen guten Link für eine Schritt für Schritt Beschreibung für dumme?
Also sowas wie Installation von vcontrold auf einem Pi
richtiges Konfigurieren der vcontrold.xml und der vito.xml evtl. mit Beispielen
Was ist die richtige Hardware? Ich habe z.B. einen USB Optolink Adapter und würde den gerne mit dem PI nutzen.
Welches ist der richtige Adapter im IO Broker und wie verwendet man diesen.

maluramichael, 2022-08-19:
@Peter71131 die wichtigste Frage zuerst, welche Anlage hast du denn? Ich selbst nutze auch den original Optolink Adapter mit einem kleinen Raspberry Pi. Bei dem Pi ist es theoretisch egal welcher das ist. So viel Leistung braucht der garnicht.

Unter https://github.com/openv/vcontrold/releases sind fertig gebaute Pakete bereitgestellt. Hier kannst du dir zum Beispiel für deinen Pi das `[vcontrold_0.98.10-176_amd64.deb](https://github.com/openv/vcontrold/releases/download/v0.98.10/vcontrold_0.98.10-176_amd64.deb)` Paket laden und mit `wget https://github.com/openv/vcontrold/releases/download/v0.98.10/vcontrold_0.98.10-176_amd64.deb && sudo dpkg -i vcontrold_0.98.10-176_amd64.deb` installieren.

Wenn du ganz mutig bist kannst du natürlich auch die aller neuste Version selbst bauen. Dafür gäbe es hier die Anleitung https://github.com/openv/vcontrold/blob/master/doc/INSTALL.md

Bei meiner 300-C konnte ich unter https://github.com/openv/vcontrold/tree/master/xml die beiden Konfigurationsdateien im Verzeichnis `300` nutzen. Das kann bei dir aber auch anders aussehen.

IO Broker hab ich selbst noch nicht genutzt. Dazu kann ich leider nichts sagen.

Peter71131, 2022-08-19:
@maluramichael Danke für die schnelle Antwort.
Also ich glaube das Ding zu haben.
[image]
Sicher bin ich mir auch nicht. Ich habe ne Vitodens 200 mit nem Außenfühler, einem Heizkreis und ner Vitosol Solar...

Ich hatte damit angefangen https://openhabdoc.readthedocs.io/de/latest/Viessmann/
Und hab es dann auch anschließend mit https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi mit den gemanagten packages probiert.

Hab da evtl. schon zu viel Chaos verursacht. Soll/Kann ich erstmal alles deinstallieren?
Habs gerade nochmals mit den armhf paketen ausgeführt.

Peter71131, 2022-08-19:
Aus irgendeinem Grund bekomme ich die vito.xml nicht rein kopiert

maluramichael, 2022-08-20:
Die vito.xml und die vcontrold.xml müssen unter `/etc/vcontrold` liegen. Wird dir irgendeine Fehlermeldung angezeigt?

Peter71131, 2022-08-20:
Ich habs mit Filezilla Probiert. Aber kommt immer Übertragung konnte nicht gestartet werden.
Ich setzte jetzt das Ding mal neu auf und Probiere es mal nochmals Stück für Stück.

Peter71131, 2022-08-21:
@maluramichael 
so ich habe noch nicht alles bis auf letzte durchkonfiguriert.
Ich bin aktuell an dem Punkt, dass Vcontrold läuft und eine Verbindung aufbaut.

Versuche ich ein Kommando bekomme ich fogende Meldung:
Connection closed by foreign host.

fehlerhafte Konfiguration? Ist die DeviceID eigentlich wichtig oder nur das verwendete Protokoll?