openv/openv #461: Vitotronic 200 KW1

traori, 2019-10-14:
Hallo,
ich besitze eine Vitotronic 200 KW1. Als Kabel habe ich einen Optolink - RS232 - Adapter. Die Verbindung mit Viess-Data (Version 1) funktioniert und ich kann Daten mitloggen bzw grafisch anzeigen lassen. Es heißt ja, es kann jede Steuerung mit dem Protokoll KW300 ausgelesen bzw. gesteuert werden. Leider bekomme ich keine Verbindung mit Vies-sion zu meiner Steuerung hin. Ich habe auch bereits die Hilfedatei durchgelesen und komme da uch nicht richtig weiter. Daher die Frage - ist es überhaupt möglich mit Vies-sion meine Steuerung auszulesen/anzusteuern ?
Danke

GolterU, 2019-10-15:
Hallo,
an meiner Heizung ("Vitodens 300-W" WB3C, Code: CB) betreibe ich seit längerem die genannte Software. Das funktioniert auch ganz gut, allerdings mussten zu Beginn einige Dinge angepasst werden:

1. Je nachdem, wie der RS232-Adapter am Rechner angeschlossen ist, fehlt u.U. die positive Betriebsspannung. Das war bei mir dann der Fall, wenn ich den Optolink-Adapter über einen USB-Adapter (PC-USB ⇆ RS232-Optolink) angeschlossen hatte. Wenn dabei "RequestToSend" nicht verwendet wird, ist keine Ub+ vorhanden (vom Pegel DTR oder RTS erzeugt). 
Benutzt man zum Anschluss des Optolink-Adapters einen alten Rechner mit originärer RS232-Schnittstelle, oder speist die Betriebsspannung separat ein, klappt es.

2. Sind die Betriebsspannungen vorhanden, funktioniert das Datenlesen grundsätzlich. Es bricht aber mit Standarddateien (bei mir) mit Fehlermeldung "Parameter 'Erweiterung AM1' (7732) could not be fetched" ab. Das liegt daran, dass Viess-sion offenbar über Adressen stolpert, die von der Heizung nicht unterstützt werden.
Ich habe dann aus der Datei "Viess-ion_CB.xml" alle die Adressen gelöscht, die zu dem Fehler führten und weitere Adressen ergänzt. Damit klappt das meiste im Programm ganz gut.

3. Trotz der umfangreichen Dokumentation bleiben manche Fragen offen. Auch die Konfiguration der Heizung ist trotz der editierbaren PNG-Dateien nicht änderbar - oder habe nicht begriffen wie es geht. Der Programmierer hat offenbar nur einen Heizkreis mit Mischer vorgesehen. Obwohl die Quelladressen der angezeigten Parameter geändert werden können, verdecken u. U. deren Ausgaben Grafikdetails im Fenster "Steuerung", z. B. die Riesen-Zimmer-Solltemperatur einen an dieser Stelle gezeichneten Heizkörper.
So wie ich das sehe, wird wohl das Programm nicht mehr weiter entwickelt. Jedenfalls ist eine vor drei Jahren angekündigte Version nicht erschienen. Schade, hier hat sich Wolfgang Hofer wirklich viel Mühe gegeben.

traori, 2019-10-18:
Hallo GolterU,
danke für deine schnelle Rückmeldung. Was ich nun immer noch nicht weiß - funktioniert Vies-sion mit dem Protokoll KW1 ? Das was ich gelesen habe, ist das es das Protokoll 300 unterstützt und abwärtskompatibel zu KW ist. Da meine Steuerung ja nur das Protokoll KW unterstützt ist die Frage, funktioniert das?
Danke

GolterU, 2019-10-21:
Soweit ich das aus der Vies-sion Doku entnehmen kann, unterstützt das Programm tatsächlich nur das Protokoll 300. Die Abwärtskompatibilität bezieht sich wohl darauf, dass neuere Geräte, die das Protokoll 330 verstehen, auch über KW-Protokoll angesprochen werden können. Leider nicht umgekehrt, so dass Vies-sion bei Dir nicht in Frage kommt - wenn Deine Heizungssteuerung wirklich nur KW kann. 
Ich würde das ganz einfach mal ausprobieren. Solange man nicht schreibend auf die Anlage zugreift, sollte das auch kein Problem sein.
Gruß
Uwe