openv/openv #463: NodeMCU MQTT Vitocal 242-s

Bojkas1985, 2019-11-18:
Hi, I have such a problem.
Vitocal 242-s heat pump with Vitotronic 200 WO1C
NodeMCU V3
Construction https://github.com/openv/openv/wiki/Bauanleitung-ESP32-Adafruit-Feather-Huzzah32-and-Proto-Wing
MQTT Broker - https://www.cloudmqtt.com/
Addresses: https://github.com/openv/openv/wiki/Adressen-W%C3%A4rmepumpen-Vitotronic-WO1C
Sketch - https://github.com/Schnup89/OpenV_NodeMCU/blob/master/VitoWifi_NodeMCU.ino

Unfortunately, he doesn't read anything at all. I do not know exactly which side I should put IR and which phototransistor, I tried both options, but I assume IR to green light. When turned on in Broker it only prints VITOWIFI / $ status / online = 1.
What can I do? I'm doing something wrong? Missing something?

Thank you

Bojkas1985, 2019-11-19:
Ok, now works for me, only addresses doesnt works properly https://github.com/openv/openv/wiki/Adressen-W%C3%A4rmepumpen-Vitotronic-WO1C
I will try today these https://github.com/bertmelis/VitoWifi/files/1505914/vt200wo1c_en.pdf

bertmelis, 2019-11-19:
Glad to hear it works.

Addresses are always a bit hit and miss. You could write some sort of scanner but with all the different formats it will be a lot of work to manually decode.

Bojkas1985, 2019-11-19:
Hi, thank you. Do you think you could help me? I need to adjust the parameters so that I can read the data from my pump. Is it necessary to observe something, or can I change the variables to my liking?
For example, from:
DPTemp getTempA ("gettempa", "allgemein", 0x0800); // Aussentemperatur
getTempA.setCallback (tempCallbackHandler)

Change to:
DPTemp getTepMessage ("outdoor_heat", "allgemein", 0x0101); //Outside temperature
getTepGear.setCallback (tempCallbackHandler)

I assume the DPTemp and HEX parameters are important

How do I write? For example, change the room temperature?
Operation HC1 Heating circuit 1 2-Byte 10 30 0x2000

Thank you very much for your help

bertmelis, 2019-11-19:
The HEX is the address and `DPTemp` is the type (which means: 2 bytes, stored as int with factor 10). All the rest you can chose.

to be able to write, you have to mark it "writeable" first:
```C++
DPTemp getTempA("gettempa", "allgemein", 0x0800, true);  // true for writeable
```
Then, in your "set" function write something like this:
```C++
  float setValue = 21.5;  // or whatever value that fits
  DPValue value(setValue);
  VitoWiFi.writeDatapoint(getTempA, value);
  VitoWiFi.readDatapoint(getTempA);  // read back to check whether it worked
```

I was working on a more convenient version, but I moved to a place without Viessmann heating 😞 . I can still support a bit, but I can't test anymore.

Bojkas1985, 2019-11-19:
Okay, I guess I understand. Then I publish a "topic" in MQTT
VITOWIFI/setTempWWsoll and "payload" already directly value "40". It is so?

bertmelis, 2019-11-19:
That depends on how you parse the MQTT payload. It could be a string, binary format...

Bojkas1985, 2019-11-19:
Ok, i can read many valuse, but i cannot write temperature.
TeplotaTUVpozadovana.setCallback(tempCallbackHandler);  read sets value

DPTemp TeplotaTUVpozadovana("TeplotaTUVpozadovana", "wasser", 0x6000); read sets temp
DPTempS TeplotaTUVzapis("TeplotaTUVzapis", "wasser", 0x6000); write temp

**in void mqtt**
mqttClient.subscribe("VITOWIFI/TeplotaTUVzapis", 0);

**in void onMqttMessage:**
  if(strcmp(topic,"VITOWIFI/TeplotaTUVzapis") == 0) {
     uint8_t setTemp = atoi(payload);
     if(setTemp>=30 && setTemp<=70){
       DPValue value(setTemp);
       VitoWiFi.writeDatapoint(TeplotaTUVzapis, value);
       VitoWiFi.readDatapoint(TeplotaTUVpozadovana);
     }
  }

in void setup 
 TeplotaTUVzapis.setWriteable(true);

If i send to mqtt broker
Topic:
VITOWIFI/TeplotaTUVzapis
Payload 35

Got back from broker
http://fii.cz/kvkubsj

This is address from PDF
http://fii.cz/qmskagwt

bertmelis, 2019-11-19:
Define like this:
DPTempS TeplotaTUVzapis("TeplotaTUVzapis", "wasser", 0x6000, true); write temp

Bojkas1985, 2019-11-19:
Doesnt help. :-/ Same like before.

bertmelis, 2019-11-20:
Why do you create 2 datapoints with different types (1 byte length vs 2 bytes) on the same address? Just create 1 and do the reading and writing on this one.

It should be DPTempS by the way, as it is 2 bytes long. You cannot write a single byte to a 2 byte datapoint.

bertmelis, 2019-11-20:
So
```C++
DPTemp TeplotaTUV("TeplotaTUV", "wasser", 0x6000, true);  // read/write temp

// in void mqtt
mqttClient.subscribe("VITOWIFI/TeplotaTUV", 0);

// in void onMqttMessage
if(strcmp(topic,"VITOWIFI/TeplotaTUV") == 0) {
  float setTemp = static_cast<float>(atof(payload));
  if(setTemp>=30 && setTemp<=70) {
    DPValue value(setTemp);
    VitoWiFi.writeDatapoint(TeplotaTUV, value);
    VitoWiFi.readDatapoint(TeplotaTUV);
  }
}
```

You might want to add some `print()` statements to do debugging. Does the payload gets parsed as it should?

Bojkas1985, 2019-11-20:
Great, thank you, I'll try it today. Probably at the moment I don't know how to use tuning. I send it directly to the MQTT broker and either it displays the result or not. I'm not exactly a programmer, I try to put everything together and I learn. :-)

Bojkas1985, 2019-11-20:
Awesome! I can write both hot water for shower and heating!
Thank you so much!