openv/openv #466: Device ID für Viessmann Calotronic 200

Josef3000, 2019-12-20:
Hallo kann mir jemand sagen ob die DeviceID für meine Calotronic 200 Steuerung auch die 2098 ist ?
Sie soll ja eigentlich baukleich sein mit der Vitotronic 200 aber mit der aktuellen Konfiguration bekomme ich keine Werte zurück.
Es hat vielleicht einen anderen Grund aber ich möchte hier schon mal sicher gehen.

Ich versuche es mit einem Raspberry ZeroW. 
Der Adapter ist direkt an die GPIO Pins gelötet also nicht USB.

TingelRingel, 2024-11-18:
Hallo Josef,

Bist du zu einem Ergebnis gekommen? Ich überlege auch das Projekt anzugehen, für die Calotronic findet man allerdings wenig Infos.

Lg

Josef3000, 2024-11-18:
Hallo
die Schrittstelle hat von 2019 bis 2023 super funktioniert. Die Heizung habe ich inzwischen durch eine Pelletsheizung getauscht.
Die Daten kamen über die IR-Schnittstelle in den Rasberry ZeroW und von dort über MQTT an Openhub. Über einige Skripte auf dem Raspi und eine einfache Weboberfläche konnte ich sogar eine Temperatursteuerung und die Party-Funktion realisieren.

Gruß
Josef

TingelRingel, 2024-11-18:
Moin,
Ach das ist natürlich super zu hören. Dann werde ich das Projekt auf jeden Fall auch angehen.
Da das Ganze bei dir nicht mehr läuft weißt du die Device ID vermutlich auch nicht mehr oder?

Lg