openv/openv #470: V200KO1B Brennerstatus oder Modulation

SpatialJ, 2020-01-12:
Hallo,
ich habe in der vito.xml mein Vitotronic wie folgt eingestellt:
<device ID="209F" name="V200KO1B" protocol="KW2"/>

Die meisten Werte lassen sich auch abfragen, nur der Brennerstatus gibt immer "NOT OK" aus.
Kann mir jemand die richtige Commando-Definition für die Steuerung nennen?
Gibt es auch die Möglichkeit die aktuelle Modulation auszulesen?

Danke!

Juchar, 2024-07-01:
I would kindly push this, anyone got lucky to get the information whether the burner is on / off for the v200 ko1b or ko2b? I searched the hell out of Google, Forums, Vitosoft xmls, databases, adress tables and even KNX PDFs and no address seems to work (0842, 0843, 55D3, 55DE, 55D8, A305). I slowly start to get desperate :-) in the diagnosis menu of the control itself it shows fine. Thank you!

Juchar, 2024-09-30:
@SpatialJ konntest du die richtige Adresse jemals herausfinden? Vielen Dank schon mal!

SpatialJ, 2024-09-30:
@Juchar leider nein :-(