openv/openv #471: Vitocal 200S, external temperature senor

sobiso, 2020-01-17:
Hi all,

It's possible to control Vitocal200S with some external device(esp8266/esp32/smart home)? I know that we can do with vitotrol 300 but i want to do something similar. I already have temp sensor in my home and home assistant with Wago PLC.

Sonnengruesser, 2020-02-01:
I have the 222S and there it works. As long as there is a optical interface available it should work

sobiso, 2020-02-03:
@Sonnengruesser can you give me some more details.

Sonnengruesser, 2020-02-03:
I think so, have a look [here ](https://github.com/openv/openv/wiki/Bauanleitung-ESP8266)for the interface and [here ](https://github.com/bertmelis/VitoWifi)for the ESP library/software

sobiso, 2020-02-03:
Do you use this VitoWiFi as reference point (room temperature) with your 222S ?

Sonnengruesser, 2020-02-03:
No, not yet. Currently I just read status info, I don't change settings. That's on my todo list.
But communication works using the linked ESP library (modified a bit), changing values should work if you know the correct addresses.

dumpfheimer, 2020-07-28:
Hi!

Did you find out how to transmit the room temperature over optolink?