openv/openv #472: Heizungswerte mittels Skript (zB Python) auslesen

mich1986, 2020-01-18:
Hallo,

vielen Dank erstmal an alle Mitwirkenden dieses Projektes,
durch die zahlenreichen detailierten Beiträge ging das aufsetzen relativ geschmeidig ;-)

Zunächst mein Systemaufbau:
- Luftwärmepumpe Vitocal 300A mit Vitronic 200 (WO1B) Steuerung
- Optische Schnittstelle wird mittels Raspi über serielle Schnittstelle angesprochen
- Der vcontrol Daemon ist eingerichtet & läuft soweit (KW Protokoll)
- Mittels **vclient -h localhost:3002 sync getTempA getTempWW ...** kann entsprechend ausgelesen werden

Da ich den Raspi auch noch für andere Zwecke nutzen will, würde ich gerne mittels
(im Hintergrund laufenden) Python Skript die Werte zyklisch auslesen und in ein
Textfile (csv) ablegen.

Das Problem des vclient Aufrufs aus dem Skript sind zum einen die Broadcast Messages
**Message from syslogd@raspberrypi at Jan 18 10:43:45 ...
 vito[12679]: konfiguriere serielle Schnittstelle /dev/serial0
Broadcast message from systemd-journald@raspberrypi (Sat 2020-01-18 10:43:45 CET):
vito[12679]: konfiguriere serielle Schnittstelle /dev/serial0**
welche ich bis Dato nicht abstellen konnte, beim Aufruf des Skripts über
**sudo nohup python /home/pi/read_values.py > /dev/null 2>&1&** 
kommen trotzdem die Messages in den Vordergrund und blockieren dann die Konsole0;0

Zum anderen würde ich vclient gerne im Sinne von 
**Rückgabewerte = vclient(Übergabewerte)**
aufrufen und die Werte selbst in ein Textfile schreiben, aktuell habe das nur direkt über
vclient mit --csvfile & --ouput Parameter geschafft.

Alternativ dachte ich an eine Abfrage über Telnet. Hier habe ich aber die 
Schwierigkeit dass ich nur einen Befehl (zB getTempA) absetzen kann; dies 
liefert aber keinen gültigen Wert, weil vorher kein Sync gemacht wurde.

Zusammengefasst:
a) Kann ich die Broadcast Messages von syslogd/systemd-journald abstellen?
b) Kann vclient aus einem Skript als Funktion mit Rückgabewert aufgerufen werden?
c) Wie kann ich über das Telnet Interface (vcontrd>) mehrere Befehle am Stück ausführen?

Vielen Dank schon mal

squashball, 2020-02-16:
Hallo Michi,
sorry, dass ich mich hier dran hängen, ohne derzeit Antworten auf deine Fragen zu haben. Ich habe dieselbe Anlage und bin auf der Suche, nach einer Config. Habt ihr euch die alle in mühevollem Ausprobieren zusammengebaut? Oder wie geht man vor?
Danke, der Squashball
PS: Wie wäre es mit iobroker als "Datenspeicher"?

sisamiwe, 2020-02-17:
Hallo,
ich habe in einem Python Script bzw. smarthomeNG Plugin das Lesen und Schreiben umgesetzt.
Bei Interesse bitte PN.

mich1986, 2020-02-19:
@squashball 
Die vcontrol.xml habe ich im wesentlichen [hieraus](https://raw.githubusercontent.com/wiki/openv/openv/files/vcontrold.xml) (ohne P300 Anteil & mit angepasster serieller Schnittstelle)
Die vito.xml habe ich mir aus dieser [Tabelle](V200WO1,VBC700) (Spalte V200WO1,VBC700) zusammengestellt

squashball, 2020-02-19:
Danke, habe die Tabelle auch gefunden und bin ganz gut voran gekommen. Hast du etwas für die SOLL-Temp des Pufferspeichers? Da finde ich nix...

mich1986, 2020-02-19:
Leider nein, nur die Vorlauf Soll-Temp...mir fehlen auch noch ein paar Parameter die ich gerne hätte...

mich1986, 2020-02-19:
@sisamiwe PN via Email?

squashball, 2020-02-19:
Hm, ja mir auch. 
Die Betriebsart klappt bei mir nicht. Versucht mit 2004 (HK1) bzw. 3004 (HK2).
Ich verstehe auch nicht, wie ich aus den generellen Solltemp für WW die aktuelle Soll bekomme, die ja von den Zeiten abhängt.
Was fehlt dir noch?

mich1986, 2020-02-19:
WW Soll ist für mich eigentlich nicht so wichtig (hier habe ich über die Zeit einen konstanten Wert).
Was mir noch fehlt ist sowas wie Stromverbrauch bzw. aufgewandte Energie; hiermit will ich ermitteln, ob es effektiver ist mit Nachtstrom (aber tendenziell kälter Umgebungstemp) oder tagsüber das WW aufzuheizen...

squashball, 2020-02-20:
Ok, bei Energie/Stromwerten bin ich auch noch nicht bzw. habe nichts gefunden. Spannende Frage, die du da hast ;)