openv/openv #475: Optolink USB-DUO, Raspberry Pi 3 - Can't connect to Vitotronic 200 WO1C

ultrazauberer, 2020-02-17:
Hello there,

I'm a total beginner to vcontrold and come now to a point where I can't help myself anymore. I have a Raspberry Pi 3 Model B and soldered an USB-Duo Optolink adapter.

I followed the openv-instructions for building vcontrold on the pi: it worked very well and I installed successfully vcontrold.

For the configs I used these two XMLs (https://github.com/openv/openv/wiki/Vorschlag__Vito-Masterdateien) from Der_Vitti and modified the serial interface to my simlinked one:

`<?xml version="1.0"?>
<V-Control xmlns:vcontrol="http://www.openv.de/vcontrol">
   <unix>
      <config>
        <serial>
                <tty>/dev/vitoir0</tty>
        </serial>
        <net>
                <port>3002</port>
                <allow ip='127.0.0.1'/>
                <allow ip='192.168.1.0/24'/>
        </net>
        <logging>
                <file>/var/log/vcontrold.log</file>
                <syslog>n</syslog>
                <debug>y</debug>
        </logging>
        <device ID="204D"/>
     </config>
...`

Also my USB to serial adapter is recognized:
`pi@heizung:~ $ ls -l /dev/vitoir0
lrwxrwxrwx 1 root root 7 Feb 17 17:48 /dev/vitoir0 -> ttyUSB0`

I have vcontrold running as a service and I can connect via telnet to the CLI. But if I'm sending anything to my Vitotronic 200 WO1C I can't receive anything and it drops an error. Here is the last of my logfile:

`[398] Mon Feb 17 18:09:48 2020 : Client verbunden 127.0.0.1:36020 (FD:2)
[398] Mon Feb 17 18:09:48 2020 : Child Prozess mit pid:727 gestartet
[727] Mon Feb 17 18:09:50 2020 : Befehl: device
[727] Mon Feb 17 18:53:10 2020 : Befehl: device
[727] Mon Feb 17 18:53:12 2020 : Befehl: commands
[727] Mon Feb 17 18:53:33 2020 : Befehl: Außentemperatur
[727] Mon Feb 17 18:53:33 2020 : Process 727 tries to aquire lock
[727] Mon Feb 17 18:53:33 2020 : Process 727 got lock
[727] Mon Feb 17 18:53:33 2020 : >FRAMER: open device /dev/vitoir0 ProtocolID 41
[727] Mon Feb 17 18:53:33 2020 : konfiguriere serielle Schnittstelle /dev/vitoir0
[727] Mon Feb 17 18:53:34 2020 : >SEND: 04
[727] Mon Feb 17 18:53:35 2020 : <RECV: len=1 05 (1320.0 ms)
[727] Mon Feb 17 18:53:35 2020 : <RECV: received 05
[727] Mon Feb 17 18:53:35 2020 : >FRAMER: closed
[727] Mon Feb 17 18:53:35 2020 : >SEND: 16
[727] Mon Feb 17 18:53:35 2020 : >SEND: 00
[727] Mon Feb 17 18:53:35 2020 : >SEND: 00
[727] Mon Feb 17 18:53:37 2020 : <RECV: len=1 05 (1990.0 ms)
[727] Mon Feb 17 18:53:37 2020 : <RECV: received 05
[727] Mon Feb 17 18:53:37 2020 : >SEND: 16
[727] Mon Feb 17 18:53:37 2020 : >SEND: 00
[727] Mon Feb 17 18:53:37 2020 : >SEND: 00
[727] Mon Feb 17 18:53:39 2020 : <RECV: len=1 05 (1990.0 ms)
[727] Mon Feb 17 18:53:39 2020 : <RECV: received 05
[727] Mon Feb 17 18:53:39 2020 : >SEND: 16
[727] Mon Feb 17 18:53:39 2020 : >SEND: 00
[727] Mon Feb 17 18:53:39 2020 : >SEND: 00
[727] Mon Feb 17 18:53:41 2020 : <RECV: len=1 05 (1980.0 ms)
[727] Mon Feb 17 18:53:41 2020 : <RECV: received 05
[727] Mon Feb 17 18:53:41 2020 : >FRAMER: could not close (3 attempts)
[727] Mon Feb 17 18:53:41 2020 : Fehler beim oeffnen /dev/vitoir0
[727] Mon Feb 17 18:53:50 2020 : Befehl: quit
[727] Mon Feb 17 18:53:50 2020 : Process 727 released lock
[727] Mon Feb 17 18:53:50 2020 : Verbindung beendet (fd:2)
[727] Mon Feb 17 18:53:50 2020 : Child Prozess pid:727 beendet`

It seems it can receive some packets. But I'm unable to debug further and I have a feeling that my USB-Duo adapter is the issue. Maybe someone can point me into the right direction?

Is there an idiot proof adapter? I think I will directly use some GPIOs of the Pi for the serial connection but before I have to order some diods.

In the end I want to get readings from the heating system and store it in my InfluxDB.

Thanks a lot in advance!

philverh, 2020-02-18:
As you have not specified which device you are trying to communicate with, the following are guesses as to what may be the issue. I see a communication problem in the log extract you attached. 
You have used a set of .xml definitions which brings a set of questions. The commands supported by a controller are defined by the device type (and potentially some sub values). In addition, the device also supports one or more of the known Viessmann communication protocols. Make sure you have a correct base by checking the following:

- Do the .xml files match the device type you have ? To check what your device type is, you need to issue a read of address 0x00F8. Check the command name associated to that address in your xml file, and try issuing this command when you have vcontrold running. Does the answer match the device type supported in the .xml files? 
- Does your device support the protocol defined in the .xml files ? If the protocol defined in the .xml file is not supported by your device, it could will be a reason for the communication failure on your IR device. 

There used to be a small program (voIdent) to determine the type of device and the protocols it supported, but I believe it required an USB IR adapter, and only ran on Windows.

ultrazauberer, 2020-02-18:
Thank you for your help!

Like I wrote I have a Vitotronic 200 WO1C. It should have the ID 204D. And I used the XMLs for this device.

Now I used the XML from the build directory and tried to get the DeviceID:
`[397] Tue Feb 18 19:25:55 2020 : TCP socket 3002 geoeffnet
[397] Tue Feb 18 19:26:24 2020 : Client verbunden 127.0.0.1:42682 (FD:2)
[397] Tue Feb 18 19:26:24 2020 : Child Prozess mit pid:629 gestartet
[629] Tue Feb 18 19:26:44 2020 : Befehl: getDevType
[629] Tue Feb 18 19:26:44 2020 : Process 629 tries to aquire lock
[629] Tue Feb 18 19:26:44 2020 : Process 629 got lock
[629] Tue Feb 18 19:26:44 2020 : >FRAMER: open device /dev/vitoir0 ProtocolID 00
[629] Tue Feb 18 19:26:44 2020 : konfiguriere serielle Schnittstelle /dev/vitoir0
[629] Tue Feb 18 19:26:44 2020 : >SEND: 04
[629] Tue Feb 18 19:26:44 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:26:44 2020 : Warte auf 05
[629] Tue Feb 18 19:26:46 2020 : <RECV: 05 (1970.0 ms)
[629] Tue Feb 18 19:26:46 2020 : >SEND: 01
[629] Tue Feb 18 19:26:46 2020 : >SEND: C7
[629] Tue Feb 18 19:26:46 2020 : >SEND: F8
[629] Tue Feb 18 19:26:46 2020 : >SEND: 04
[629] Tue Feb 18 19:26:46 2020 : >SEND: 04
[629] Tue Feb 18 19:26:46 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:26:48 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:26:50 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:26:52 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:26:54 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:26:54 2020 : <RECV: received 05 05 05 05
[629] Tue Feb 18 19:26:54 2020 : Recv Timeout: 7960 ms  > 150 ms (Retry: 2)
[629] Tue Feb 18 19:26:54 2020 : >SEND: 04
[629] Tue Feb 18 19:26:54 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:26:54 2020 : Warte auf 05
[629] Tue Feb 18 19:26:56 2020 : <RECV: 05 (1990.0 ms)
[629] Tue Feb 18 19:26:56 2020 : >SEND: 01
[629] Tue Feb 18 19:26:56 2020 : >SEND: C7
[629] Tue Feb 18 19:26:56 2020 : >SEND: F8
[629] Tue Feb 18 19:26:56 2020 : >SEND: 04
[629] Tue Feb 18 19:26:56 2020 : >SEND: 04
[629] Tue Feb 18 19:26:56 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:26:58 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:00 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:02 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:04 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:04 2020 : <RECV: received 05 05 05 05
[629] Tue Feb 18 19:27:04 2020 : Recv Timeout: 7960 ms  > 150 ms (Retry: 1)
[629] Tue Feb 18 19:27:04 2020 : >SEND: 04
[629] Tue Feb 18 19:27:04 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:27:04 2020 : Warte auf 05
[629] Tue Feb 18 19:27:06 2020 : <RECV: 05 (1990.0 ms)
[629] Tue Feb 18 19:27:06 2020 : >SEND: 01
[629] Tue Feb 18 19:27:06 2020 : >SEND: C7
[629] Tue Feb 18 19:27:06 2020 : >SEND: F8
[629] Tue Feb 18 19:27:06 2020 : >SEND: 04
[629] Tue Feb 18 19:27:06 2020 : >SEND: 04
[629] Tue Feb 18 19:27:06 2020 : >FRAMER: no preset result
[629] Tue Feb 18 19:27:08 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:10 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:12 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:14 2020 : <RECV: len=1 05 (1990.0 ms)
[629] Tue Feb 18 19:27:14 2020 : <RECV: received 05 05 05 05
[629] Tue Feb 18 19:27:14 2020 : Recv Timeout: 7960 ms  > 150 ms (Retry: 0)
[629] Tue Feb 18 19:27:14 2020 : Recv Timeout, Abbruch
[629] Tue Feb 18 19:27:14 2020 : Fehler beim ausfuehren von getDevType
[629] Tue Feb 18 19:27:39 2020 : Befehl: quit
[629] Tue Feb 18 19:27:42 2020 : Befehl: quit
[629] Tue Feb 18 19:27:42 2020 : Process 629 released lock
[629] Tue Feb 18 19:27:42 2020 : Verbindung beendet (fd:2)
[629] Tue Feb 18 19:27:42 2020 : Child Prozess pid:629 beendet
pi@heizung:~ $`

Do you think my adapter is working and it's a configuration problem?

philverh, 2020-02-19:
It looks like the adapter is at least able to receive something from the heating controller (The "05" bytes in the log.) 
In the log I see you are trying to use GWG as communications protocol (The initial operation is a SYNC, sending 04 then waiting for 05, and the following send is prefixed with "01 C7"). Could you try changing the protocol to P300 in vito.xml. The following is an extract from my vito.xml
`<devices>
        <device ID="2046" name="Vitocal 300-G VBC700_BW_WW" protocol="P300"/>
    </devices>`
Of course the P300 protocol needs to be defined in the vcontrold.xml "protocols" section. 
And don't forget to reboot the Pi to make sure the new xml files are loaded.

This would tell us if your device is able to use the P300 protocol.

ultrazauberer, 2020-02-19:
Hi philverh,

thanks for your input. I googled and found the configs for the WO1C Vitotronic here:
https://www.schlocke.de/viessmann-heizungswerte-mit-openv-auslesen-und-mit-volkszaehler-darstellen/

I tried it with these configs but no luck. Here is my current header of the vito.xml:

> <vito>
>         <devices>
>                 <device ID="204D" name="V200-S" protocol="P300"/>
>         </devices>
>         <commands>
> 
> <command name="getDevType" protocmd="getaddr">
>          <addr>00F8</addr>
>          <len>2</len>
>          <unit>UN</unit>
>          <description>Information - Allgemein: Anlagentyp (204D)</description>
>      </command>
> [...]

The ID should be right for the WO1C. Also the protocol is P300. My start of the vcontrold.xml looks like this:

> <?xml version="1.0"?>
> <V-Control xmlns:vcontrol="http://www.openv.de/vcontrol">
>   <unix>
>      <config>
>         <serial>
>                 <tty>/dev/vitoir0</tty>
>         </serial>
>         <net>
>                 <port>3002</port>
>                 <allow ip='127.0.0.1'/>
>                 <allow ip='192.168.1.0/24'/>
>         </net>
>         <logging>
>                 <file>/var/log/vcontrold.log</file>
>                 <syslog>n</syslog>
>                 <debug>y</debug>
>         </logging>
>         <device ID="204D"/>
>      </config>
>   </unix>
>   <units>
> [...]
>   </units>
>   <protocols>                                                                                                                 <protocol name='P300'>
>           <pid>41</pid>
>           <macros>
>                 <macro name='GETADDR'>
>                     <command>SEND 00 01</command>
>                 </macro>
>           </macros>
>           <commands>
>                 <command name="getaddr">
>                      <send>GETADDR $addr $hexlen;RECV $len $unit</send>
>                 </command>
>                 <command name="setaddr">

Then I try telnet localhost 3002:

> pi@heizung:~ $ telnet localhost 3002
> Trying ::1...
> Trying 127.0.0.1...
> Connected to localhost.
> Escape character is '^]'.
> vctrld>getDevType
> ERR: >FRAMER: could not close (3 attempts)
> Fehler beim oeffnen /dev/vitoir0
> vctrld>

What's confusing me is the current log:

> [397] Wed Feb 19 17:58:08 2020 : TCP socket 3002 geoeffnet
> [397] Wed Feb 19 17:59:14 2020 : Client verbunden 127.0.0.1:57890 (FD:2)
> [397] Wed Feb 19 17:59:14 2020 : Child Prozess mit pid:651 gestartet
> [651] Wed Feb 19 17:59:19 2020 : Befehl: getDevType
> [651] Wed Feb 19 17:59:19 2020 : Process 651 tries to aquire lock
> [651] Wed Feb 19 17:59:19 2020 : Process 651 got lock
> [651] Wed Feb 19 17:59:19 2020 : >FRAMER: open device /dev/vitoir0 ProtocolID 41
> [651] Wed Feb 19 17:59:19 2020 : konfiguriere serielle Schnittstelle /dev/vitoir0
> [651] Wed Feb 19 17:59:19 2020 : >SEND: 04
> [651] Wed Feb 19 17:59:21 2020 : <RECV: len=1 05 (1540.0 ms)
> [651] Wed Feb 19 17:59:21 2020 : <RECV: received 05
> [651] Wed Feb 19 17:59:21 2020 : >FRAMER: closed
> [651] Wed Feb 19 17:59:21 2020 : >SEND: 16
> [651] Wed Feb 19 17:59:21 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:21 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:23 2020 : <RECV: len=1 05 (1980.0 ms)
> [651] Wed Feb 19 17:59:23 2020 : <RECV: received 05
> [651] Wed Feb 19 17:59:23 2020 : >SEND: 16
> [651] Wed Feb 19 17:59:23 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:23 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:25 2020 : <RECV: len=1 05 (1990.0 ms)
> [651] Wed Feb 19 17:59:25 2020 : <RECV: received 05
> [651] Wed Feb 19 17:59:25 2020 : >SEND: 16
> [651] Wed Feb 19 17:59:25 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:25 2020 : >SEND: 00
> [651] Wed Feb 19 17:59:27 2020 : <RECV: len=1 05 (2000.0 ms)
> [651] Wed Feb 19 17:59:27 2020 : <RECV: received 05
> [651] Wed Feb 19 17:59:27 2020 : >FRAMER: could not close (3 attempts)
> [651] Wed Feb 19 17:59:27 2020 : Fehler beim oeffnen /dev/vitoir0
> [651] Wed Feb 19 17:59:41 2020 : Befehl: quit
> [651] Wed Feb 19 17:59:41 2020 : Process 651 released lock
> [651] Wed Feb 19 17:59:41 2020 : Verbindung beendet (fd:2)
> [651] Wed Feb 19 17:59:41 2020 : Child Prozess pid:651 beendet
> pi@heizung:~ $

Why it can't open my serial connection? This confuses me:

> pi@heizung:~ $ ls -lah /dev/vitoir0
> lrwxrwxrwx 1 root root 7 Feb 19 17:58 /dev/vitoir0 -> ttyUSB0

I will try your suggestion with the KW2 protocol and use for this the standard XMLs from the build directory.

Thanks in advance!

ultrazauberer, 2020-02-19:
Here are my results with the generic vcontrold.xml and vito.xml. Modified to use my serial device and the KW2 protocol.

Looks more promising:
```
pi@heizung:~ $ telnet localhost 3002
Trying ::1...
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
vctrld>getTempA
ERR: Ergebnis falsch, Abbruch
Fehler beim ausfuehren von getTempA
vctrld>getDevType
UNKNOWN
vctrld>quit
good bye!
Connection closed by foreign host.
```

And the logfile:
```
[396] Wed Feb 19 18:18:21 2020 : TCP socket 3002 geoeffnet
[396] Wed Feb 19 18:19:55 2020 : Client verbunden 127.0.0.1:56862 (FD:2)
[396] Wed Feb 19 18:19:55 2020 : Child Prozess mit pid:632 gestartet
[632] Wed Feb 19 18:20:00 2020 : Befehl: getTempA
[632] Wed Feb 19 18:20:00 2020 : Process 632 tries to aquire lock
[632] Wed Feb 19 18:20:00 2020 : Process 632 got lock
[632] Wed Feb 19 18:20:00 2020 : >FRAMER: open device /dev/vitoir0 ProtocolID 00
[632] Wed Feb 19 18:20:00 2020 : konfiguriere serielle Schnittstelle /dev/vitoir0
[632] Wed Feb 19 18:20:00 2020 : >SEND: 04
[632] Wed Feb 19 18:20:00 2020 : >FRAMER: no preset result
[632] Wed Feb 19 18:20:00 2020 : Warte auf 05
[632] Wed Feb 19 18:20:01 2020 : <RECV: 05 (290.0 ms)
[632] Wed Feb 19 18:20:01 2020 : >SEND: 01
[632] Wed Feb 19 18:20:01 2020 : >SEND: F7
[632] Wed Feb 19 18:20:01 2020 : >SEND: 08
[632] Wed Feb 19 18:20:01 2020 : >SEND: 00
[632] Wed Feb 19 18:20:01 2020 : >SEND: 02
[632] Wed Feb 19 18:20:01 2020 : >FRAMER: no preset result
[632] Wed Feb 19 18:20:03 2020 : <RECV: len=1 05 (1990.0 ms)
[632] Wed Feb 19 18:20:05 2020 : <RECV: len=1 05 (1990.0 ms)
[632] Wed Feb 19 18:20:05 2020 : <RECV: received 05 05
[632] Wed Feb 19 18:20:05 2020 : Errstr matched, Ergebnis falsch (Retry:-1)
[632] Wed Feb 19 18:20:05 2020 : Ergebnis falsch, Abbruch
[632] Wed Feb 19 18:20:05 2020 : Fehler beim ausfuehren von getTempA
[632] Wed Feb 19 18:20:24 2020 : Befehl: getDevType
[632] Wed Feb 19 18:20:24 2020 : >SEND: 04
[632] Wed Feb 19 18:20:24 2020 : >FRAMER: no preset result
[632] Wed Feb 19 18:20:24 2020 : Warte auf 05
[632] Wed Feb 19 18:20:24 2020 : <RECV: 05 (540.0 ms)
[632] Wed Feb 19 18:20:24 2020 : >SEND: 01
[632] Wed Feb 19 18:20:24 2020 : >SEND: F7
[632] Wed Feb 19 18:20:24 2020 : >SEND: 00
[632] Wed Feb 19 18:20:24 2020 : >SEND: F8
[632] Wed Feb 19 18:20:24 2020 : >SEND: 02
[632] Wed Feb 19 18:20:24 2020 : >FRAMER: no preset result
[632] Wed Feb 19 18:20:26 2020 : <RECV: len=1 05 (1990.0 ms)
[632] Wed Feb 19 18:20:28 2020 : <RECV: len=1 05 (1990.0 ms)
[632] Wed Feb 19 18:20:28 2020 : <RECV: received 05 05
[632] Wed Feb 19 18:20:28 2020 : 05 05 -> UNKNOWN
[632] Wed Feb 19 18:20:28 2020 : UNKNOWN
[632] Wed Feb 19 18:20:48 2020 : Befehl: quit
[632] Wed Feb 19 18:20:49 2020 : Process 632 released lock
[632] Wed Feb 19 18:20:49 2020 : Verbindung beendet (fd:2)
[632] Wed Feb 19 18:20:49 2020 : Child Prozess pid:632 beendet
pi@heizung:~ $
```

I only receive 05 packets... I wonder if the sender is not working and the heating system always sends the 05 packets.

I have a Viessmann Vitocal 333G-NC BWT1.07 with a Vitotronic 200 (WO1C regarding to the service manual, I couldn't find that value in the menu of the heating system).

philverh, 2020-02-25:
Indeed, only receiving 05 could mean the sending part does not work. 05 is indeed sent periodically by the Viessmann controller, just to sync things. 
Double check the IR led orientation to make sure the polarity is right. 
When sending something, you should get back something more than "05". Even getting back an error  would help in determining if the sending part works.

ultrazauberer, 2020-03-02:
Hello again, I found the problem: The IR LED didn't work correctly. I changed it and now it works. Thanks for the help!

closed 2020-03-02 by ultrazauberer