openv/openv #476: Bauteilauswahl LED und Transistor

jnsw, 2020-02-26:
Ich find' es ziemlich schwierig die korrekte LED sowie Transistor zu finden. Reicht wirklich die Wellenlänge aus? Oder muss ich noch auf Anstiegs-, Abfallzeiten oder anderes ähnliches achten?

Schön wäre es, wenn man auflistet, welche Bauteile alle erprobenermaßen funktionieren. 

Man bekommt auf eBay leider die SFH487-2 nicht und SFH 309 FA wurde mir bereits falsch zugesendet und ist statt wie beschrieben schwarz oder dunkel bei mir ein durchsichtiger Transistor, ist das überhaupt möglich bei dem Bauteil?

Und das Wiki könnte man auch etwas besser strukturieren, da gefühlt jede Bauanleitung ihr eigenes Süppchen kocht und mal etwas erwähnt, oder mal auch nicht mal grundlegendes textuell erwähnt wie z.B. das die SFH487-2 andersrum gepolt ist als es bei LEDs üblich ist, oder sehe ich das falsch?

jnsw, 2020-02-26:
Nvm, habe das gefunden: https://github.com/openv/openv/issues/134

Sonnengruesser, 2020-02-26:
Ich hab die TSML1000/TEMT1000 verwendet und funktioniert auch. Die Datenraten sind sehr gering, also dürfte jede halbwegs vernünftige IR Diode funktionieren.
Die SFH309 gibts in transparent für sichtbares Licht und in schwarz für IR. Hast also vermutlich die falsche bekommen.

GolterU, 2020-02-26:
Vielleicht helfen Dir meine Aufzeichnungen zur Bauelementeauswahl. Der Text ist nicht 
 sonderlich strukturiert, enthält aber alle notwendigen Angaben: [BE-Auswahl-Hinweise.txt](https://github.com/openv/openv/files/4257816/BE-Auswahl-Hinweise.txt).
Wichtig ist meiner Erfahrung nach schon die Sende-/Empfangscharakteristik bezüglich der Wellenlänge. "Jede vernünftige IRED" wird nicht funktionieren. [Hier](https://github.com/openv/openv/wiki/Bauanleitung) zur Illustration ein paar diesbezügliche Grafiken.

Gruß
Golter

jnsw, 2020-02-27:
@Sonnengruesser danke, die vorgeschlagenen Bauteile arbeiten im 940/950nm-Bereich, scheinbar funktioniert das auch? Habe solche Bauteile zum Teil auch da, werde ich mal versuchen, danke!

@GolterU danke dir! Die Illustrationen hatte ich auch entdeckt, die sind echt hilfreich, danke dafür! Komisch, dass dein Vorredner es mit 60nm drüber dennoch klappt, falls dem wirklich so ist, oder fällt das noch in den "Rahmen". reichen 90% Ausschlag beim Empfänger?

Sonnengruesser, 2020-02-27:
Funktioniert einwandfrei. Ich habe mich an die "Bauanleitung USB" gehalten, da sind die Teile definiert.

Von der IR Schnittstelle für Smartmeter weiß ich, dass laut Standard zwischen 800 und 1000 nm definiert ist. Wenn die internen LEDs/Transistoren breitbanding sind, kann man extern beliebige schmalbandige Teile verwenden.

GolterU, 2020-02-28:
@jnsw Ich denke, allzu pingelig muss man nicht sein, was den Wellenlängen-Peak anbelangt. Vor allem bei der Empfangsstrecke (Heizung/Steuerung -> Optolink-Adapter) kann man ja gut mit dem Oszi messen, damit der Eingangspegel passt. Schwieriger ist schon die umgekehrte Richtung, wenn man nicht in der Heizungssteuerung selbst 'rumbasteln will. Ein entsprechender Test, bei welcher Amplitude der Empfänger korrekte Daten empfängt und 25% Aufschlag für eventuelle Schwankungen (bei möglichst guter Ausfilterung von Tageslicht) sollte funktionieren. So habe ich es jedenfalls gemacht. S. Stromlaufpläne hier [https://github.com/openv/openv/wiki/Bauanleitung](url) und [https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer](url).
Hilfreich ist auf jeden Fall bei Optolink-Nachbauten die Parameter der von Viessmann verwendeten Bauelemente zu kennen. IMHO verwenden sie als Sende-Element den Typ [SFH7222](http://www.icpdf.com/OSRAM_datasheet/SFH331_pdf_5575603/) (grüne LED plus IRED, 880 nm Peak) und als Empfänger [SFH331](http://www.icpdf.com/OSRAM_datasheet/SFH7222_pdf_5152957/) (rote LED und Fototransistor, 860 nm Peak). Ob sich Viessmann nach irgendwelchen Standards richtet, wage ich zu beweifeln - eher nach eigenem Gusto die BE auswählt.

Gruß
Golter

fitim7, 2020-03-19:
> Ich find' es ziemlich schwierig die korrekte LED sowie Transistor zu finden. Reicht wirklich die Wellenlänge aus? Oder muss ich noch auf Anstiegs-, Abfallzeiten oder anderes ähnliches achten?
>
> Schön [...]

Ich stehe aktuell vor dem selben Problem. Was hast du nun als Lösung bzw. alternative genutzt?