openv/openv #477: Read but no write to new 333-G Inverter.

MissMusic, 2020-02-28:
First of all, thanks a lot for vcontrol! Thanks to this project i now have logging to a database up and running.

My current setup is a RPi 3 connected to the pump using the officiall Viessmann optolink USB-interface stolen from the VitoConnect 100.

I've read about every topic of issues but have found none to get me any further then this. My new inverter pump model is too new, and not listed in the Vitosoft 300 software files, so i had to do a lot of trial and error with the XML-files to get anything working and mapping out the memory addresses of interest took some time.

As of now I'm running the P300 protocol and reading data works just fine. The next step is writing data, and here is where I'm stuck.

I've done a lot of trial and error, small changes in the xml-files, downloaded and tried other files posted here that seems to work for others, tried the older protocols, you name it, nothing works. From my understanding it should be quite straight forward doing a write, but i can't seem to get it.

Is there anyone that have a clue for what can be wrong, or have any tips on what to try next? I'm all out of ideas.

Here are my xml-files:

[vcontrold.xml](https://mega.nz/#!B4djEaLD!7_dZ1bqVwgPsSvVSAjsZnTC4cVJ78hMJjhZKSug3IPo)
[vito.xml](https://mega.nz/#!d0d3ESzb!9X6OnnFKmHSEfuLafzba2miRQ4pYH-9ZfQk-R3NYUpw)

And here are log outputs from vcontrold: 

This is the log from when i try to write:

[2680] Fri Feb 28 13:44:39 2020 : Client verbunden 192.168.1.101:49764 (FD:2)
[2680] Fri Feb 28 13:44:39 2020 : Child Prozess mit pid:11702 gestartet
[11702] Fri Feb 28 13:44:39 2020 : Befehl: setDHWTarget 55
[11702] Fri Feb 28 13:44:39 2020 : Process 11702 tries to aquire lock
[11702] Fri Feb 28 13:44:39 2020 : Process 11702 got lock
[11702] Fri Feb 28 13:44:39 2020 : >FRAMER: open device /dev/IR_interface_0 ProtocolID 41
[11702] Fri Feb 28 13:44:39 2020 : konfiguriere serielle Schnittstelle /dev/IR_interface_0
[11702] Fri Feb 28 13:44:39 2020 : >SEND: 04
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 05 (2010.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 05
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: closed
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 16
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 00
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 00
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 06 (10.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 06
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: opened
[11702] Fri Feb 28 13:44:41 2020 : Send Exp:V*10 [V=55.000000]
[11702] Fri Feb 28 13:44:41 2020 : Typ: short (Bytes: 26 02 )
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 41
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 07
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 00
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 02
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 60
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 00
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 01
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 26
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 02
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 92
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 06 (30.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 06
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: addr was still active FE06
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: Command send
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: no preset result
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 41 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 41
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 06 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 06
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 03 (10.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 02 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 60 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 00 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 01 (10.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 04 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: len=1 70 (0.0 ms)
[11702] Fri Feb 28 13:44:41 2020 : <RECV: received 03 02 60 00 01 04 70
[11702] Fri Feb 28 13:44:41 2020 : >FRAMER: ERROR address 6000 code 4
[11702] Fri Feb 28 13:44:41 2020 : Fehler recv, Abbruch
[11702] Fri Feb 28 13:44:41 2020 : Fehler beim ausfuehren von setDHWTarget 55
[11702] Fri Feb 28 13:44:41 2020 : Befehl: quit
[11702] Fri Feb 28 13:44:41 2020 : >SEND: 04
[11702] Fri Feb 28 13:44:43 2020 : <RECV: len=1 05 (2010.0 ms)
[11702] Fri Feb 28 13:44:43 2020 : <RECV: received 05
[11702] Fri Feb 28 13:44:43 2020 : >FRAMER: closed
[11702] Fri Feb 28 13:44:43 2020 : Process 11702 released lock
[11702] Fri Feb 28 13:44:43 2020 : Verbindung beendet (fd:2)
[11702] Fri Feb 28 13:44:43 2020 : Child Prozess pid:11702 beendet

And this is when i read:
[2680] Fri Feb 28 13:46:32 2020 : Client verbunden 192.168.1.101:37262 (FD:2)
[2680] Fri Feb 28 13:46:32 2020 : Child Prozess mit pid:11704 gestartet
[11704] Fri Feb 28 13:46:32 2020 : Befehl: DHWTarget
[11704] Fri Feb 28 13:46:32 2020 : Process 11704 tries to aquire lock
[11704] Fri Feb 28 13:46:32 2020 : Process 11704 got lock
[11704] Fri Feb 28 13:46:32 2020 : >FRAMER: open device /dev/IR_interface_0 ProtocolID 41
[11704] Fri Feb 28 13:46:32 2020 : konfiguriere serielle Schnittstelle /dev/IR_interface_0
[11704] Fri Feb 28 13:46:32 2020 : >SEND: 04
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 05 (620.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: received 05
[11704] Fri Feb 28 13:46:33 2020 : >FRAMER: closed
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 16
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 00
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 00
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 06 (10.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: received 06
[11704] Fri Feb 28 13:46:33 2020 : >FRAMER: opened
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 41
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 05
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 00
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 01
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 60
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 00
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 02
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 68
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 06 (30.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: received 06
[11704] Fri Feb 28 13:46:33 2020 : >FRAMER: addr was still active FE06
[11704] Fri Feb 28 13:46:33 2020 : >FRAMER: Command send
[11704] Fri Feb 28 13:46:33 2020 : >FRAMER: no preset result
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 41 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 07 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: received 41 07
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 01 (10.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 01 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 60 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 00 (10.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 02 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 F4 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 01 (0.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: len=1 60 (10.0 ms)
[11704] Fri Feb 28 13:46:33 2020 : <RECV: received 01 01 60 00 02 F4 01 60
[11704] Fri Feb 28 13:46:33 2020 : Typ: short (in float: 500.000000)
[11704] Fri Feb 28 13:46:33 2020 : (FLOAT) Exp:V/10 [B0:F4 B1:01 B2:00 B3:00 B4:00 B5:00 B6:00 B7:00 B8:00 B9:00 ]
[11704] Fri Feb 28 13:46:33 2020 : 50.000000 Degrees Celsius
[11704] Fri Feb 28 13:46:33 2020 : Befehl: quit
[11704] Fri Feb 28 13:46:33 2020 : >SEND: 04
[11704] Fri Feb 28 13:46:35 2020 : <RECV: len=1 05 (2010.0 ms)
[11704] Fri Feb 28 13:46:35 2020 : <RECV: received 05
[11704] Fri Feb 28 13:46:35 2020 : >FRAMER: closed
[11704] Fri Feb 28 13:46:35 2020 : Process 11704 released lock
[11704] Fri Feb 28 13:46:35 2020 : Verbindung beendet (fd:2)
[11704] Fri Feb 28 13:46:35 2020 : Child Prozess pid:11704 beendet

speters, 2020-02-28:
You seem to be using an outdated version of `vcontrold`.
Please use the latest one from https://github.com/openv/vcontrold and report back if problem persists. 

---

Du scheinst eine alte Version des `vcontrold` zu betreiben.
Bitte verwende eine aktuelle von https://github.com/openv/vcontrold und melde Dich, falls das Problem noch besteht.

philverh, 2020-02-28:
When you read the temp, your vito.xml states that you want to get a value from virt adddress 6000 with a length of 2, and a format of UT. When trying to set the temperature, you only use a length of 1...  for the same address and with the same format. Won't work.

MissMusic, 2020-02-28:
@speters I'll give that a try, thanks for pointing this out.

@philverh 
THANK YOU! I don't know how i missed that. Setting the DHW Target temp now works!

Any chance you can help me with the setOperatingMode command? I can't figure out what format i should use for input.

closed 2026-09-08 by MissMusic