openv/openv #478: 1* WW-Bereitung an einer Vitotronic 200, WO1B

squashball, 2020-02-29:
Hi zusammen,

leider scheinen alle bekannten Threads zu diesem Thema tot zu sein.
Hat es jemand geschafft, die einmalige Warmwasser-Bereitung zu aktivieren?

Danke, der Squashball

MrIcemanLE, 2020-03-09:
Hast du diesen Patch hier schon gefunden und probiert? Es scheint einen Bug bei der Umsetzung des P300 zu geben. Ich weiß nicht genau, ob der hier auf GitHub schon bekannt ist.

https://sourceforge.net/p/vcontrold/patches/3/

hmueller01, 2020-03-09:
> Hast du diesen Patch hier schon gefunden und probiert? Es scheint einen Bug bei der Umsetzung des P300 zu geben. Ich weiß nicht genau, ob der hier auf GitHub schon bekannt ist.
> 
> https://sourceforge.net/p/vcontrold/patches/3/

Hi, diese Patches sind alle in die aktuelle openv/vcontrold Version eingeflossen.

squashball, 2020-03-22:
Habe es doch hinbekommen. Geht über Adresse B020!
Danke euch.

m4ding, 2020-05-08:
Eine Frage: welche Device ID verwendest du?

squashball, 2020-05-09:
Ich verwende 2049, wobei das ja glaube ich egal ist, oder? Ich habe jedenfalls keine der vorhandenen Befehle benutzt, sondern mir die Sachen zusammen gesucht...