openv/openv #479: what is needed to make it work with 222-W ?

Zixim, 2020-03-02:
It looks as though my 222-W has no optolink port.
Can some one confirm this ?

If that is true, then what extra (Viessman) hardware would I need to get OpenV up & running ?

Sonnengruesser, 2020-03-20:
It should have. Try to slide open the covers next to the display. If you see the "V" with the status LEDs you are fine.

Zixim, 2020-03-20:
There are no sliding parts on the front panel, no LEDs, nothing except the touchscreen display.
This is the HMI i have on the device : https://www.youtube.com/watch?v=7EXRQGVT4WM

Zixim, 2020-10-27:
Has anyone installed Optolink on a touchscreen front panel ?

SebSeb11, 2024-02-09:
Just checked the manual. Looks like the touchscreen version does not have an optolink interface.