openv/openv #480: Vitotronic 200 WO1C - addresses and data types

ultrazauberer, 2020-03-03:
Hello great community,

with your help I managed to get my Raspberry Pi and the optolink USB Duo adapter working. Now I'm working on the XML files to get all the data I need from the heating system (Vitocal 333G-NC with Vitotronic 200 WO1C).

I used the config files provided here (https://www.schlocke.de/viessmann-heizungswerte-mit-openv-auslesen-und-mit-volkszaehler-darstellen/) as a base and started to modify it because some commands delivered strange results.
With these addresses (https://github.com/openv/openv/wiki/Adressen-W%C3%A4rmepumpen-Vitotronic-WO1C) I modified the JAZ commands like this:
```
     <command name="getJAZ" protocmd="getaddr">
         <addr>1680</addr>
         <len>1</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl (0..10)</description>
     </command>
     <command name="getJAZHeiz" protocmd="getaddr">
         <addr>1681</addr>
         <len>1</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl Heizen (0..10)</description>
     </command>
     <command name="getJAZWW" protocmd="getaddr">
         <addr>1682</addr>
         <len>1</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Jahresarbeitszahl WW (0..10)</description>
     </command>
```
Now I'm getting good values:
```
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
vctrld>getJAZ
48.000000
vctrld>getJAZHeiz
50.000000
vctrld>getJAZWW
41.000000
```
I just need to divide with 10 and the results are exactly the same like on the menu of the vitotronic. Perfect!

But now I'm trying to implement completly new commands: I want to get the consumption of the electrical energy (kWh) and the produced heat (kWh).
It seems the values are located at the addresses 1640 and 1660. I created these XML commands:
```
     <command name="getEnergieFaktor" protocmd="getaddr">
         <addr>163F</addr>
         <len>1</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Faktor Energiebilanz (1 = 0.1 kWh, 10 = 1 kWh, 100 = 10 kWh)</description>
     </command>
     <command name="getEnergieHeizwaerme" protocmd="getaddr">
         <addr>1640</addr>
         <len>4</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Heizwärme "Heizbetrieb", Verdichter 1</description>
     </command>
     <command name="getEnergieHeizelektro" protocmd="getaddr">
         <addr>1660</addr>
         <len>4</len>
         <unit>CO</unit>
         <description>Statistik - Energiebilanz: Elektroenergie "Heizbetrieb", Verdichter 1</description>
     </command>
```
I have no clue if I used the right datatype for the commands, but now I'm getting this in vcontrold:
```
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
vctrld>getEnergieFaktor
1.000000
vctrld>getEnergieHeizwaerme
158047.000000
vctrld>getEnergieHeizelektro
31281.000000
vctrld>
```
The first command gets the factor (1=0.1kWh, 10=1kWh, 100=10kWh). The last two commands should represent my needed data.
I looked into the menu of my Vitotronic and saw this for this week:
Energy consumption: 23.5 kWh - produced heat: 121.0 kWh

Now I know following:
158047.0 * 0.1 = 15804.7
15804.7 / 121.0 = 130.6174 (hmm, looks like the value from getEnergieHeizwaerme needs to divided by 128)

Now I'm wondering if my thoughts could be right. I'm guessing a lot (length of the data from the command and the data type).

Maybe someone can point me into the right direction?

MrIcemanLE, 2020-03-04:
I also trieb to figure out what adresses are used by the WO1C (DevType 204C) for the energy consumption and the COP.

I tried your adresses at my device and found no coincidences with the values showed in the display (see attached image)
[image]

ultrazauberer, 2020-03-04:
> I also tried to figure out what adresses are used by the WO1C (DevType 204C) for the energy consumption and the COP.
> 
> I tried your adresses at my device and found no coincidences with the values showed in the display (see attached image)
> [image]

Can you describe which addresses you read out? The third column of your table nearly matches with your values in the menu of the Vitotronic.

Also the ratio between the two values nearly matches the JAZ provided by the menu: 4.36 vs 4.42

What's strange: my values didn't change from yesterday to today.

MrIcemanLE, 2020-03-04:
I just copied your three commands from above. So the addresses are the same. First one is the thermal energy (0x1640) and the second one ist the electrical energy (0x1660).

My values didn't change either. I will monitor them and see if they change from Sunday to Monday or something else.

[image]

ultrazauberer, 2020-03-04:
I'll hope next week there are different values in the register. Than it means the register stores the values of the last week.
I can't imagine no one else tried before to get the power consumption of the heating system over the optolink interface.

What will be difficult is the decoding of the received values. Do you also have an energy factor of 1?

MrIcemanLE, 2020-03-04:
> I can't imagine no one else tried before to get the power consumption of the heating system over the optolink interface.

That are exactly my thoughts. I already asked in some issues/threads but with no luck.

If found that _0x1670_ delivers a value of 6070 ... but no idea what this could be.

In the table in your first link I found all the addresses from 0x1640 upwards. I already tried all of them but only the first one is showing a value.

The length of the data point in the table is 2 but the xml only returns a value if the <len> is set to 4. Does that mean that Viessmann has counted always two bytes and the vcontrol needs the true length of bytes?

> Do you also have an energy factor of 1?
yes

ultrazauberer, 2020-03-06:
When I was opening this topic I already included also the addresses of the boiling water (0x1670; 0x1671; 0x1660; 0x1661) and the 0x1670 and 0x1660 resulting in values where the ratio again matches my COP/JAZ for getting hot water.

I'm saving all the data into an influxdb (5 minutes periodes). So I hope I see next week some changes? Than the hard part begins. ;-)

MrIcemanLE, 2020-03-06:
So far "we" have 3 out of 4 values which are maybe connected to thermal an electrical energy:

0x1640
0x1660
0x1670

So we are missing a fourth one to get all the information we're interested in. Hopefully the values are changing on Sunday or Monday.

ultrazauberer, 2020-03-06:
I got this 13:10 today:

```
pi@heizung:~ $ vclient -h localhost:3002 -c getEnergieFaktor,getEnergieHeizwaerme,getEnergieHeizelektro,getEnergieHeizwa
ermeWW,getEnergieWWelektro
getEnergieFaktor:
1.000000
getEnergieHeizwaerme:
158047.000000
getEnergieHeizelektro:
31281.000000
getEnergieHeizwaermeWW:
27834.000000
getEnergieWWelektro:
6763.000000
pi@heizung:~ $
```

So all 4 values. Still waiting for a change and then I have to translate the values.

MrIcemanLE, 2020-03-06:
What address am i missing? I only got those three mentiioned above. Can you list your addresses or commands again?

ultrazauberer, 2020-03-06:
I attached my current XML files.
[WO1C.zip](https://github.com/openv/openv/files/4298554/WO1C.zip)

Sonnengruesser, 2020-03-06:
what happens with these?

Name | Adresse | #Bytes | Einheit | Umrechnung | Beschreibung
-- | -- | -- | -- | -- | --
WaermeEnergie_Heizen_12Monate | 0x1640 | 4 | kW | Div10 | WaermeEnergie_Heizen_12Monate (die   Wochenzähler werden Montags,0:00 Uhr zurückgesetzt, die Jahreszähler sind   rollierend)
ElektroEnergie_Heizen_12Monate   0x1650 | 0x1650 | 4 | kW | Div10 | ElektroEnergie_Heizen_12Monate (die   Wochenzähler werden Montags,0:00 Uhr zurückgesetzt, die Jahreszähler sind   rollierend)
WaermeEnergie_Warmwasser_12Monate   0x1660, 4 bytes, factor=0.1 | 0x1660 | 4 | kW | Div10 | WaermeEnergie_Warmwasser_12Monate (die   Wochenzähler werden Montags,0:00 Uhr zurückgesetzt, die Jahreszähler sind   rollierend)
ElektroEnergie_Warmwasser_12Monate | 0x1670 | 4 | kW | Div10 | ElektroEnergie_Warmwasser_12Monate   (die Wochenzähler werden Montags,0:00 Uhr zurückgesetzt, die Jahreszähler   sind rollierend)
ElektroEnergie_Warmwasser_1Woche | 0x16D0 | 2 | kW | Div10 | ElektroEnergie_Warmwasser_1Woche (die   Wochenzähler werden Montags 0:00 Uhr zurückgesetzt)
ElektroEnergie_Heizen_1Woche | 0x16D1 | 2 | kW | Div10 | ElektroEnergie_Heizen_1Woche (die   Wochenzähler werden Montags 0:00 Uhr zurückgesetzt)
WaermeEnergie_Warmwasser_1Woche | 0x16D2 | 2 | kW | Div10 | WaermeEnergie_Warmwasser_1Woche (die   Wochenzähler werden Montags 0:00 Uhr zurückgesetzt)
WaermeEnergie_Heizen_1Woche | 0x16D3 | 2 | kW | Div10 | WaermeEnergie_Heizen_1Woche (die   Wochenzähler werden Montags 0:00 Uhr zurückgesetzt)

ultrazauberer, 2020-03-06:
I implemented the values and got this:

```
pi@heizung:~ $ vclient -h localhost:3002 -c getEnergieEWocheWW,getEnergieEWocheHeizen,getEnergieWWocheWW,getEnergieWWocheHeizen
SRV ERR:
>FRAMER: Command send
>FRAMER: ERROR address 16D2 code 1
Fehler recv, Abbruch
Fehler beim ausfuehren von getEnergieWWocheWW
SRV ERR:  ERROR address 16D2 code 1
>FRAMER: Command send
>FRAMER: ERROR address 16D3 code 1
Fehler recv, Abbruch
Fehler beim ausfuehren von getEnergieWWocheHeizen
getEnergieEWocheWW:
0.000000
getEnergieEWocheHeizen:
0.000000
getEnergieWWocheWW:
server error
getEnergieWWocheHeizen:
server error
pi@heizung:~ $
```

MrIcemanLE, 2020-03-06:
The first four addresses in this table are the ones we are talking about. So if this is a 12 month value ist maybe changes every month and shows the summ of the last 12 months.

0x16D0 to 0x16D3 with the length of 2 only delivers 0.
With a length of 4 I get some data on 0x16D0 but those values are changing nearly on Everly request

MrIcemanLE, 2020-03-06:
What if the answer of the weekly data 0x16D? is much longer and contains all weeks at a certain byte position. In the control unit you can see all weekly data and this data must be stored some how and maybe requested. So for 52 weeks and 2 bytes per week the answer could be 104 bytes long. But I have no idea how to figure that out.

ultrazauberer, 2020-03-09:
Okay, the values are changing from sunday to monday. The bad thing: the counters for the heating itself decreases and the counter for the hot water increases.
```
	0x1650	0x1640	0x1660	0x1670
KW10	31281	158047	27834	6763
KW11	30781	156319	28091	6813
```
If I subtract the values (each week) and divide by 10 I get this:
`50.0, 172.8, 25.7, 5.0`

Up to now I have no clue how to get the right solution.

ultrazauberer, 2020-03-09:
Heureka... if it's a rolling sum than I have an idea:
```
KW10 ratio 5,052491928007417
KW11 ratio 5,078425002436568
```
If the value represents the last 12 months (52 weeks) I know why my values for the heating are decreasing: last year my source (brine system) for the heating system wasn't enough and I had really bad brine temperatures. So if the first week of last year doesn't count anymore to the sum, than my ratio (JAZ) increases. Like a FIFO buffer.

MrIcemanLE, 2020-03-09:
If your heating is already running for more than one year and if the values are showing the sum of the last 12 month (as suspected by @Sonnengruesser ) the values can go up and down depending on the week that ist now droped (one year ago) and the week that is added (last week). If this is true one could store these values at the 1st of january to have a yearly overview.

For a weekly documentation of the consumption we need to figure out how the 0x16D? addresses work.

MissMusic, 2020-03-12:
Awesome thread!

If you can figure out how to get weekly (or monthly) numbers that would be a great addition to sum up the total energy consumption for the house.

Wolf-Raider, 2020-08-28:
looks like the WO1C heatpump does it different. I found out, by sniffing the display (its I2C kommunikation simular to the 300 Protokoll) that the heatpump has function call IDs. (0x07 in 300 Protokoll) 
Wenn electric data is shown, there is an ID 0xB800 witch contains data for each day.
https://github.com/openv/openv/wiki/FunctionCalls-WO1C

MissMusic, 2020-08-28:
Wolf-Raider
So exiting! Please keep us posted if you manage to figure out how to extrapolate what data is what, and preferably what is needed in the xml-file to get at it using openv.

MrIcemanLE, 2020-10-05:
@Wolf-Raider I think I've got a little bit further in terms of interpretation of the Answer. Here are the responses of my WO1C when I call the 0xB800 using 0x02 and 0x0X for the weekday:

```
02   00   14   29   62 00 11 00 1C 00 0A 00 00 00 00 00
02   01   14   28   F7 00 2F 00 D4 00 35 00 00 00 00 00
02   02   14   28   5D 01 41 00 3A 01 4F 00 00 00 00 00
02   03   14   28   CB 01 55 00 3A 01 4F 00 00 00 00 00
02   04   14   28   F3 01 5C 00 A5 01 6B 00 00 00 00 00
02   05   14   28   08 02 60 00 04 02 82 00 00 00 00 00
02   06   14   28   60 02 70 00 D1 02 BB 00 00 00 00 00
```
As you suspected the first two bytes are from the request
3. byte: `0x14 = 20` is (maybe) the year,
4. byte: `0x28 = 40` and `0x28 = 41` is the number of the week in the year
5.-12. byte: ... the data ... ;-)
Maybe the 6th, 8th, 10th and 12th byte are scaling factor for the odd bytes:
`0x00 = 10^0`
`0x01 = 10^1`
`0x02 = 10^2` 
I'll do some math to figure this out. The Problem is, that it is nearly impossible to compare the results to the display, which only shows weekly data.

MrIcemanLE, 2020-10-05:
Sorry for the double post. I tried to use `0x00` and `0x01` as the first argument at the function call. The response at 0x01 says that the length is `0x0A = 10`. The number of the week and the year seems to be correct. But the data ... no idea so far.

Below each table I tried to get some values. The first col is the decimal value of the col above concatenated with the right one. The second col is the decimal value of only one col. Can't find any sense in this data

[image]

lanalnuel, 2020-10-06:
This with the id 0x02 looks very well if you swap the bytes. So i think also vitocal sums the values to the end of the week.

Look at the last line:
I think you have had (for the last week): 0x0260 thermical energy got from 0x0070 electrical energy and 0x021d thermical enegy for hot water and used there 0x00bb electrical energy. Could this be possible? (May the factor could be 10^-1?)

lanalnuel, 2020-10-06:
I also know, that there is a hidden menue in the display of the vitocal wo1c, where these values are shown.
Follow the discussion in this thread https://www.viessmann-community.com/t5/Experten-fragen/Energiebilanz-wird-bei-neuer-WO1C-SW-nicht-mehr-richtig/qaq-p/51252/comment-id/44897#M44897
Its german, but there has someone written what to do to unlock this. Including acces to expert level!

MissMusic, 2020-10-06:
Why the secrecy?
For level 1 coding, press Menu + OK and hold for a few seconds.
To get to level 2 do as above for level 1 first, from there press ? + Back and hold for a few seconds.  (it's possible it's OK + Back i can't remember) Now it should ask for a password or code, it's: 3442541331

@lanalnuel Do you happen to know where in the menu to find these energy readings?

lanalnuel, 2020-10-06:
@MissMusic its in service -enter level 1- than go to diagnostic, than there should be energybilance(?).

If not, you have to check this:
"Para 8812 = Wert +1
oder Para 8817 auf 1023

Dann noch 8818 auf 31"
=>
Id 8812 actual value +1
Id 8817 and 8818 see value above.
These ida can be found in service level 2.

MrIcemanLE, 2020-10-07:
> This with the id 0x02 looks very well if you swap the bytes. So i think also vitocal sums the values to the end of the week.
>
> Look at the last line:
> I think you have had (for the last week): 0x0260 [...]

Yeahh!! Finally we found the weekly data. The values are the same as in the control unit.

Now I have to figure out how I can return the converted values using the xml-Files. So far I wrote my own macro in the control.xml and used a command in the Vito.xml to get the data of each day of the week

[image]

vito.xml
```
    <command name="getEnergy" protocmd="setfunc">
          <addr>B800</addr>
          <len>16</len>
          <unit>PL</unit>
          <description>FunctionCall</description>
    </command>
```

vcontrol.xml:
```
    <unit name='Plain'>
      <abbrev>PL</abbrev>
      <calc get='V' set='V'/>
      <type>short</type>
    </unit>
...
        <command name="setfunc">
          <send>SEND 00 07 $addr 02 02;SEND BYTES $unit;RECV $len</send>
        </command>
```

MissMusic, 2020-10-10:
@MrIcemanLE How do you invoke the getEnergy-command? What parameters should I add? Week number, or week and day?
I added the commands and unit to my xml-files, but I can't figure out how to use it.

MrIcemanLE, 2020-10-10:
```
vclient -h 127.0.0.1 -p 3002 -c 'getEnergy 0'
vclient -h 127.0.0.1 -p 3002 -c 'getEnergy 1'
vclient -h 127.0.0.1 -p 3002 -c 'getEnergy 2'
..
vclient -h 127.0.0.1 -p 3002 -c 'getEnergy 6'
```

MissMusic, 2020-10-19:
So, I tried to log the daily energy to influxdb, but I'm missing something.
The energy seems to be aggregated for the week?
This is the converted response-data:

getEnergy 1:

year: 2020
week: 42
day: 2
Henergy: 117.2
Helect: 22.7
DHWenergy: 10.2
DHWelec: 2.5
Total_Energy: 127.4
Totalelec: 25.2

getEnergy 2:

year: 2020
week: 42
day: 3
Henergy: 180
Helect: 34.9
DHWenergy: 13
DHWelec: 3.3
Total_Energy: 193
Totalelec: 38.2

getEnergy 3:

year: 2020
week: 42
day: 4
Henergy: 222.9
Helect: 43.2
DHWenergy: 16.4
DHWelec: 4.2
Total_Energy: 239.3
Totalelec: 47.4

Is this as expected? I thought the values I got back would be for that day only.

To get the daily usage I would have to save the data for all days in a week, and at the end do some calculations like:
Day 0 = Day 0 returned values.
Day 1 = Day 1 returned values minus Day 0 returned values.
Day 2 = Day 2 returned values minus Day 1 returned values.

Sure, no problem, just get yesterdays total, subtract the total for the day before that! No, lets also reset the counter each week, My god. -.-

MrIcemanLE, 2020-10-19:
As @lanalnuel mentioned above, the control unit sums up till Sunday and restarts on monday. For my opinion the weekly dataf is enough. How much more information do you get from daily data.

How do you get the single values out of the response? Are you doing some post processing or do you have a solution inside the vclient?

MissMusic, 2020-10-19:
Ok, yea, I was just not really following what was stated.

I'm using Node-Red and it connects through telnet.

After changing the response to a string and filtering out the response-message i split it to an array and stitch it back together. It truly is a Frankenstein kind of solution, but it do seem to work.

`let arr = msg.payload.split(/[ ,]+/);
let yearend = parseInt(arr[2], 16);
let year = Number("20"+yearend);
let day = parseInt(arr[1], 16);
let week = parseInt(arr[3], 16);
let Henergy = Number((parseInt((arr[5]+arr[4]), 16)*0.1).toFixed(1));
let Helect = Number((parseInt((arr[7]+arr[6]), 16)*0.1).toFixed(1));
let DHWenergy = Number((parseInt((arr[9]+arr[8]), 16)*0.1).toFixed(1));
let DHWelec = Number((parseInt((arr[11]+arr[10]), 16)*0.1).toFixed(1));
let totalenergy = Number((Henergy + DHWenergy).toFixed(1));
let totalelec = Number((Helect + DHWelec).toFixed(1));

msg.energydata = { "year": year, "week": week, "day": day, "Henergy": Henergy, "Helect":Helect, "DHWenergy":DHWenergy, "DHWelec":DHWelec, "Total_Energy":totalenergy, "Totalelec":totalelec};

return msg;`

MissMusic, 2020-10-21:
So i made it!
A bit OT but might be interesting to someone.
I now have a flow in Node-Red that outputs daily energy readings.
It runs each day, fetching yesterdays data and stores it in an influxdb. If yesterday is a monday, it sends the data directly after formating, if its any other day of the week, it also fetches the day before yesterday and subtracts it from yesterdays data before sending it to influx.
It can then be visualized with Grafana as such:
[image]

ultrazauberer, 2020-10-21:
> So i made it!
> A bit OT but might be interesting to someone.
> I now have a flow in Node-Red that outputs daily energy readings.
> It runs each day, fetching yesterdays data and stores it in an influxdb. [...]

Very nice chart! Can you show us your Node-Red flow? I'm using vcontrold to send the data directly via CURL to my influxdb and than display the values in a node-red chart. It would be nice if I could use Node-Red directly. :-)

MissMusic, 2020-10-21:
Here is my flow, other than Node-Red i use some Home Assistant features to set a running state (that stops my flow collecting temperatures from the pump to connecting at the same time), and also to save the date of last successfull run, that way the flow can trigger every 20 mins but only run one time per day, and by doing so it's not missed if the system happens to be offline/not working on the exact time a once a day-job is triggered.

`[{"id":"2f25d7ff.357ec","type":"tab","label":"Flow 2","disabled":false,"info":""},{"id":"1ebf0b82.924184","type":"inject","z":"2f25d7ff.357ec","name":"","repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"date","x":92,"y":480,"wires":[["2c6a9f5f.e294c"]]},{"id":"2c6a9f5f.e294c","type":"change","z":"2f25d7ff.357ec","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":272,"y":480,"wires":[["cd37e235.3786b"]]},{"id":"cd37e235.3786b","type":"function","z":"2f25d7ff.357ec","name":"quit session","func":"msg.payload = 'quit\\n';\nnode.send(msg);\n","outputs":1,"noerr":0,"x":470,"y":480,"wires":[["1c69f4ef.e51273"]]},{"id":"10b141ad.427f06","type":"function","z":"2f25d7ff.357ec","name":"getEnergy","func":"msg.measurement = \"HeatEnergy\"\nmsg.payload = 'getEnergy '+msg.yesterdaynum+'\\n';\nnode.send(msg);","outputs":1,"noerr":0,"initialize":"","finalize":"","x":470,"y":512,"wires":[["1c69f4ef.e51273"]]},{"id":"1c69f4ef.e51273","type":"tcp request","z":"2f25d7ff.357ec","server":"192.168.130.3","port":"3002","out":"sit","splitc":"0","name":"","x":718,"y":512,"wires":[["f3ab6ab9.c8939"]]},{"id":"f3ab6ab9.c8939","type":"function","z":"2f25d7ff.357ec","name":"convert payload to string","func":"msg.payload = msg.payload.toString();\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":718,"y":560,"wires":[["4da4fb16.c72474"]]},{"id":"4da4fb16.c72474","type":"switch","z":"2f25d7ff.357ec","name":"Filter output","property":"payload","propertyType":"msg","rules":[{"t":"cont","v":"good","vt":"str"},{"t":"eq","v":"vctrld>","vt":"str"},{"t":"cont","v":"OK","vt":"str"},{"t":"cont","v":"Degrees","vt":"str"},{"t":"cont","v":"02 0","vt":"str"},{"t":"nnull"}],"checkall":"false","repair":false,"outputs":6,"x":678,"y":640,"wires":[["e0b48d77.082bd"],[],[],[],["b5232be0.f34818"],[]]},{"id":"b5232be0.f34818","type":"function","z":"2f25d7ff.357ec","name":"Split to array","func":"let arr = msg.payload.split(/[ ,]+/);\nlet yearend = parseInt(arr[2], 16);\nlet year = Number(\"20\"+yearend);\nlet day = parseInt(arr[1], 16);\nlet week = parseInt(arr[3], 16);\nlet Henergy = Number((parseInt((arr[5]+arr[4]), 16)*0.1).toFixed(1));\nlet Helect = Number((parseInt((arr[7]+arr[6]), 16)*0.1).toFixed(1));\nlet DHWenergy = Number((parseInt((arr[9]+arr[8]), 16)*0.1).toFixed(1));\nlet DHWelec = Number((parseInt((arr[11]+arr[10]), 16)*0.1).toFixed(1));\nlet totalenergy = Number((Henergy + DHWenergy).toFixed(1));\nlet totalelec = Number((Helect + DHWelec).toFixed(1));\n\nmsg.energydata = { \"year\": year, \"week\": week, \"day\": day, \"Henergy\": Henergy, \"Helect\":Helect, \"DHWenergy\":DHWenergy, \"DHWelec\":DHWelec, \"Total_Energy\":totalenergy, \"Totalelec\":totalelec};\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":678,"y":704,"wires":[["b886a449.bbff98"]]},{"id":"d651babf.b68da8","type":"change","z":"2f25d7ff.357ec","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"energydata","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":672,"y":832,"wires":[["712b3f02.e03bb"]]},{"id":"712b3f02.e03bb","type":"function","z":"2f25d7ff.357ec","name":"Convert to date and timestamp","func":"datetime = weekDateToDate (msg.payload.year, msg.payload.week, msg.payload.day)\n\nfunction weekDateToDate (year, week, day) {\n  const firstDayOfYear = new Date(year, 0, 1)\n  const days = 2 + day + (week - 1) * 7 - firstDayOfYear.getDay()\n  return new Date(year, 0, days)\n}\ndatetime.setHours(datetime.getHours() - 2);\ndatetime.setDate(datetime.getDate()+1);\n\ntimestamp = Date.parse(datetime)\nmsg.payload.date = datetime\nmsg.payload.timestamp = timestamp\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":562,"y":928,"wires":[["3bc8afd3.fa8ac"]]},{"id":"3bc8afd3.fa8ac","type":"function","z":"2f25d7ff.357ec","name":"Filter payload to be sent influx","func":"msg.payload = {\n    Henergy: msg.payload.Henergy,\n    Helect: msg.payload.Helect,\n    DHWenergy: msg.payload.DHWenergy,\n    DHWelec: msg.payload.DHWelec,\n    Total_Energy: msg.payload.Total_Energy,\n    Totalelec: msg.payload.Totalelec,\n    time: msg.payload.timestamp\n}\nreturn msg;\n","outputs":1,"noerr":0,"initialize":"","finalize":"","x":546,"y":1040,"wires":[["f80b4b5.7b403b8","1dfd3e66.eeda0a","c8716c14.071938","d01284c1.819a48"]]},{"id":"f80b4b5.7b403b8","type":"influxdb out","z":"2f25d7ff.357ec","influxdb":"4a253c7f.2f38fc","name":"Send to energy measurement","measurement":"","precision":"ms","retentionPolicy":"","x":882,"y":1040,"wires":[]},{"id":"c8716c14.071938","type":"debug","z":"2f25d7ff.357ec","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","statusVal":"","statusType":"auto","x":822,"y":1088,"wires":[]},{"id":"1dfd3e66.eeda0a","type":"delay","z":"2f25d7ff.357ec","name":"","pauseType":"delay","timeout":"1000","timeoutUnits":"milliseconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst [... cut ...] "str"}],"checkall":"false","repair":false,"outputs":2,"x":698,"y":736,"wires":[["9035240b.7ecd68"],["7cf879e7.c8ca3"]]},{"id":"4a253c7f.2f38fc","type":"influxdb","z":"","hostname":"a0d7b954-influxdb","port":"8086","protocol":"http","database":"viessmann","name":"Influx-viessmann","usetls":false,"tls":""},{"id":"638e1dd1.bf0a54","type":"server","z":"","name":"Home Assistant","legacy":false,"addon":true,"rejectUnauthorizedCerts":true,"ha_boolean":"y|yes|true|on|home|open","connectionDelay":true}]`

dima-357, 2020-11-21:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

arfoll, 2021-12-19:
I'd be very interested in your latest vito.xml configuration if you have it for the 0x204D (200-S)

StefanLSA, 2022-03-06:
I am trying to implement the function calls for my Vitocal 333 G heat pump, which seems to be compatible with WO1C. I forked https://github.com/joppi588/pyvcontrol (i will not release my fork until it works)

Can you tell me if this request is the correct one?

41 08 00 07 b8 00 02 02 02 cd

Because I always get 0x15 as result (=Error). I'd tryed serveral combinations, but I always get an 0x15. Is the description in https://github.com/openv/openv/wiki/FunctionCalls-WO1C complete? Do I need the lenght byte of the result?

Wolf-Raider, 2022-03-06:
i think your length is wrong.
Try 41 07 00 07 b8 00 02 02 02 cc

ultrazauberer, 2022-03-06:
> I'd be very interested in your latest vito.xml configuration if you have it for the 0x204D (200-S)
[vcontrold.zip](https://github.com/openv/openv/files/8193410/vcontrold.zip) I added the latest version of my XML files. The functioncall is working. You need to do the byte magic in post.

> I am trying to implement the function calls for my Vitocal 333 G heat pump, which seems to be compatible with WO1C. I forked https://github.com/joppi588/pyvcontrol (i will not release my fork until [...]

I just added the functioncall command to my vito.xml and vcontrold.xml. Maybe I can support you further if the answer from Wolf-Raider isn't enough.

@all
Is there a way to compute the right values of the functioncall in vcontrold? Or is it only possible in post? I had a long break (almost 2 years) and I'm redoing all the stuff to port vcontrold to my new raspberry pi. In my current mindset I think a docker container for vcontrold would be nice because I'm already running IOTstack (with NodeRed, influxdb, ...). I've never done this and any tips are welcome.

dima-357, 2022-03-07:
I think you are not accessing the system correctly.
the system always answers 0x5, after a correct call it will answer 0x16, which means it is listening, and then you can send request telegrams. Read the protocol, everything is written there
 
 
С уважением. 
Кислов Дмитрий Владимирович  
т. +7 9535212450 , +7 9218227202

Best regards  
Kislov Dmitry Vladimirovich  
т. +7 9535212450 , +7 9218227202

StefanLSA, 2022-03-07:
> i think your length is wrong. Try 41 07 00 07 b8 00 02 02 02 cc

Thank you, there was an issue in pyvcontrol. Now the function call works on my Vitocal333G. Take a look at my fork https://github.com/StefanLSA/pyvcontrol ... I'm able to read out the function call with variable response length

In viTools.py there is a scanner for the function call. The results are very interesting:
- the first parameter seems to be working from 0 to 12
- function 0 and 1 contains t.in + t.out from "Betriebstagebuch"
- function 4 an 6 look interesting, but I don't know what this values mean ... outside temp? "Vorlauf?" "Rücklauf?"

HEX  -- "Header" - (decoded as signed shorts) / [the same values div 10]

==========Function # 0=========== Betriebstagebuch? (t.in, t.out, WP1?, WP2?, AC?, NC?)
00 00 16 0a 1a 00 07 00 54 01 00 00 00 00 00 00 -- (0, 0, 22, 10) - (26, 7, 340, 0, 0, 0) / [2.6, 0.7, 34.0, 0.0, 0.0, 0.0]
00 01 16 09 19 00 06 00 69 01 00 00 00 00 00 00 -- (0, 1, 22, 9) - (25, 6, 361, 0, 0, 0) / [2.5, 0.6, 36.1, 0.0, 0.0, 0.0]
00 02 16 09 19 00 06 00 48 01 00 00 00 00 00 00 -- (0, 2, 22, 9) - (25, 6, 328, 0, 0, 0) / [2.5, 0.6, 32.8, 0.0, 0.0, 0.0]
00 03 16 09 1a 00 07 00 69 01 00 00 00 00 00 00 -- (0, 3, 22, 9) - (26, 7, 361, 0, 0, 0) / [2.6, 0.7, 36.1, 0.0, 0.0, 0.0]
00 04 16 09 17 00 04 00 82 01 00 00 00 00 00 00 -- (0, 4, 22, 9) - (23, 4, 386, 0, 0, 0) / [2.3, 0.4, 38.6, 0.0, 0.0, 0.0]
00 05 16 09 17 00 04 00 50 01 00 00 00 00 00 00 -- (0, 5, 22, 9) - (23, 4, 336, 0, 0, 0) / [2.3, 0.4, 33.6, 0.0, 0.0, 0.0]
==========Function # 1=========== Betriebstagebuch? value 1+2 are identical to func 0
01 00 16 0a 1a 00 07 00 07 00 -- (1, 0, 22, 10) - (26, 7, 7) / [2.6, 0.7, 0.7]
01 01 16 09 19 00 06 00 02 00 -- (1, 1, 22, 9) - (25, 6, 2) / [2.5, 0.6, 0.2]
01 02 16 09 19 00 06 00 05 00 -- (1, 2, 22, 9) - (25, 6, 5) / [2.5, 0.6, 0.5]
01 03 16 09 1a 00 07 00 fc ff -- (1, 3, 22, 9) - (26, 7, -4) / [2.6, 0.7, -0.4]
01 04 16 09 17 00 04 00 04 00 -- (1, 4, 22, 9) - (23, 4, 4) / [2.3, 0.4, 0.4]
01 05 16 09 17 00 04 00 09 00 -- (1, 5, 22, 9) - (23, 4, 9) / [2.3, 0.4, 0.9]
==========Function # 2=========== Energy
02 00 16 0a 12 01 33 00 56 00 19 00 00 00 00 00 -- (2, 0, 22, 10) - (274, 51, 86, 25, 0, 0) / [27.4, 5.1, 8.6, 2.5, 0.0, 0.0]
02 01 16 09 73 02 74 00 5c 00 1b 00 00 00 00 00 -- (2, 1, 22, 9) - (627, 116, 92, 27, 0, 0) / [62.7, 11.6, 9.2, 2.7, 0.0, 0.0]
02 02 16 09 92 03 aa 00 99 00 2d 00 00 00 00 00 -- (2, 2, 22, 9) - (914, 170, 153, 45, 0, 0) / [91.4, 17.0, 15.3, 4.5, 0.0, 0.0]
02 03 16 09 ec 04 ec 00 c3 00 36 00 00 00 00 00 -- (2, 3, 22, 9) - (1260, 236, 195, 54, 0, 0) / [126.0, 23.6, 19.5, 5.4, 0.0, 0.0]
02 04 16 09 59 06 30 01 f1 00 41 00 00 00 00 00 -- (2, 4, 22, 9) - (1625, 304, 241, 65, 0, 0) / [162.5, 30.4, 24.1, 6.5, 0.0, 0.0]
02 05 16 09 74 07 66 01 37 01 56 00 00 00 00 00 -- (2, 5, 22, 9) - (1908, 358, 311, 86, 0, 0) / [190.8, 35.8, 31.1, 8.6, 0.0, 0.0]
==========Function # 3===========
03 00 16 0a 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 0, 22, 10) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
03 01 16 09 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 1, 22, 9) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
03 02 16 09 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 2, 22, 9) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
03 03 16 09 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 3, 22, 9) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
03 04 16 09 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 4, 22, 9) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
03 05 16 09 00 00 00 00 00 00 00 00 00 00 00 00 -- (3, 5, 22, 9) - (0, 0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0, 0.0]
==========Function # 4===========
04 00 16 0a 02 00 f6 00 1b 00 1b 00 24 00 -- (4, 0, 22, 10) - (2, 246, 27, 27, 36) / [0.2, 24.6, 2.7, 2.7, 3.6]
04 01 16 09 03 00 07 01 24 00 19 00 20 00 -- (4, 1, 22, 9) - (3, 263, 36, 25, 32) / [0.3, 26.3, 3.6, 2.5, 3.2]
04 02 16 09 02 00 e8 00 2b 00 14 00 1e 00 -- (4, 2, 22, 9) - (2, 232, 43, 20, 30) / [0.2, 23.2, 4.3, 2.0, 3.0]
04 03 16 09 02 00 2d 01 21 00 16 00 01 00 -- (4, 3, 22, 9) - (2, 301, 33, 22, 1) / [0.2, 30.1, 3.3, 2.2, 0.1]
04 04 16 09 02 00 32 01 2f 00 1b 00 01 00 -- (4, 4, 22, 9) - (2, 306, 47, 27, 1) / [0.2, 30.6, 4.7, 2.7, 0.1]
04 05 16 09 03 00 e8 00 26 00 18 00 25 00 -- (4, 5, 22, 9) - (3, 232, 38, 24, 37) / [0.3, 23.2, 3.8, 2.4, 3.7]
==========Function # 5===========
05 00 16 0a 00 00 00 00 00 00 00 00 00 00 -- (5, 0, 22, 10) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
05 01 16 09 00 00 00 00 00 00 00 00 00 00 -- (5, 1, 22, 9) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
05 02 16 09 00 00 00 00 00 00 00 00 00 00 -- (5, 2, 22, 9) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
05 03 16 09 00 00 00 00 00 00 00 00 00 00 -- (5, 3, 22, 9) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
05 04 16 09 00 00 00 00 00 00 00 00 00 00 -- (5, 4, 22, 9) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
05 05 16 09 00 00 00 00 00 00 00 00 00 00 -- (5, 5, 22, 9) - (0, 0, 0, 0, 0) / [0.0, 0.0, 0.0, 0.0, 0.0]
==========Function # 6===========
06 00 16 0a 57 00 fc 00 00 00 00 00 -- (6, 0, 22, 10) - (87, 252, 0, 0) / [8.7, 25.2, 0.0, 0.0]
06 01 16 09 47 00 22 01 00 00 00 00 -- (6, 1, 22, 9) - (71, 290, 0, 0) / [7.1, 29.0, 0.0, 0.0]
06 02 16 09 3d 00 0a 01 00 00 00 00 -- (6, 2, 22, 9) - (61, 266, 0, 0) / [6.1, 26.6, 0.0, 0.0]
06 03 16 09 28 00 40 01 00 00 00 00 -- (6, 3, 22, 9) - (40, 320, 0, 0) / [4.0, 32.0, 0.0, 0.0]
[... cut ...]
0b 03 16 09 00 00 00 00 -- (11, 3, 22, 9) - (0, 0) / [0.0, 0.0]
0b 04 16 09 00 00 00 00 -- (11, 4, 22, 9) - (0, 0) / [0.0, 0.0]
0b 05 16 09 00 00 00 00 -- (11, 5, 22, 9) - (0, 0) / [0.0, 0.0]
==========Function # 12===========
0c 00 16 0a 00 00 -- (12, 0, 22, 10) - (0,) / [0.0]
0c 01 16 09 00 00 -- (12, 1, 22, 9) - (0,) / [0.0]
0c 02 16 09 00 00 -- (12, 2, 22, 9) - (0,) / [0.0]
0c 03 16 09 00 00 -- (12, 3, 22, 9) - (0,) / [0.0]
0c 04 16 09 00 00 -- (12, 4, 22, 9) - (0,) / [0.0]
0c 05 16 09 00 00 -- (12, 5, 22, 9) - (0,) / [0.0]

Wolf-Raider, 2022-03-23:
i think 3,5,7 will show values, if you have a two stage system with a second compressor.
4 and 6 are compressor stuff. i found out that the values in 4 are the same as in screen "Belastungsklassen" and 6 are the runtime in minutes for hot water, heating, cooling, xxx.

h0nIg, 2022-09-10:
i have implemented it the following way (without day parameter for the lazy ones)
This implementation requires https://github.com/openv/vcontrold/pull/109 and https://github.com/openv/vcontrold/pull/110

```
    <command name="getWeekDay" protocmd="getaddr">
      <addr>08E0</addr>
      <len>8</len>
      <unit>WD</unit>
      <description>Ermittle Wochentag</description>
    </command>
    <command name="getWocheEnergieEWW" protocmd="setfunc">
        <addr>B800</addr>
        <len>16</len>
        <unit>FEEWW</unit>
        <precommand>getWeekDay</precommand>
        <description>Statistik - Energiebilanz: Elektroenergie "WW-Betrieb 1 Woche", Verdichter 1</description>
    </command>
    <command name="getWocheEnergieEHeizen" protocmd="setfunc">
        <addr>B800</addr>
        <len>16</len>
        <unit>FEEH</unit>
        <precommand>getWeekDay</precommand>
        <description>Statistik - Energiebilanz: Elektroenergie "Heizbetrieb 1 Woche", Verdichter 1</description>
    </command>
    <command name="getWocheEnergieWWW" protocmd="setfunc">
        <addr>B800</addr>
        <len>16</len>
        <unit>FEWWW</unit>
        <precommand>getWeekDay</precommand>
        <description>Statistik - Energiebilanz: Waermeenergie "WW-Betrieb 1 Woche", Verdichter 1</description>
    </command>
    <command name="getWocheEnergieWHeizen" protocmd="setfunc">
        <addr>B800</addr>
        <len>16</len>
        <unit>FEWH</unit>
        <precommand>getWeekDay</precommand>
        <description>Statistik - Energiebilanz: Waermeenergie "Heizbetrieb 1 Woche", Verdichter 1</description>
    </command>
```

```
    <unit name='weekday'>
      <abbrev>WD</abbrev>
      <icalc get="B4"/>
      <type>char</type>
      <entity>Day of the week</entity>
    </unit>
    <unit name='FuncEnergyEWW'>
      <abbrev>FEEWW</abbrev>
      <icalc get="(BB &lt;&lt; 8 | BA)" set="P4"/>
      <type>uint</type>
      <entity>kW</entity>
    </unit>
    <unit name='FuncEnergyEHeat'>
      <abbrev>FEEH</abbrev>
      <icalc get="(B7 &lt;&lt; 8 | B6)" set="P4"/>
      <type>short</type>
      <entity>kW</entity>
    </unit>
    <unit name='FuncEnergyWWW'>
      <abbrev>FEWWW</abbrev>
      <icalc get="(B9 &lt;&lt; 8 | B8)" set="P4"/>
      <type>short</type>
      <entity>kW</entity>
    </unit>
    <unit name='FuncEnergyWHeat'>
      <abbrev>FEWH</abbrev>
      <icalc get="(B5 &lt;&lt; 8 | B4)" set="P4"/>
      <type>short</type>
      <entity>kW</entity>
    </unit>
```

$unit is important here, it was missing within the previous implementation
```
        <command name="setfunc">
          <send>SEND 00 07 $addr 02 02;SEND BYTES $unit;RECV $len $unit</send>
        </command>
```

MrFly72, 2022-09-12:
This sound very interesting and I would like to test it with a Vitodens 300 20CB.
But am I missing something or where is the definition of the unit WD (which I think is weekday?).
In all my config this is not included, and though I get an error if I edit these fields in addition.
Any idea?

h0nIg, 2022-09-12:
@MrFly72 update above

MrFly72, 2022-09-13:
Cool. Managed to put it into my files. now.
But the addresses seem to be different for a 20CB.
Again on the hunt for the right Addresses ....

h0nIg, 2022-09-13:
@MrFly72 i can just reference content like https://github.com/openv/openv/issues/409, get a vitosoft demo and checkout these ecn* files. You then try to get the SystemTime, which i had to search for as well. E.g. it is different across some devicdes: https://github.com/openv/vcontrold/blob/master/xml/300/vito.xml#L1206

MrFly72, 2022-09-13:
Found the time, it's 088E for my device.
Did you find the energy values also in the extracts?
I will look at it tomorrow when sitting in front of pc. On iPad it's rather great ;-)

maringera, 2022-09-16:
@h0nIg 
Would you be so kind and provide us the XML for vcontrold?
Many thanks

MrFly72, 2022-09-17:
@h0nIg do you remember which ecn you found the usage value in and what it was named at?
I looked at the ecn and the DP... file, but they are huge and I dit not find the logic so far how to find your values you have up there for "Verbrauch", as this might help me to find the 20CB values.

h0nIg, 2022-09-17:
@MrFly72 they are not documented, found them via this issue and the referenced pyvcontrol PR which add the same functionality somewhere else.
https://github.com/openv/openv/wiki/Adressen-Ger%C3%A4te-VT-200-(WO1C)
https://github.com/openv/openv/wiki/Adressen-W%C3%A4rmepumpen-Vitotronic-WO1C

@all i have fixed the workday input, as it is at least a problem for me during 0am-1am, because the value suddenly drops to "0":

```
    <unit name='FuncEnergyEWW'>
      <abbrev>FEEWW</abbrev>
      <calc get="((BB*2*2*2*2*2*2*2*2) + BA) / 10"/><!-- 2^8 is a bit shift of 8 -->
      <!--
        P4 = workday
        P5 = hour
        The values are "0" between 0am-1am and we need to take the value from the previous day for continous reporting.
        x | (~x + 1) will cause a bit flip in the most significant bit IF the value is greater than 0.
        We need to shift this specific bit to the last significant bit, negate it to have a 1 byte result and AND it to receive a "1" between 0am-1am and "0" for the rest of the time.
        If we are at week day "0" we can not do this and can continue to use "0" as value.
      -->
      <icalc set="P4 - (((~((P5 | ((~P5) + 1)) >> 31)) &amp; 1) &amp; (((P4 | ((~P4) + 1)) >> 31) &amp; 1))"/>
      <type>short</type>
      <entity>kWh</entity>
    </unit>
```

full file: https://github.com/h0nIg/vcontrol-vitoconnect-bridge/blob/main/vcontrold.xml

vich-667, 2022-12-23:
Did someone used the datapoints `Aussentemperatur 0x01C1`? In all pages I could find here it's described as a two bye value. But on Monitoring the communication of the Vitoconnect to my **Vitotronic 200 (WO1C)** I always see **3Bytes**.

When I try to read from the adress I get an ERROR as response when I ask for 2 bytes. If I request 3 bytes I get an response.

```
Aussentemperatur            0x01C1  decoding of 3 bytes unclear 0x06 0x00 0x40, might mean 0,6, but what is 0x40?
Anlagenvorlauftemperatur    0x01CA  decoding o f3 bytes unclear 0x00 0x00 0x46
Pufferspeichertemperatur    0x01CB  decoding of 3 bytes unclear 0x00 0x00 0x46
Warmwassertemperaturoben    0x01CD  decoding of 3 bytes unclear 0xde 0x01 0x40, first two bytes are 47.8 what makes sense, but what is 0x40?
Warmwassertemperaturunten   0x01CE  decoding of 3 bytes unclear 0x00 0x00 0x46
RaumtemperaturHK1           0x01D6  decoding of 3 bytes unclear 0x00 0x00 0x86
```

So it seems to me that the first bytes are the *UT* value, but does anyone know what is the meaning of the third byte?

Wolf-Raider, 2022-12-23:
Looks like a status byte. 0x40 for 'ok' 0x46\0x86 for 'not connected'

ultrazauberer, 2023-04-19:
> i have implemented it the following way (without day parameter for the lazy ones) This implementation requires [openv/vcontrold#109](https://github.com/openv/vcontrold/pull/109) and [openv/vcontrold#110](https://github.com/openv/vcontrold/pull/110)
> 
> ```
> $ vclient -c 'getWocheEnergieEWW'
> getWocheEnergieEWW:
> 11.800000 kWh
> ```
> 
> ```
>     <command name="getWeekDay" protocmd="getaddr">
>       <addr>08E0</addr>
>       <len>8</len>
>       <unit>WD</unit>
>       <description>Ermittle Wochentag</description>
>     </command>
>     <command name="getWocheEnergieEWW" protocmd="setfunc">
>         <addr>B800</addr>
>         <len>16</len>
>         <unit>FEEWW</unit>
>         <precommand>getWeekDay</precommand>
>         <description>Statistik - Energiebilanz: Elektroenergie "WW-Betrieb 1 Woche", Verdichter 1</description>
>     </command>
>     <command name="getWocheEnergieEHeizen" protocmd="setfunc">
>         <addr>B800</addr>
>         <len>16</len>
>         <unit>FEEH</unit>
>         <precommand>getWeekDay</precommand>
>         <description>Statistik - Energiebilanz: Elektroenergie "Heizbetrieb 1 Woche", Verdichter 1</description>
>     </command>
>     <command name="getWocheEnergieWWW" protocmd="setfunc">
>         <addr>B800</addr>
>         <len>16</len>
>         <unit>FEWWW</unit>
>         <precommand>getWeekDay</precommand>
>         <description>Statistik - Energiebilanz: Waermeenergie "WW-Betrieb 1 Woche", Verdichter 1</description>
>     </command>
>     <command name="getWocheEnergieWHeizen" protocmd="setfunc">
>         <addr>B800</addr>
>         <len>16</len>
>         <unit>FEWH</unit>
>         <precommand>getWeekDay</precommand>
>         <description>Statistik - Energiebilanz: Waermeenergie "Heizbetrieb 1 Woche", Verdichter 1</description>
>     </command>
> ```
> 
> ```
>     <unit name='weekday'>
>       <abbrev>WD</abbrev>
>       <icalc get="B4"/>
>       <type>char</type>
>       <entity>Day of the week</entity>
>     </unit>
>     <unit name='FuncEnergyEWW'>
>       <abbrev>FEEWW</abbrev>
>       <calc get="((BB*2*2*2*2*2*2*2*2) + BA) / 10"/> 
>       <icalc set="P4"/>
>       <type>short</type>
>       <entity>kWh</entity>
>     </unit>
>     <unit name='FuncEnergyEHeat'>
>       <abbrev>FEEH</abbrev>
>       <calc get="((B7*2*2*2*2*2*2*2*2) + B6) / 10"/>
>       <icalc set="P4"/>
>       <type>short</type>
>       <entity>kWh</entity>
>     </unit>
>     <unit name='FuncEnergyWWW'>
>       <abbrev>FEWWW</abbrev>
>       <calc get="((B9*2*2*2*2*2*2*2*2) + B8) / 10"/>
>       <icalc set="P4"/>
>       <type>short</type>
>       <entity>kWh</entity>
>     </unit>
>     <unit name='FuncEnergyWHeat'>
>       <abbrev>FEWH</abbrev>
>       <calc get="((B5*2*2*2*2*2*2*2*2) + B4) / 10"/>
>       <icalc set="P4"/>
>       <type>short</type>
>       <entity>kWh</entity>
>     </unit>
> ```
> 
> $unit is important here, it was missing within the previous implementation
> 
> ```
>         <command name="setfunc">
>           <send>SEND 00 07 $addr 02 02;SEND BYTES $unit;RECV $len $unit</send>
>         </command>
> ```
> 
> if you need to check for a specific weekday, you can define the unit in the following way and remove the precommand. You do not depend on any of the PR's listed above:
> 
> ```
>     <unit name='FuncEnergyEWW'>
>       <abbrev>FEEWW</abbrev>
>       <calc get="((B7*2*2*2*2*2*2*2*2) + B6) / 10" set="V"/>
>       <type>short</type>
>       <entity>kW</entity>
>     </unit>
> ```
> 
> ```
> $ vclient -c 'getWocheEnergieEWW 0'
> getWocheEnergieEWW 0:
> 0012 kW
> $ vclient -c 'getWocheEnergieEWW 1'
> getWocheEnergieEWW 1:
> 001D kW
> $ vclient -c 'getWocheEnergieEWW 2'
> getWocheEnergieEWW 2:
> 002B kW
> $ vclient -c 'getWocheEnergieEWW 3'
> getWocheEnergieEWW 3:
> 003C kW
> $ vclient -c 'getWocheEnergieEWW 4'
> getWocheEnergieEWW 4:
> 004C kW
> $ vclient -c 'getWocheEnergieEWW 5'
> getWocheEnergieEWW 5:
> 0062 kW
> $ vclient -c 'getWocheEnergieEWW 6'
> getWocheEnergieEWW 6:
> 0081 kW
> ```

Awesome work @h0nIg! I updated yesterday my vcontrold and using now your provided xml files.
Now I can use your newly implemented methods.

One question: when does the function call makes most sense? I guess sundays at 23:59? Because then you have the most complete data from the week? Or is there an option to get data from last week?

ultrazauberer, 2023-04-24:
Ok, now I got my problem confirmed. The heating pump was working between 23:00 Sunday and 01:00 Monday. My cronjob gets the values every hour and 59 minutes.
In my database the value for heating energy is 39 kWh for last week and on the heating pump itself it shows me 40,1 kWh for last week. What can I do to get the right value for last week? Is there the possibility to read out the values from last week?

budisek, 2024-01-19:
@MrIcemanLE Hi, please did you find any way how to read energy consumption data? I have the same device WO1C (DevType 204C).

MrIcemanLE, 2024-02-19:
I use the Node-Red Flow from @MissMusic [Link to comment](https://github.com/openv/openv/issues/480#issuecomment-713443463) and my commands from [here](https://github.com/openv/openv/issues/480#issuecomment-705099106)

timbrd, 2024-10-04:
> Did someone used the datapoints `Aussentemperatur 0x01C1`? In all pages I could find here it's described as a two bye value. But on Monitoring the communication of the Vitoconnect to my **Vitotronic [...]

Did you ever find a way to read these hot water related temperature values? I am hitting the [same problem](https://github.com/philippoo66/optolink-splitter/issues/44).

timbrd, 2024-10-04:
> Did someone used the datapoints `Aussentemperatur 0x01C1`? In all pages I could find here it's described as a two bye value. But on Monitoring the communication of the Vitoconnect to my **Vitotronic [...]

@vich-667 Did you ever find a way to read these hot water related temperature values? I am hitting the [same problem](https://github.com/philippoo66/optolink-splitter/issues/44).

[4 more replies]