openv/openv #482: Alternative zu SFH487-2

fitim7, 2020-03-19:
Was kann ich hier als alternative für die SFH487-2 nutzen? Leider werde ich nicht fündig bei der SFH487-2.

bertmelis, 2020-03-19:
I might have some spares. Can't send them now though. Postal services blocked outbound international items.

speters, 2020-03-20:
Die Schaltung ist da recht anspruchslos, nimm einfach eine andere IR-Diode.
Meinen eigenen Adapter macht es seit Jahren mit IR-Diode und -Transistor aus der Gabellichtschranke (kann man trennen) einer alten mechanischen Maus.

The circuit is quite simple, just use another IR diode.
My own adapter has been making it for years with IR diode and IR transistor from a photo interrupter (cut in half) of an old mechanical mouse.

fitim7, 2020-03-20:
> Die Schaltung ist da recht anspruchslos, nimm einfach eine andere IR-Diode.
> Meinen eigenen Adapter macht es seit Jahren mit IR-Diode und -Transistor aus der Gabellichtschranke (kann man trennen) einer alten mechanischen Maus.
> Eine andere Quelle könnte eine Sendediode einer alten IR-Fernbedienung sein. Nur den Empfänger kann man hier nicht herauspulen, die TSOP1736-ähnlichen Empfänger sind für diesen Zweck nutzlos.
> 
> The circuit is quite simple, just use another IR diode.
> My own adapter has been making it for years with IR diode and IR transistor from a photo interrupter (cut in half) of an old mechanical mouse.
> Another source is the IR diode of an IR remote control. But this does not come with a phottransistor, the TSOP1738 style receivers are useless for this task.

Danke! Ich muss erstmal schauen ob ich eine alte Maus noch irgendwo rumliegen habe. Da ich aber mehrer solcher Teile bauen wollte (für den Stromkasten und Gas) habe ich eher auf ein Produkt oder so gehofft, dass es im 10er Pack für um die 5 Euro gibt. Leider verwirrt mich da einfach die Wellenlänge usw. ... Könnte man z.B. die SFH 4346 nutzen? Diese hat aber eine Wellenlänge von 940nm, anstatt 880nm. Wie gesagt, leider kenne ich mich nicht aus was das angeht.

speters, 2020-03-20:
Wenn du mehrere Geräte bauen möchtest, dann bestelle doch die Bauteile bei einem Elektronikversender (mouser, reichelt.de, tme.eu ... s.a. https://www.mikrocontroller.net/articles/Elektronikversender) .

Deine SFH 4346 hätte im Bereich der maximalen Empfindlichkeit des heizungsseitigen Phototransistors  (880nm?) keine 10% der maximalen Strahlstärke.
Das könnte noch langen, aber wenn du eh Bauteile besorgen musst, dann nimm doch gleich was passendes.

fitim7, 2020-03-20:
> Wenn du mehrere Geräte bauen möchtest, dann bestelle doch die Bauteile bei einem Elektronikversender (mouser, reichelt.de, tme.eu ... s.a. [...]

Das würde ich ja gerne sofort tun. Wenn ich halt nur wüsste welches Teil passend wäre, denn der SFH487-2 ist weder auf Mouser, noch auf Reichelt verfügbar. Bei Reichelt würde ich sofort bestellen, wenn ich ein alternatives passendes Teil finden würde. Kann mir da jemand vielleicht etwas raussuchen?

speters, 2020-03-20:
Reichelt hat z.B. die IRL81A da (https://www.reichelt.de/infrarotdiode-880-nm-25-side-led-irl-81-a-p60591.html?&trstct=pol_5&nbc=1), Mouser z.B HSDL4260 (https://www.mouser.de/Optoelectronics/Infrared-Data-Communications/_/N-6qrgf?Ns=Pricing%7c0), TME hat auch viele https://www.tme.eu/de/katalog/?search=880nm&s_field=1000014&s_order=asc&limit=20&currency=EUR&page=1) und auch bei Pollin wird man wahrscheinlich (https://www.pollin.de/p/sortiment-infrarot-leuchtdioden-100-teilig-800152) oder sicher (https://www.pollin.de/p/ir-sendediode-tshf5210-120744) fündig.

fitim7, 2020-03-21:
> I might have some spares. Can't send them now though. Postal services blocked outbound international items.

Btw. Thank you my friend! Yes, the actual situation is really confused all of over the world. Stay healthy!

fitim7, 2020-03-21:
> Reichelt hat z.B. die IRL81A da (https://www.reichelt.de/infrarotdiode-880-nm-25-side-led-irl-81-a-p60591.html?&trstct=pol_5&nbc=1), Mouser z.B HSDL4260 [...]

Danke für die Mühe! Ich schau mir das mal an :)

derfritz1, 2020-10-18:
Moin! Ich habe die SFH487-p von Reichelt genommen. Funktioniert zumindest in der Raspberry-Pi Schaltung 1:1 ohne Anpassungen statt der sfh487-2.

Leo-wq, 2020-11-03:
Hi, wenn ich das richtig verstanden habe wäre die optimale Wellenlänge 880nm, kann ich einfach "irgendeine" 880nm nehmen oder müsste ich dann evtl die Widerstände anpassen? Bzw worauf müsste ich da genau achten?

speters, 2020-11-03:
Solange du nicht irgendeinen exotischen Typ zu fassen hast (Hochleistungs-LED oder irgendeine superwinzig SMD LED), brauchst du dir da keinen großen Kopf machen. Vorwiderstand um 100Ω.

Falls du eine Gabellichtschranke auseinanderpulst: Das Bauteil mit dem klaren Gehäuse ist i.d.R. die LED, das andere der Phototransistor.

Leo-wq, 2020-11-03:
Würden die beiden auch funktionieren? 

https://www.conrad.de/de/p/osram-fototransistor-4-57-x-5-84-mm-1100-nm-40-lpt-80-a-153470.html

https://www.conrad.de/de/p/osram-irl-81-a-ir-emitter-880-nm-10-4-57-x-5-84-mm-radial-bedrahtet-153692.html

speters, 2020-11-03:
Grundsätzlich ja. Ich würde beim Phototransistor jedoch einen nehmen, der sichtbares Licht herausfiltert. Das sind meist die Typen mit dunklem Gehäuse. Ohne diese Filter könnte Umgebungslicht und das Licht der Status-LED der Heizung stören.

Leo-wq, 2020-11-03:
> Solange du nicht irgendeinen exotischen Typ zu fassen hast (Hochleistungs-LED oder irgendeine superwinzig SMD LED), brauchst du dir da keinen großen Kopf machen. Vorwiderstand um 100Ω.
>
> Falls du [...]

Im Plan, https://github.com/openv/openv/wiki/Bauanleitung-RaspberryPi sind allerdings 10KΩ Wiederstände drin... welche soll ich nun nehmen?

speters, 2020-11-03:
Die 10kΩ sind kein Vorwiderstand für die LED, sondern für die Tansistoren. Die sind gut so. Wenn du Widerstände besorgen musst, nimm die 10k. Wenn du was anderes zwischen 2k2 und 15k hast, dann ist das aber auch OK.

Der Vorwiderstand für die LED hat in dem o.g. Bauplan 180Ω. Da ist so ungefähr alles i.O., was in der Größenordnung liegt.
Auch hier wieder: Wenn du den Widerstand erst besorgen muss, nimm einen, der genau passt, ansonsten irgendwas zwischen 50 und 220 Ohm.

Leo-wq, 2020-11-03:
Okay, dann nehme ich also 
https://www.conrad.de/de/p/osram-fototransistor-3-mm-1100-nm-12-sfh-309-fa-153883.html (Empfänger)
https://www.conrad.de/de/p/osram-sfh-485-p-ir-emitter-880-nm-40-5-mm-radial-bedrahtet-153831.html (Sender)

Wiederstände müsste ich alle erst noch besorgen... 
Im Plan ist jetzt nur ein Vorwiederstand für den Sender, soll trotzdem noch einer für den Empfänger eingeplant werden?

Für die Transistoren hinsichtlich der Qualität eine Empfehlung?

speters, 2020-11-03:
Halte dich an den Schaltplan.
Bei den Transistoren ist die "Qualität" (falls du damit den Verstärkungsfaktor meinst - bei BC547 z.B. BC547A, BC547B, BC547C) egal. Der Typ muss halt passen (NPN oder PNP) und bei der Bauform ist wahrscheinlich TO92 Gehäuse angesagt.
Und wie gesagt: Zwingend die Beinchenbelegung bei den Transistoren nachschauen, da gibt es nämlich keine feste Regel.

interose, 2022-01-30:
Hallo zusammen,
ich war ebenfalls auf der Suche nach einem Ersatz und habe diese hier verwendet:
https://www.reichelt.de/silizium-pin-fotodiode-400-1100nm-40-5-mm-sfh-203-p60568.html

Jetzt hab ich alles mit einer NodeMCU zusammengebaut und verschiedene Dinge getestet, aber ich empfange einfach keine Werte. Anschließend nochmal genau die Beschreibung gelesen, "...und bei 880 nm (SFH 203 FA)"

Bedeutet das, dass ich eigentlich diese hier benötige:
https://www.reichelt.de/silizium-pin-fotodiode-800-1100nm-40-5-mm-sfh-203-fa-p60573.html

Oder müsste es mit beiden funktionieren und der Fehler liegt woanders. Hat jemand einen Rat für mich?

philverh, 2022-01-31:
I would avoid the photodiode which can "see" green light, except if you have a good filter in front of it to only let IR light through.
The original diode even has a "dark" body to filter out visible light. Try the 800-1100nm one.

interose, 2022-02-05:
Hi okay - I have bought the other one and tested it as well. Unfortunately without any luck.
I created the sketch as it is demonstrated at [https://github.com/openv/openv/wiki/Bauanleitung-ESP8266] but without the sending part and with a basic infrared lib. It works with a tv remote control, the raw bytes were printed at the console. But if I put the diode next to the green LED from the vitodens - nothing happens.

Does the vitodens constantly transmitting the values each second via the green LED or do I have to enable this?
For recieving values, do I need the sending part of the sketch, e.g. send a command to the vitodens "Please send me your values"?