openv/openv #483: Telnet schließt sich nach Wert-abfrage

C1500, 2020-03-22:
Hi
ich hab ein Problem mit vcontrold auf dem PI
die Telnet Verbindung bricht immer wieder ab.
Wenn ich als erstes nach öffnen der Verbindung "getTempA" eingebe kommt ein "ERR: command unknown"
Versuch ich es nochmal mit "getTempA" schließt sich die Telnet Verbindung einfach.
Kennt das jemand und kann mir hier helfen?
Gruß C1500

philverh, 2020-03-22:
That means your vito.xml does not contain that command.You need to define "getTempA" in your vito.xml, with the correct address for your device before it can be executed.
vcontrold cannot run without a correct vcontrold.xml / vito.xml pair. By "correct" I mean a pair containing the commands for the device type you want to control. 

Once telnet has started, try running "commands" to see what is defined in the loaded vito.xml. Hopefully that matches what you defined. 

When done, try running the command associated with address 0x00F8 in your vito xml. That should return the device type you have.

hmueller01, 2020-10-23:
Hi,

kann ich mit `vcontrold version 0.98.9-8-gae6873e` nicht reproduzieren:
```
> telnet localhost 3002
vctrld>getTempA
ERR: command unknown
vctrld>getTempA
ERR: command unknown
vctrld>getTempA
ERR: command unknown
vctrld>getTempATiefpass
14.000000 Grad Celsius
vctrld>
```
Bitte mal mit der letzen git Version testen. Danke.

PS: Fehler am vcontrold sollten in den openv/vcontrold issue tracker, nicht hierher.