openv/openv #484: Q: Wie funktioniert das Original-Kabel (v) mit dem ESP8266?

Huntercover, 2020-04-18:
Hi,

ich habe ein Buchse <> Buchse Kabel gebastelt und frage mich ob ich hier Rx mit Tx tauschen muss? Wie auch immer, ich die Callbacks werden nicht durchlaufen.
Hat jemand das Kabel am Laufen? Danke im Voraus !

Grüße
Andreas

Huntercover, 2020-09-27:
ich nutze die beiden Bauteile (Sender, Empänger) aus dem Kabel an einem ESP8266, funktioniert nun

adorobis, 2023-11-19:
Hello @Huntercover 
Could you please share how you have connected the original cable to ESP8266?
Many thanks!

AndreasHobby, 2023-11-19:
Hi, first of all:
Did not know why, but the recommended parts (..) are not working for me - maybe I missed to use the right resistor.

Anyway, I do not remember 100%, but I cut some lines inside the sensor and make use only of the receiver and transmitter of the original cable (as I wrote). Positive effect is that the cable matching perfectly to the Vitocal and I will never use the original cable for anything else.

Finally I just use the cable directly on free defined pins (+3.3V + GND)with the following init command:
  VitoWiFi.setup(&Serial1, 36, 4);

Regards, Andreas

adorobis, 2023-11-19:
Great, many thanks. I was searching further and seems that the cable has a built in USB-UART adapter so most likely it won't work directly with esp. I'll probably remove the inside of the cable and replace it with my own components (which work very well) - I've built it according to the following diagram: https://deploy-preview-2737--esphome.netlify.app/components/optolink.html

AndreasHobby, 2023-11-19:
you are welcome :) yes, right, this was the reason,