openv/openv #486: Vitodens 200-W / Typ HO1B

ZE135, 2020-05-07:
Hallo zusammen,
ich hoffe sehr ihr könnt mir helfen.  
Die Verbindung mit der Heizung klappt soweit. 
Ich verwende das Protokoll P300 mit VScotHO1 (ID=20CB). Version von VCONTROLD: 0.98.2_IPv6
Einige Befehle funktioneren soweit.

Nun zu meinen Problemen: 
1.) getDeviceType gibt leider ein "UNKNOWN" zurück. 
2.) Ich versuche die Betriebsart auszulesen. Mit folgenden Adressen habe ich es versucht:
2323 -> RED
2500 -> ABSCHALT
2301 -> WW
3301 -> WW

Die Heizung steht in der Tat auf Warmwasser. Ein Umschalten bewirkt nichts. In diversen Forenbeiträgen habe ich auch gelesen, dass mit einem set die Betriebsart gesetzt werden kann. Die 
Vito habe ich folgendermaßen ergänzt:

> 
	<command name='setBetriebsartTo0' protocmd='setaddrTo0'>
             <addr>2323</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Setze Betriebsart</description>
        </command>
	<command name='setBetriebsartTo1' protocmd='setaddrTo1'>
             <addr>2323</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Setze Betriebsart</description>
        </command>
	<command name='setBetriebsartTo2' protocmd='setaddrTo2'>
             <addr>2323</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Setze Betriebsart</description>
        </command>
	<command name='setBetriebsartTo3' protocmd='setaddrTo3'>
             <addr>2323</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Setze Betriebsart</description>
        </command>
	<command name='setBetriebsartTo4' protocmd='setaddrTo4'>
             <addr>2323</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Setze Betriebsart</description>
        </command>
und in der vcontrold.xml folgendes:

	<command name="setaddr">
		     <send>SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
		</command>
		<command name="setaddrTo0">
		     <send>SETADDR $addr $hexlen 00;RECV 1</send>
		</command>
		<command name="setaddrTo1">
		     <send>SETADDR $addr $hexlen 01;RECV 1</send>
		</command>
		<command name="setaddrTo2">
		     <send>SETADDR $addr $hexlen 02;RECV 1</send>
		</command>
		<command name="setaddrTo3">
		     <send>SETADDR $addr $hexlen 03;RECV 1</send>
		</command>
		<command name="setaddrTo4">
		     <send>SETADDR $addr $hexlen 04;RECV 1</send>
		</command>
		<command name="setaddrTo5">
		     <send>SETADDR $addr $hexlen 05;RECV 1</send>
		</command>

Hat jemand eine Idee was ich falsch mache?  Ich bin echt am verzweifeln....

speters, 2020-05-08:
Bitte update zuerst auf eine aktuelle Version von `vcontrold`. Probleme mit älteren Versionen können durchaus auf Fehlern basieren, die in aktuellen Versionen ausgemerzt wurden.

Bei deiner Heizung müsste die Adressse "0x2323" für die Betriebsartumschaltung des ersten Heizkreises korrekt sein.
Die Werte sind folgendermaßen zu interpretieren: 0=Abschalt;1=Nur WW;2=Heizen + WW;3=Dauernd Reduziert;4=Dauernd Normal.

Mit deiner VScotHO1 könntest du auch mal [vitalk](https://github.com/openv/vitalk) ausprobieren, um auszuschliessen, dass es an deiner Hardware (Selbstbauadapter?) liegt.

ZE135, 2020-05-08:
Vielen Dank für Deine Rückmeldung. 
Eigentlich ist das die letzte Version. Ich hatte [diese ](https://github.com/openv/vcontrold/archive/v0.98.9.zip) installiert. Angehzeit wird allerdings immer 0.98.2. Was wird denn bei Dir angezeigt?

Warum ist die Adresse "0x2323" eigntlich nicht in der Vito.xml als standard hinterlegt?

vitalk kenne ich nicht, probiere es aber gerne aus. 

Ne Idee wo ich eine komplette und vollständige Vito- und Vcontrol.xml bekommen könnte?

ZE135, 2020-05-09:
Hat niemand eine Idee und kann mir weiterhelfen????

ZE135, 2020-05-10:
Hat niemand eine Idee, bzw. eine funktionierende XML für lesen und schreiben???

ZE135, 2020-05-17:
lets switch writing english. Hopefully I can approach more and get my issue solved. 

I am using P300 protocoll (VScotH01 ID=20CB). Installed VControlD version: v0.98.9 but in the applicaiton it shown: 0.98.2_IPv6
Most of the commands are working proberly. 
Where I am struggling is described as follows:

1.) getDeviceType return "UNKNOWN". Any clue why?
2.) I am trying to read out the operation mode. I have tried following addresses:

2323 -> RED
2500 -> ABSCHALT
2301 -> WW
3301 -> WW

I am am switching the status directly, I am not able to see the changes. I am really struggling to understand which address is correct for this matter. 

Furthermore I would like to change by set-command the status. I changed the vito.xml and vcontrol.xml accordingly (see code above).

Does anybody has any clue ?

longinius, 2020-05-17:
In [diesem Beitrag](https://forum.iobroker.net/topic/2715/viessmann-adapter/451) sind am Ende vom Post vito.xml und vcontrold.xml angehängt. Die Konfiguration ist für die VScotHO1 mit Protokoll P300. Die Konfiguration ist umfangreich und bietet so ziemlich alle Funktionen.

Das ändern der Betriebsart dauert immer eine Zeit bis die Anlage die Änderung anzeigt. Über die Schnittstelle sind die Änderungen direkt zu sehen.

OrionVIII, 2021-05-20:
hi, maybe have a look here: [https://github.com/OrionVIII/OpenV_NodeMCU](url)
It's working with the Home Assistant project (sending data via MQTT). It's made with an ESP8266 and still working, :-)

ZE135, 2021-05-22:
Hi elbowz,

not sure I undertsood you correct. Currently I am using openv as an interface to receive / send data to my boiler. 
In general this is done by telnet-commands. Since I have my complete house integrated in openhab, I am exchaning the values via Rest-API.
This is working fine for me. 
[This ](https://intranet-der-dinge.de/smarthome/haus/technik/heizung/)was a very nice "trigger" to start with the entire project.

Hope this helps...

Regards

elbowz, 2021-05-24:
> hi, maybe have a look here: [https://github.com/OrionVIII/OpenV_NodeMCU](url)
> It's working with the Home Assistant project (sending data via MQTT). It's made with an ESP8266 and still working, :-)

Better of what I hoped...It does more then what expected with my boiler, It's amazing. Thanks!

> This is working fine for me.
> [This ](https://intranet-der-dinge.de/smarthome/haus/technik/heizung/)was a very nice "trigger" to start with the entire project.

Thanks! I'll go deeply in the project and try to realize a similar one.