openv/openv #487: Find out Device ID of Viessmann Vitotrol 100 with Optolink

m4ding, 2020-05-08:
Hi,

I have a running iobroker instance and have installed vcontrold with this step by step guide (German language): https://github.com/openv/openv/wiki/vcontrold-mit-Raspberry-Pi

Everything seems to be ok except that no data is displayed in iobroker. However, the connection is ok according to iobroker. I think this is because of the incorrect Device ID which has to be correctly set in vcontrold.xml.
Example:
`<config>
   <serial>
      <tty>/dev/ttyUSB0</tty>
   </serial>
   <net>
      <port>3002</port>
      <allow ip='127.0.0.1'/>
      <allow ip='192.168.0.0/24'/>
   </net>
   **<device ID="2098"/>**
</config>`

Unfortunately, I haven't found a hint how to find out the Device ID of my Optolink cable which is connected via USB to my RPi4. 

Any help is much appreciated.

m4ding, 2020-05-08:
Hi,

I have found out my Device ID using this article: https://github.com/openv/openv/wiki/Ger%C3%A4te
However, when I run vcontrold it says:
[8136] Fri May  8 17:17:47 2020 : Device 204D nicht definiert
Fehler beim Laden von /etc/vcontrold/vcontrold.xml, terminiere!

Device 204D not defined, terminating. 
My Viessmann heater is nearly three years old. Does this Device not exist yet in vcontrol?

Mine is:

Vitotronic 200 | 204D | 300,KW2 | Vitotronic 200 (WO1C) | Wärmepumpenregelung (Vitocal 2xx-G/3xx-G mit Vitotronic 200 (Typ WO1C)
-- | -- | -- | -- | --

schaedeldx, 2022-09-26:
There are two folders with xml-files. For your configuration you have to copy the files under "300". I have the same constellation and it works.