openv/openv #488: Vitotronik 300 identity 20B8

dima-357, 2020-05-09:
Hello.
I have a boiler room. 3 boilers with Vitotronik 100, they are connected in cascade on the Vitotronik 300.
I connected to the optical port of the Vitotronik 300, and I receive data from the Vitotronik 300 at the addresses from the list (Viessmann-addresses.xls)
In this list there are addresses of the Vitotronik 300 variables and there are several addresses of the variable boilers in the cascade, for example I can see the temperature of the boilers at the addresses
boiler 1 0xA202,
boiler 2 0xA242,
boiler 3 0xA282.
But I can not see the rest of the information on the boilers.
 
Tell me please,
What addresses should I contact the Vitotronik 300
so that he gives me information on each of the boilers
- flue gas temperature
- burner in operation
- burner power
- boiler pump in operation
- temperature at the outlet of the boiler
- temperature at the inlet to the boiler
- boiler valve position
- boiler failure

bertmelis, 2020-05-09:
I don't know myself but you csn try to reverse engineer by installing Viessmann's diagnostic software. (or unzip the package)

A large XML that comes along with this software should contain a lot more datapoints.

dima-357, 2020-05-09:
Where can I get this software and more XML?

vitoopen, 2020-05-09:
@dima-357 
you need extended addressing that is not supported by vcontrold.
I have a cascade control too. One vitotronic 333 and two boilers with vitotronic 100 connected by LON.
I have a custom software for communication. 
```
 * Addressing:
 * 
 * telegram           0x41, <len>, <tg-type>, <command>, <checksum>
 * command            <cmd-type>, <address>, <datalen>
 * address            <byte1>, <byte2>
 * 
 * regular (V333MW1)  0x41,<len>,0x00,0x01,<addr1>,<addr0>,<datalen>,<checksum>
 * extended (V100GC1) 0x41,<len>,0x10,0x0A,<plant>,<device>,<cmds>,<command1..n>,<checksum>
 * 
```

my values:
plant = 1
device = 1 for first V100, 2 for second V100 (LON bus address)
cmds = number of commands

sample telegrams (the final 0xff is replaced by the checksum before sent)
```
BYTE RQ_BR1_TEMP_STUFEN[] =     // Kessel-, Abgas-Temperatur und Brennerstufen 1 u. 2
{  
   0x41,0x1D,0x10,0x0A,0x01,0x01,0x04,
   0x05,0x00,0x01,0x08,0x10,0x02,
   0x05,0x00,0x01,0x08,0x16,0x02,
   0x05,0x00,0x01,0x08,0x42,0x01,
   0x05,0x00,0x01,0x08,0x49,0x01, 
   0xFF
};
```
the same telegram for the second V100
```
BYTE RQ_BR2_TEMP_STUFEN[] =     // Kessel-, Abgas-Temperatur und Brennerstufen 1 u. 2
{  
   0x41,0x1D,0x10,0x0A,0x01,0x02,0x04,
   0x05,0x00,0x01,0x08,0x10,0x02,
   0x05,0x00,0x01,0x08,0x16,0x02,
   0x05,0x00,0x01,0x08,0x42,0x01,
   0x05,0x00,0x01,0x08,0x49,0x01, 
   0xFF
};
```

I hope this gives some ideas to solve your addressing.
The benefit of extended addressing (supported by 300er-protocol only) is that you can send multiple commands with one telegram.

bertmelis, 2020-05-09:
> 
> 
> Where can I get this software and more XML?

It's called Vitosoft 300. You can find a trial version on Google. (don't know if I'm allowed to link directly. It's hidden on Viessmann's website)

dima-357, 2020-05-10:
Addressing by LON in the cascade is obtained
plant = 1
device = 1 or 2
And the rest means device = 3, device = 4?

vitoopen, 2020-05-10:
This is just an example of my plant. The value for `device` depends on the LON address of your V100 devices.

dima-357, 2020-05-11:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

dima-357, 2020-05-20:
> This is just an example of my plant. The value for `device` depends on the LON address of your V100 devices.