openv/openv #491: Kein Signal an ESP8266

empty88, 2020-06-07:
Hallo zusammen,

ich habe mich auch an einem Optolink Adapter auf Basis eines ESP8266 (Wemos D1 mini) gegeben.
Leider bekomme ich von meiner Heizung nicht das sekündliche Signal.

Ich verwende einen SFH309FA auf der Empfangsseite und habe die Schaltung von der Variante mit dem ESP8266 von hier übernommen. Halte ich eine TV-Fernbedienung maximal 3 cm vor den Fototransistor bekomme ich die Daten. An der Heizung bekomme ich nichts.

Den Fototransistor halte ich wie vorgesehen an die grüne LED von meiner Vitodens 200.
Selbst wenn ich über einen SFH 487P OSO direkt vor dem Fototransistor Daten sende, bekomme ich nichts. Die LED funktioniert, mit Handykamera überprüft.

Es scheint mir als wären die ankommenden Signale zu schwach und ich hab keine Ahnung warum.

Ist es eigentlich richtig, das an meinem D1 mini zwischen RX und GND 3,3 V anliegen? Vielleicht ist der hinüber?

Viele Grüße

speters, 2020-06-07:
Die 3,3V zwischen RX und GND kommen dadurch zustande, dass der interne Pull-Up Widerstand vom EPS8266 im "Urzustand" angeschaltet ist.

Probier es mal ohne internen Pull-Up (Modus des Portpins auf INPUT statt INPUT_PULLUP).
Alternativ könntest auf den externen Pull-Up verzichten und nur dem internen vertrauen - da besteht dann aber eher die Gefahr, dass man den Port beschädigt.

An der Schaltung und den Bauelementen kann man eigentlich nicht viel verkehrt machen.
Liegt es vielleicht eher doch an der ESP8266 Firmware (z.B. Port als Output, Schnittstelle nicht richtig konfiguriert, ...)?

empty88, 2020-06-08:
Hallo @speters ,
vielen Dank für deine Antwort, es liegt auf jeden Fall an dem RX-PIN. Habe mal mit SoftwareSerial ein anderes PIN-Paar für die serielle Kommunikation ausprobiert und siehe da:
1. Die TV-Fernbedienung wird auch aus größerer Entfernung erkannt
2. ich bekomme endlich das langersehnte 05, 05, 05... von der Heizung

INPUT als Portmodus hat leider nichts geholfen, aber du hast mich auf den richtigen Weg gebracht.

Danke nochmal! :-)

closed 2020-06-08 by speters

interose, 2022-02-07:
@empty88 
Hi ich habe das gleiche Problem. Ich schaffe es bei meiner Vitodens einfach nicht das sekündliche Signal zu empfangen. Ich habe es mal auch mit SoftwareSerial (also diese Lib hier für den [ESP8266](https://github.com/plerup/espsoftwareserial/) ) versucht. 

Eine Fernbedienung wird erkannt, d.h. wenn ich sie davor halte und irgendwelche Tasten drücke, dann bekomme ich auf der Konsole ein paar Fragezeichen. Wenn ich die Schaltung vor die grüne LED der Vitodens halte passiert nichts. Hier mal der simple Codeschnipsel.

```
#include <SoftwareSerial.h>

#define MYPORT_TX 12
#define MYPORT_RX 13

SoftwareSerial testSerial;

void setup() {
  Serial.begin(74880);   // Open serial communications and wait for port to open:
  while (!Serial) {
    ; // wait for serial port to connect.
  }

  pinMode(MYPORT_RX, INPUT);

  testSerial.begin(4800, SWSERIAL_8N2, MYPORT_RX, MYPORT_TX, false);
  if (!testSerial) {
    Serial.println("Invalid SoftwareSerial pin configuration, check config");
    while (1) {
      delay (1000);
    }
  }
}

void loop() {
  while (testSerial.available() > 0) {
    Serial.write(testSerial.read());
  }
}
```

Ich hab auch schon verschiedene Pins versucht, aber leider ebenfalls ohne Erfolg.

Mit welchem Codeschnipsel hast du es denn versucht?

empty88, 2022-02-07:
@interose 

Dein while im setup verhindert, dass das setup beendet werden kann. Das hat da nichts zu suchen. Erst wenn das setup durchgelaufen ist, wird loop() ausgeführt.
`    while (1) {
      delay (1000);
    }`

interose, 2022-02-07:
@empty88 

Ahhhhh, okay das funktioniert schon mal - vielen Dank!

Hast du dann auch die [VitoWiFi](https://github.com/bertmelis/VitoWifi) benutzt? Und falls ja mit welchen Pins? 
Leider verlangt die setup Methode hier ein HardwareSerial Interface.

interose, 2022-02-07:
@empty88 okay - ja bei mir funktioniert es (bzw. sollte es) in die andere Richtung. Ich habe schon zwei ESP's (Stromzähler & Gaszähler) die per WiFi die Werte an ein REST Interface schicken und jetzt wollte ich die Vitodens ebenfalls noch "anschließen".

Ich fürchte ich bin jetzt auch mit den Dioden durcheinander gekommen -> siehe [hier](https://github.com/openv/openv/discussions/536#discussioncomment-2127564)

Vielen Dank auf jeden Fall mal!