openv/openv #495: USB Duo Sniffer Block Extension

vich-667, 2020-08-25:
I stated to extend the scematic of the USB Duo Sniffer

#494 # General Project Idea

- use the USB Duo Sniffer curcuit to have the Heater still online and grab some tempreate values from the commu

nication
- block the running communication and use the "PC" to change the temperature or ask for 1x WW

### Schematic
Basicly I used the existing [schematic ](https://github.com/openv/openv/wiki/Bauanleitung-USB_Duo-Sniffer) from @GolterU and extend it to block the communicaton of one port with the dtr signal from the other port.
[image]
At the moment i tried to simulate it with LTSpice and a first prototype wat tested with the heater. If you are interested in give me a hint what you think about it. Did I need to change or improve something?

### Software
Not done yet, only some basic test skripts. Generally I plan to write the appication in python.

drummer3333, 2020-11-22:
I had a simliar idea, but using the included OPTO 1 connected to a PI 4 and the USB OTG in slave mode connected to the vitoconnect 100

m1k3f15h, 2023-02-16:
Hy @vich-667,

I like the Idea :)
i already ordered the parts for a DUO-Sniffer..
First step will be to "only listen" to the answers from the heating..
whitch software do you use?

Mike

vich-667, 2023-02-19:
Hi!
I have written my own SW using Python. I hope I will find some time during the next days, to finally test my stuff. Than I will publish it here on github.

m1k3f15h, 2023-02-20:
Hi,
yeah that would be great :)
I hope i find some time to build my Duo-Sniffer :-D

vich-667, 2023-04-16:
Here you can find all details about my project: https://github.com/vich-667/OptolinkDuoBlock