openv/openv #497: Not receiving "05" response to "04" sync message

eddyg, 2020-10-23:
I built a Raspberry Pi opto interface following this [excellent blog post](https://projects.webvoss.de/2017/11/05/interfacing-vitovalor-300-p-with-a-raspberry-pi/#Building_the_OptoLink_Connector) that was referenced in the [Bauanleitung Raspberry Pi](https://github.com/openv/openv/wiki/Bauanleitung-RaspberryPi) wiki entry.

I believe the optical UART interface components are functioning properly: I can run `screen /dev/ttyAMA0 4800` on the Pi, point the infrared emitter at the infrared receiver, and when they are aimed at each other properly, I can see characters I type in `screen` echoed back to me; when they aren't pointed at each other, I don't.

For reference, I used an [Osram IRL 81A](https://www.osram.com/ecat/Radial%20Sidelooker%20IRL%2081%20A/us/en/class_pim_web_catalog_103489/global/prd_pim_device_2219704/) as the IR emitter (as suggested in the above blog entry since the SFH487-2 is no longer available) and an [Osram SFH 309 FA](https://www.osram.com/ecat/Radial%20T1%20SFH%20309%20FA/com/en/class_pim_web_catalog_103489/global/prd_pim_device_2219658/) as the IR receiver.

I was able to `git clone` `vcontrold` and compile it on the Pi without issue. I then copied the XML files from the (kw)[https://github.com/openv/vcontrold/tree/master/xml/kw] directory to `/etc/vcontrold` and modified `/etc/vcontrold/vcontrold.xml` with:
```
        <tty>/dev/ttyAMA0</tty>
```
and set my device:
```
      <device ID="2098"/>
```
to match the Vitotronic 200 KW2 I want to monitor.

`vcontrold` starts up fine, but when I connect and send a command, it _never_ sees the **05** response to the **04** sync byte:

```
vctrld>debug on
vctrld>getTempA
DEBUG:Fri Oct 23 11:18:14 2020 : Command: getTempA
DEBUG:Fri Oct 23 11:18:14 2020 : >SENT: 04
DEBUG:Fri Oct 23 11:18:14 2020 : >FRAMER: no preset result
DEBUG:Fri Oct 23 11:18:14 2020 : Waiting for 05
DEBUG:Fri Oct 23 11:18:19 2020 : read timeout
DEBUG:Fri Oct 23 11:18:19 2020 : Error in wait, terminating
DEBUG:Fri Oct 23 11:18:19 2020 : Error executing getTempA
ERR: read timeout
Error in wait, terminating
Error executing getTempA
```

Given that I can point the IR emitter/receiver at each other and get characters echoed (which indicates to me the hardware it wired properly and working), what else can I try?

enzingerm, 2020-12-02:
You could verify whether your serial port receives the periodically sent `0x05` sync byte from the heating control at all.

eddyg, 2021-01-19:
Thanks @enzingerm for the hints about the sync and ACK bytes. It was helpful to learn that the `0x05` byte gets sent periodically. 

My probem ended up being related to the IR LEDs I was using.

I switched to using an [Osram SFH 300 FA-3/4](https://www.osram.com/ecat/Radial%20T1%203-4%20SFH%20300%20FA/com/en/class_pim_web_catalog_103489/prd_pim_device_2219652/) as the receiver and [Osram SFH 4556](https://www.osram.com/ecat/Radial%20T1%203-4%20SFH%204556/com/en/class_pim_web_catalog_103489/prd_pim_device_2219779/) as the emitter and all is well!

closed 2021-01-19 by eddyg