openv/openv #498: Error returned when reading any address

sebastianw-dev, 2020-10-26:
Hello,

I've setup a Raspberry Pi 4 + the Optolink adapter of a "Vitoconnect 100" connected to the USB. I have a Vitocal 200-A with a Vitotronic 200 WO1C. The device ID seems to be 204C.

To test it I tried to read the outside temperature using the following command, using the P300 protocol:

`vclient -h 127.0.01 -p 3002 -m -c getTempA`

This is the extract of the log file recorded:

```
[1204] Mon Oct 26 23:38:19 2020 : Command: getTempA
[1204] Mon Oct 26 23:38:19 2020 : Process 1204 tries to aquire lock
[1204] Mon Oct 26 23:38:19 2020 : Process 1204 got lock
[1204] Mon Oct 26 23:38:19 2020 : >FRAMER: open device /dev/ttyUSB0 ProtocolID 41
[1204] Mon Oct 26 23:38:19 2020 : Configuring serial interface /dev/ttyUSB0
[1204] Mon Oct 26 23:38:19 2020 : >SENT: 04
[1204] Mon Oct 26 23:38:21 2020 : <RECV: len=1 05 (2000.0 ms)
[1204] Mon Oct 26 23:38:21 2020 : <RECV: received 05
[1204] Mon Oct 26 23:38:21 2020 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_LINK_STATUS(05) (was FFFF)
[1204] Mon Oct 26 23:38:21 2020 : >FRAMER: closed
[1204] Mon Oct 26 23:38:21 2020 : >SENT: 16
[1204] Mon Oct 26 23:38:21 2020 : >SENT: 00
[1204] Mon Oct 26 23:38:21 2020 : >SENT: 00
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 06 (10.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: received 06
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was FE05)
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: opened
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 41
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 05
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 00
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 01
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 01
[1204] Mon Oct 26 23:38:22 2020 : >SENT: C1
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 02
[1204] Mon Oct 26 23:38:22 2020 : >SENT: CA
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 06 (30.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: received 06
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: framer_set_actaddr framer_current_addr = C101 (was FFFF)
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: Command send
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: no preset result
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 41 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 06 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: received 41 06
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 03 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 01 (10.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 01 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 C1 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 01 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 04 (10.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: len=1 D1 (0.0 ms)
[1204] Mon Oct 26 23:38:22 2020 : <RECV: received 03 01 01 C1 01 04 D1
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was C101)
[1204] Mon Oct 26 23:38:22 2020 : >FRAMER: ERROR address 01C1 code 4
[1204] Mon Oct 26 23:38:22 2020 : Error in recv, terminating
[1204] Mon Oct 26 23:38:22 2020 : Error executing getTempA
[1204] Mon Oct 26 23:38:22 2020 : Command: quit
[1204] Mon Oct 26 23:38:22 2020 : >SENT: 04
[1204] Mon Oct 26 23:38:24 2020 : <RECV: len=1 05 (2000.0 ms)
[1204] Mon Oct 26 23:38:24 2020 : <RECV: received 05
[1204] Mon Oct 26 23:38:24 2020 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_LINK_STATUS(05) (was FFFF)
[1204] Mon Oct 26 23:38:24 2020 : >FRAMER: closed
[1204] Mon Oct 26 23:38:24 2020 : Process 1204 released lock
[1204] Mon Oct 26 23:38:24 2020 : Closed connection (fd:1)
[1204] Mon Oct 26 23:38:24 2020 : Child process with PID 1204 terminated
```

As far as I understood the protocol it seems that after sending the request to read address `01C1` it starts to receive the good answer. However, after _41 06_ the 3rd byte is a _03_, indicating an error.

I have no idea what's wrong or if I'm completely wrong with and the issue is something else. Did anyone of you had a similar issue or has an idea what could be the root cause? 
I'm using the vcontrold.xml file I downloaded together with the code. I just modified the device ID. Is that an issue? Any adaptation of the P300 protocol needed for my device?

Thank you for your support!

Sebastian

sebastianw-dev, 2020-10-28:
It turned out that simply the address for the outside temperature was wrong. I incorrectly assumed that for all WO1C control the addresses will be similar, but that seems to be not the case. 

Can any provide the list of datapoints for the type 204C control?

Thanks, Sebastian

simat-git, 2021-12-13:
Hi Sebastian

Did you ever manage to get vcontrol talking to your Vitocal 200-A ?

If so, please can i have a copy of your XML files as i  also have a Vitocal 200-A with a Vitotronic 200 WO1C

Best regards

Simon.

sebastianw-dev, 2021-12-27:
Hi Simon,

these are the files I'm using ... of course I did not use or test all commands/ addresses. I propose you try getTempA first, because that should be working at least.
[config.zip](https://github.com/openv/openv/files/7781745/config.zip)