openv/openv #499: Desperate for E3 HMI advice ? How to connect ?

Zixim, 2020-10-27:
Please, it there _anyone_ with a generation E3 HMI (Heating Manager Interface) active in this github project ?  
  
The E3 is a touchscreen-only front panel, there is **no V-shaped connector** (or any other) visible. There is **no sliding door/panel** either.

### How to connect Optolink to this Generation E3 HMI ??? 

[image]

speters, 2022-02-01:
Markus Fritze (@sarnau) wrote about the WIFI communication in his [docs or InsideViessmann](https://github.com/sarnau/InsideViessmannVitosoft/blob/main/VitosoftCommunication.md).
If it is as simple at it sounds, it might be possible to redirect TCP traffic from port 45317 on the Viessmann device to a pseudo-serial device on your PC/SBC/Raspberry using e.g. [socat](https://www.acmesystems.it/socat).

sarnau, 2022-02-01:
I've indeed done a "router" to convert Viessmann serial to their Wifi protocol. So their software detects the system and can read it. It is way too slow, so I never really used it beyond that. Also: their demo version only goes so far.