openv/openv #502: VplusHO1 - 20C8  Adressen

jenskaschub, 2020-11-12:
Hallo,
im Großen und Ganzen läuft mein openv auf einem raspi3. Aber: ich suche mich seit Tagen Wund bei folgenden Adressen:
es ist die Vitoladens 300-W / 20C8

- getBrennerStatus
- getBrennerStufe
- getMischerM2

- getLeistungIst: da bekomme ich wenigesten eine Ausgabe jedoch 193% bei Stufe 1 und 228% bei Stufe 2. Müüste das nicht 65 und 100% sein?

Danke
Jens

dima-357, 2020-11-21:
Please tell me the request address V100
boiler pump in operation or not /
It is not in the list of addresses of Viessmann-adresses

jenskaschub, 2020-11-21:
if you mean the intern pump, on mine it's 7660

SpatialJ, 2021-02-07:
@jenskaschub Bist Du hier weiter gekommen? Ich bin auch auf der Suche nach einer möglichst vollständigen vito.xml für die VplusHO1

jenskaschub, 2021-02-07:
Ne, leider nicht. Aber ich einige wichtige mit denen ich effektif meine Heizung überwachen und auch steuern kann. Die oben erwähnten fehlen leider noch, bzw. liefern noch falsche Werte. 
Die Daten logge ich übrigends über ioBroker in influx und lass sie dann über Grafana darstellen.