openv/openv #504: OptoPi low volume series / Kleinserie

Ion3, 2020-11-22:
[English below]

Dieses Konzept scheint sehr passend zu sein für die neueren Heizungen mit dem kleinen Schacht. Ist eine Kleinserie geplant? 

Gerne kann die Platine auch noch ein kleines Stück kleiner sein, so dass ein schickes schwarzes 3D-gedrucktes Gehäuse drum rum passt. Ich wäre gerne bereit die Gehäuse dazu beizusteuern. 

Hat jemand den OptoPi in Verbindung mit openHAB am Laufen? Mir ist hier noch nicht ganz klar, was alles an Software hierfür benötigt wird. (z.B. brauch es zwingend einen Raspi oder kann auch ein Adruino mit LAN das [Binding](https://www.openhab.org/addons/bindings/vitotronic/) füttern?).

[ENGLISH]

This concept seems to be very suitable for the newer heaters with the small compartment. Is a small series planned? 

The board can be a little bit smaller, so that a nice black 3D-printed case fits around it. I would be happy to contribute the case.  

Does anyone have the OptoPi running in conjunction with openHAB? I'm not quite sure what kind of software is needed here. (e.g. does it compulsively need a Raspi or will an Adruino with LAN also feed the [binding](https://www.openhab.org/addons/bindings/vitotronic/)?).

SergejKrut, 2021-04-06:
Hallo,
wo und wie kann ich Platine OptoPi bestellen?
https://github.com/openv/openv/wiki/Bauanleitung-OptoPi

dankeschön

STB3, 2021-04-17:
Ihr könnt mal hier schauen, ich hebe eine Kleinserie für USB aufgelegt: [https://github.com/openv/openv/wiki/Bauanleitung-USB](url). Aktuell sind noch Teile verfügbar.