openv/openv #506: Keine Ölstand abfrage.

insiderla, 2020-12-01:
Hallo,
ich möchte  von meiner Heizung den Ölstand auslesen und habe nun folgende Zeilen der Vito.xml und Vcontrold.xml hinzugefügt. Leider bekomm ich immer den selben Wert bei meiner Abfrage im Telnet

vctrld>getOel
0.000000 Liter
oder
vctrld>getOel
84215,046875 Liter

die Werte verändern sich seit Tagen nicht. Was mach ich falsch?

Heizung: V200 KW2

VITO.xml:
        <command name='getOel' protocmd='getaddr'>
             <addr>7574</addr>
             <len>4</len>
             <unit>LI</unit>
             <description>Ermittle Oelstand</description>
             <device ID="2053"/>
        </command>
        <command name='setOel' protocmd='setaddr'>
             <addr>7574</addr>
             <len>4</len>
             <unit>LI</unit>
             <description>Ermittle Oelstand</description>
             <device ID="2053"/>
        </command>
        <command name='getLeistung' protocmd='getaddr'>
             <addr>55E3</addr>
             <len>1</len>
             <unit>PRN</unit>
             <description>Aktuelle Leistung</description>
             <device ID="2053"/>

Vcontrold.xml
        <unit name='Liter'>
           <abbrev>LI</abbrev>
           <calc get='V/1000' set='V*1000'/>
           <type>int</type>
           <entity>Liter</entity>
        </unit>

Vielen Dank für eure Hilfe

insiderla, 2020-12-01:
`VITO.xml:
        <command name='getOel' protocmd='getaddr'>
             <addr>7574</addr>
             <len>4</len>
             <unit>LI</unit>
             <description>Ermittle Oelstand</description>
             <device ID="2053"/>
        </command>
        <command name='setOel' protocmd='setaddr'>
             <addr>7574</addr>
             <len>4</len>
             <unit>LI</unit>
             <description>Ermittle Oelstand</description>
             <device ID="2053"/>
        </command>
        <command name='getLeistung' protocmd='getaddr'>
             <addr>55E3</addr>
             <len>1</len>
             <unit>PRN</unit>
             <description>Aktuelle Leistung</description>
             <device ID="2053"/>

Vcontrold.xml
        <unit name='Liter'>
           <abbrev>LI</abbrev>
           <calc get='V/1000' set='V*1000'/>
           <type>int</type>
           <entity>Liter</entity>
        </unit>
`

enzingerm, 2020-12-02:
Das Problem liegt an der Kommunikation mit der Heizung. Die Heizung liefert 4 mal das Byte `0x05` zurueck (was eigentlich zum Synchronisieren verwendet wird), der Client interpretiert es jedoch als Zahl (welche dann nach der Division durch 1000 den dir angezeigten Wert ergibt). Kannst du andere Werte auslesen?

insiderla, 2020-12-16:
ja andere Werte wie getTempA liefern die richtigen Werte.
Kann meine Konfiguration falsch sein?

VG