openv/openv #510: Kompatibilität mit Vitodens 200-W Typ B2HF?

marraufu, 2020-12-16:
Hallo zusammen,

ich bin dabei mir eine neue Heizungsanlage zuzulegen und habe eine Angebot über die Vitodens 200-W Typ B2HF vorliegen. 
Meine Frage ist nun, ob ich die Steuerung mit openv auf dem raspberry auslesen kann?

Viele Grüße
Marko

OrionVIII, 2020-12-26:
Meine ist eine Vitodens 200-W B2HB-35. Da funktioniert es ganz gut (Die Kommunikation läuft über einen Node3MCU, der über MQTT mit dem Homeassistant verbunden ist: [https://github.com/OrionVIII/OpenV_NodeMCU](url) ). Habe aber leider keine Ahnung, wo der Unterschied zwischen B2HB und B2HF liegt.
VG Norbert

HelmutFi, 2021-12-30:
Das würde mich auch interessieren. Hab auch eine Vitodens 200-W B2HF-19 im Keller hängen.