openv/openv #517: Dokumentation für die XML Files

sarnau, 2021-07-23:
Hallo,

Ich habe die XML Files von Vitosoft mal dokumentiert. Damit kann man für seine individuelle Viesmannn Anlage an alle Parameter rankommen. Zudem habe ich einen kleinen Optolink2MQTT Server für Raspberry Pi geschrieben und dabei die Details der VS1/VS2 Kommunikation dokumentiert.

[https://github.com/sarnau/InsideViessmannVitosoft](https://github.com/sarnau/InsideViessmannVitosoft)

speters, 2021-12-17:
Vielen Dank nicht nur für die tolle Zusammenfassung der bisherigen Erkenntnisse, sondern vor allem dafür, dass du da noch ein ganzes Stück weiter eingetaucht bist!

aliexpress987, 2022-03-03:
Das Tool ist wirklich super! Tolle Leistung! Leider nützt es mir konkret nicht viel, da ich als Privatmensch keinen Zugriff auf eine Vitosoft habe, demzufolge liegt das data Verzeichnis mit den xml's nicht vor.
Ich bräuchte eigentlich nur das Ausgabeergebnis des Tools, eventuell reduziert auf den Typ 2053 GWG_VBEM (Vitodens 200 WB2), damit ich meine vcontrol commands endlich mal vervollständigen könnte.

sarnau, 2022-03-03:
Doch, jeder kann Vitosoft direkt downloaden und somit an die XML Daten gelangen. Siehe [https://github.com/sarnau/InsideViessmannVitosoft/blob/main/VitosoftSoftware.md](https://github.com/sarnau/InsideViessmannVitosoft/blob/main/VitosoftSoftware.md)

aliexpress987, 2022-03-03:
Schande über mich, ich hatte die Links im README.md nicht angeklickt ;-) Meinen herzlichsten Dank für den schnellen Hinweis. Dann muss ich mal ne Windows Kiste booten und schauen ....

aliexpress987, 2022-03-03:
So, Stunden später :-). Übrigens bekommt man über den Link bei Version 8.0.5.0 inzwischen die 8.0.6.2, also kein nachträgliches Update erforderlich.

Erfolge: PrintDatapoints.py läuft fehlerfrei durch.

Probleme: Alle anderen Skripte laufen nicht durch.
Beispiel: python3 PrintEventTypes.py beginnt loszulaufen und printet einiges am Bildschirm, bis 
```
0xCD18 Virtual_READ *1 50-300:5 Int/2 Byte:0/2 --- (0C) Flow rate level 3
0xCD74 Virtual_READ *1 Int/2 Byte:0/2 --- (3A) Bathroom switch
0xCD76 Virtual_READ *1 0-1440:1 Int/2 Byte:0/2 --- (3B) Duration intensive ventilation on bathroom switch
0xCD78 Virtual_READ *1 Int/2 Byte:0/2 --- (3C) Differential pressure switch, combustion equipment
Traceback (most recent call last):
  File "PrintEventTypes.py", line 144, in <module>
    parse_ecnEventType();
  File "PrintEventTypes.py", line 140, in parse_ecnEventType
    print('%6s %s %s %s --- %s' % (event['Address'],event['FCRead'],border,parameter,IDstr))
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2013' in position 72: ordinal not in range(256)
```
Ein lokales Problem mit der Codetabelle?

2. Problem - eigentlich das interessantere:
python3 PrintEventsForDatapoint.py  -  stürzt ohne weitere Ausgabe ab
``
Traceback (most recent call last):
  File "PrintEventsForDatapoint.py", line 478, in <module>
    parse_DPDefinitions('GWG_VBEM_00')
  File "PrintEventsForDatapoint.py", line 139, in parse_DPDefinitions
    root = etree.fromstring(re.sub(' xmlns="[^"]+"', '', open(DATAPATH + "DPDefinitions.xml",'r').read()))
  File "/usr/lib/python3.7/xml/etree/ElementTree.py", line 1315, in XML
    parser.feed(text)
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 1, column 1
``
Was nun? Python 3.7 zu alt?

sarnau, 2022-03-03:
Sieht nach einen Encoding Problem aus: UTF-8 vs. latin-1. Keine Ahnung, wie das unter Windows verhält. Ich habe einen Mac bzw. Linux.

aliexpress987, 2022-03-04:
Also ich verwende Linux. Problem 2 hat sich komischerweise von selbst gelöst und PrintEventsForDatapoint läuft fehlerfrei durch. Ich vermute, dass vorher Speichermangel die Ursache war.

Problem 1 - ist eher ein Fehler in den xml Daten, wo Viessmann einen Windows-spezifischen Character verwendet hat:
`The unicode character u'\02013' is the "en dash". It is contained in the Windows-1252 (cp1252) character set (with the encoding x96), but not in the Latin-1 (iso-8859-1) character set. `

sarnau, 2022-03-04:
Speicher, ja das XML ist _sehr_ groß. Hmm, eventuell fiel das bisher nicht auf, weil der String mit dem en dash nicht bei jeder Anlage benutzt wird?

aliexpress987, 2022-03-04:
Also das trat hier mit PrintEventTypes auf - also eigentlich generisch. Hab's eben nochmal laufen lassen - heute läufts fehlerfrei durch. Habe wirklich nichts am Environment geändert, muss ebenfalls ein Folgefehler eines Speichermangels gewesen sein.

Also im Moment, alles gut, alles läuft :-)

ultrazauberer, 2022-03-07:
> Hallo,
>
> Ich habe die XML Files von Vitosoft mal dokumentiert. Damit kann man für seine individuelle Viesmannn Anlage an alle Parameter rankommen. Zudem habe ich einen kleinen Optolink2MQTT Server [...]

Danke für deine Arbeit! Verstehe ich das richtig, dass man mit deinem Python Tool und der Viessmann Software alle Datenpunkte/Adressen der Anlage auslesen kann? Quasi auch das Menü der Energiebilanzen (Service Level 1) von der WO1C Vitotronic 200?

Markus030789, 2022-07-20:
Hallo zusammen,
leider habe ich auch das hier Problem 2:

`PS Z:\> & C:/Users/marku/AppData/Local/Programs/Python/Python310/python.exe c:/Users/marku/Desktop/InsideViessmannVitosoft-main/PrintEventsForDatapoint.py
Traceback (most recent call last):
  File "c:\Users\marku\Desktop\InsideViessmannVitosoft-main\PrintEventsForDatapoint.py", line 479, in <module>
    parse_DPDefinitions('V200KO2B')
  File "c:\Users\marku\Desktop\InsideViessmannVitosoft-main\PrintEventsForDatapoint.py", line 140, in parse_DPDefinitions
    root = etree.fromstring(re.sub(' xmlns="[^"]+"', '', open(DATAPATH + "DPDefinitions.xml",'r').read()))
  File "C:\Users\marku\AppData\Local\Programs\Python\Python310\lib\xml\etree\ElementTree.py", line 1344, in XML
    parser.feed(text)
xml.etree.ElementTree.ParseError: not well-formed (invalid token): line 1, column 1`

Ich bin mit VSCode auf Windows 10 unterwegs, vermutlich ist die DPDefinitions.XML einfach zu groß. Kann mir wohl jemand die Ergebnisse aus der `PrintEventsForDatapoint.py` für meine Anlage `V200KO2B` zukommen lassen?

Viele Grüße
Markus

BenNiklas, 2023-11-29:
Hallo Allerseits,

konnte das Encoding-Problem auf meinem System mit folgender Änderungen in Zeile 139 von "PrintEventsForDatapoint.py" beheben:

`root = etree.fromstring(re.sub(' xmlns="[^"]+"', '', open(DATAPATH + "DPDefinitions.xml", 'r', encoding='utf-8').read()))`

Beste Grüße,
Ben