openv/openv #525: Ferienbetreib setzen

swhof123, 2021-12-15:
Erstmal vielen Dank für all die Infos hier. 
Ich möchte gerne bei meiner Vitodens200 mit Regelung Vitotronic200 Typ HO1b den Ferienbetrieb über mein FHEM starten und beenden können. 
Habe dazu vito.xml so erweitert:
```   <command name='getVitoBetriebFerien' protocmd='getaddr'>
                <addr>2535</addr>
                <len>1</len>
                <unit>BFB</unit>
                <description>Ferienbetrieb</description>
        </command>        
        <command name='getVitoFerienBeginn' protocmd='getaddr'>
                <addr>2309</addr>
                <len>8</len>
                <unit>TI</unit>
                <description>Abreisetag</description>
        </command>
        <command name='setVitoFerienBeginn' protocmd='setaddr'>
                <addr>2309</addr>
                <len>8</len>
                <unit>TI</unit>
                <description>Setze Abreisetag</description>
        </command>
        <command name='getVitoFerienEnde' protocmd='getaddr'>
                <addr>2311</addr>
                <len>8</len>
                <unit>TI</unit>
                <description>Rückreisetag</description>
        </command>
        <command name='setVitoFerienEnde' protocmd='setaddr'>
                <addr>2311</addr>
                <len>8</len>
                <unit>TI</unit>
                <description>Setze Rückreisetag</description>
        </command>
```

Was funktioniert sind die getBefehle.
Was nicht klappt ist das Setzen von Start und Endzeit

```
vctrld>setVitoFerienBeginn 20 21 12 16 04 00 00 00
DEBUG:Wed Dec 15 11:50:08 2021 : Befehl: setVitoFerienBeginn 20 21 12 16 04 00 00 00
DEBUG:Wed Dec 15 11:50:08 2021 : Setzen von explizieter Zeit noch nicht unterstuetzt
DEBUG:Wed Dec 15 11:50:08 2021 : Fehler Unit Wandlung:
DEBUG:Wed Dec 15 11:50:08 2021 : Fehler beim ausfuehren von setVitoFerienBeginn 20 21 12 16 04 00 00 00
ERR: Setzen von explizieter Zeit noch nicht unterstuetzt
Fehler Unit Wandlung:
Fehler beim ausfuehren von setVitoFerienBeginn 20 21 12 16 04 00 00 00
```
Alternativ ohne gesetzet Zeit nimmt er die Systemzeit:
```
vctrld>setVitoFerienBeginn
DEBUG:Wed Dec 15 11:47:27 2021 : Befehl: setVitoFerienBeginn
DEBUG:Wed Dec 15 11:47:27 2021 : aktuelle Sys.Zeit 20 21 12 15 03 11 47 27
DEBUG:Wed Dec 15 11:47:27 2021 : >SEND: 04
DEBUG:Wed Dec 15 11:47:27 2021 : >FRAMER: no preset result
DEBUG:Wed Dec 15 11:47:27 2021 : Warte auf 05
DEBUG:Wed Dec 15 11:47:29 2021 : <RECV: 05 (1890.0 ms)
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: 01
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: F4
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: 23
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: 09
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: 08
DEBUG:Wed Dec 15 11:47:29 2021 : >SEND: 20
DEBUG:Wed Dec 15 11:47:29 2021 : >FRAMER: no preset result
DEBUG:Wed Dec 15 11:47:31 2021 : <RECV: len=1 05 (2290.0 ms)
DEBUG:Wed Dec 15 11:47:31 2021 : <RECV: received 05
DEBUG:Wed Dec 15 11:47:31 2021 : 05 -> SYNC (NOT OK)
DEBUG:Wed Dec 15 11:47:31 2021 : SYNC (NOT OK)
```

Was mich stutzig macht, ist dass die Suche die Fehlermeldung "Setzen von explizieter Zeit noch nicht unterstuetzt
", die ja auf Grund ihres Rechtschreibfehlers unique ist, nicht im vcontrold Sourcecode findet. 
Leider verstehe ich nicht genug vom Programmieren um dem auf den Grund zu gehen. Wer kann helfen?

closed 2021-12-15 by swhof123

daniele2209, 2023-10-17:
Hallo @swhof123 
hast Du ein Beispiel für mich wie Du die Ferienzeit mittels vControld setzt?
Mich würde der XML-Ausschnitt und der entsprechende vcontrol-Befehl interessieren.
Also welche Parameter du genau übergibst.
Danke schon mal
Gruss Daniel

swhof123, 2023-10-18:
Hi, 
xml steht oben. 
Ich habe es schlussendlich nie in FHEM integriert, da ich es auch nicht so oft brauche. 
Greife daher per telnet auf den Client zu:  z.B.
`telnet localhost 3002`
Befehle sind: 
`getVitoFerienBeginn: Abreisetag
setVitoFerienBeginn: Setze Abreisetag Format YYYY-MM-DDT23:59:59+0100
getVitoFerienEnde: Rückreisetag
setVitoFerienEnde: Setze Rückreisetag Format YYYY-MM-DDT00:00:00+0100`
Am Besten vor Abreise setzten und remote nur ändern, wenn sich die Pläne geändert haben.
Hilft dir das weiter?