openv/openv #530: Sammelstörung 0x0A82 liefert immer den Wert 0

vito-user, 2022-01-09:
Hallo,
seit einiger Zeit läuft das Auslesen der Daten problemlos. Jetzt ist seit langem mal eine Störung am Gerät aufgetreten und ich musste leider feststellen, dass die Abfrage der Adresse 0A82 immer noch 0 (keine Störung) liefert. Ich habe eine Vitodens 200-W mit einer Vitotronic 200 (WO1B) Steuerung und folgende Einträge in der VITO.XML:
`<vito>
	<devices>
		<device ID="20CB" name="VScotH01" protocol="P300" />
	</devices>
....
		
		<command name='getStatusStoerung' protocmd='getaddr'>
			<addr>0A82</addr>
			<len>1</len>
			<unit>ES</unit>
			<description>Status Störung</description>
		</command>
`
Ich habe schon diverse andere Adressen ausprobiert, aber ohne Erfolg. Die Abfrage der Fehlerhistorie funktioniert problemlos. Nur bekommt man damit eben nicht den aktuellen Zustand der Heizung raus.
Hat jemand eine Idee, was ich noch probieren könnte?

Viele Grüße, Uwe

JanMali, 2022-01-19:
Hallo @vito-user,

das Problem hatte ich auch. Probier's mal mit Adresse 7590. Wenn der Wert ≠ 00 ist, steht eine Störung an.

Gruß Jan

vito-user, 2022-01-19:
Hallo Jan,
danke für den Tipp. Die Adresse wurde akzeptiert (andere Adressversuche scheiterten schon beim Abruf innerhalb von vcontrold). Jetzt muss ich nur mal bei Gelegenheit eine Störung an der Heizung provozieren, um zu sehen, ob's klappt.
Viele Grüße, Uwe

Markus030789, 2022-07-17:
Hallo zusammen,
bei mir leider der gleiche Effekt dass mit 0A82 trotz aktueller Störung "00" ausgegeben wird, ebenfalls die Anlage Vitotronic 200 (WO1B). Ich habe wie vorgeschlagen den Wert einmal auf 7590 geändert, jetzt erhalte ich "FF" - Das ist zwar ≠ 00, befürchte aber dass damit nicht "Störung" gemeint ist sondern dass der gesendete Wert einfach Fehlerhaft ist und nicht verarbeitet werden kann. Hier einmal vcontrol mit "debug on"

```
vctrld>debug on
vctrld>getVitoStatusStoerung
DEBUG:Sun Jul 17 18:57:44 2022 : Command: getVitoStatusStoerung
DEBUG:Sun Jul 18:57:44 2022 : >SENT: 04
DEBUG:Sun Jul 17 18:57:44 2022 : >FRAMER: no preset result
DEBUG:Sun Jul 17 18:57:44 2022 : Waiting for 05
DEBUG:Sun Jul 17 18:57:46 2022 : <RECV: 05 (1880.0 ms)
DEBUG:Sun Jul 17 18:57:46 2022 : >SENT: 01
DEBUG:Sun Jul 17 18:57:46 2022 : >SENT: 
DEBUG:Sun Jul 17 18:57:46 2022 : >SENT: 75
DEBUG:Sun Jul 17 18:57:46 2022 : >SENT: 90
DEBUG:Sun Jul 17 18:57:46 2022 : >SENT: 01
DEBUG:Sun Jul 17 18:57:46 2022 : >FRAMER: no preset result
DEBUG:Sun Jul 17 18:57:46 2022 : <RECV: len=1 FF (10.0 ms)
DEBUG:Sun Jul 17 18:57:46 2022 : <RECV: received FF
DEBUG:Sun Jul 17 18:57:46 2022 : FF -> ?
DEBUG:Sun Jul 17 18:57:46 2022 : ?
```

Auszug aus meiner vito.xml
```
		<!-- STOERUNG -->
		<command name='getStatusStoerung' protocmd='getaddr'>
			<addr>7590</addr>
			<len>1</len>
			<unit>ES</unit>
			<description>Status Störung</description>
		</command>
```

Ich warte noch einmal ab bis die Störung hoffentlich bald behoben ist ob ich dann wie erwartet eine "00" zurückbekomme, vermute aber dass ich auf dem falschen Weg bin. Welche Adressen würden noch in Frage kommen und hat es bei dir mit dem Wert 7590 geklappt, @vito-user ?

Viele Grüße
Markus

vito-user, 2022-07-17:
Hallo Markus,
bis jetzt war meine Heizung noch nicht wieder in Störung, und eine Störung provozieren wollte ich auch nicht. Steht bei mir also noch auf der Beobachtungsliste.
Viele Grüße, Uwe

Markus030789, 2022-07-19:
Hallo zusammen,
unser Klempner war heute da um die Störung zu beseitigen, die Anlage scheint wieder zu laufen. Wie befürchtet hilft mir der vorgeschlagene Wert 7590 nicht, trotz Entstörung erhalte ich ein "FF" bzw. gemapptes "?". Auch der Wert 0A82 liefert wieder 0, genau wie während der Störung.

@vito-user Hast du denn aktuell den Wert 7590 bei dir drin? Du hattest geschrieben dass der Wert bei dir angenommen wird. Was erhältst du denn aktuell von vcontrold zurück?

Viele Grüße
Markus

JanMali, 2022-07-20:
Hallo Markus,
was genau hast du denn für eine Heizung? Ich hatte die 7590 aus den XML-Dateien der Vitosoft. Vielleicht findet man zu deinem speziellen Typ dort auch einen Wert, der funktioniert. 
Gruß Jan

Markus030789, 2022-07-20:
Hallo @JanMali ,
ich habe die
```
Vitotronic 200 Typ KO1B
V200KO1B (ID=209F) (Protocol=P300)
```
Ich habe mir das Vitosoft Tool eben geladen und durchforste die XMLs, bin aber noch ganz am Anfang und finde mich erst einmal zurecht. Vielleicht hast du die passende Adresse ja direkt zur Hand?

Eine Frage noch: Sind die Adressen Case Sensitive? Manchmal finde ich z.B. 0A82 oder 0a82 - Macht das einen Unterschied?

Gruß
Markus