openv/openv #539: Vitotronic 200 -VscotH01 - ViessData 2.0

Michel-V1264, 2022-03-16:
**Subject: Viess-Data 2.0 - Vitotronic 200 (VscotHO1)**

Hello,
I would first like to say that I am impressed by the tremendous work that has been done by the Open-V community and the level of programming skill of the contributors.
For my part, after installation on a Windows PC, I note that the Viess-Data 2.0 software communicates very well with a Vitotronic 200 regulation by means of an Optolink. The Serial Port Monitor software shows continuous communication without any error message. 

[image]

However, in the Viess-Data\bin\Release directory, no data is saved! Therefore, I have no data displayed in the "Graph" tab. Could anyone help me to solve this problem? 
Thank you in advance !!

GolterU, 2022-03-18:
Hi,
my system also works with the Vitotronic 200 (VscotHO1). I tested Viess-Data as an evaluation program, but wasn't really satisfied with it. It only seems to only support one heating circuit.
I therefore use [Viess-sion](https://github.com/openv/openv/wiki/Vies-sion) version 1.205 under Windows 10. Although the program has not been further developed for years, it fulfills the purpose of continuous data recording of all essential heating parameters. However, you have to adapt the "Viess-ion_CB.xml" file with trial and error.
Regards
Uwe

Michel-V1264, 2022-03-19:
Hi Uwe,
Thank you very much for your answer and for sharing your experience with Viess-Data.  After this, I'll also try the Vies-sion software. 
Best regards.
Michel