openv/openv #552: Get room temperature

corsac-s, 2022-08-18:
Hi,

I'm currently experimenting interactions with my Wiessmann heater (Vitodens 222-W with Vitotronic 200 (type H01B, and getDevType returns `20 CB 03 4F 00 00 01 3D`).

The heater has sensors for both external and room temperatures (displayed on the LCD screen), but it seems I can only query for external temperature (address `0x0800`), and I wasn't able to identify in `vito.xml` the address for the room temperature. Did I miss something obvious?

dirk-schulz, 2022-10-16:
On my Vitodens 200 I got the current room temperature on 0x0896.
Maybe give it a try.
I'm searching for set room temp (soll temperature) or partymode or burner status, but struggling to find the correct address.

corsac-s, 2022-10-17:
> On my Vitodens 200 I got the current room temperature on 0x0896. Maybe give it a try. I'm searching for set room temp (soll temperature) or partymode or burner status, but struggling to find the correct address.

Yes indeed, I never updated this issue but I found the reference to 0x0896 somewhere and it does seem to work for me. For now I didn't try to set the target temperature but there are references to:

- 0x2307: ``set reduced target room temperature``
- 0x2321: ``set standard target room temperature``
- 0x2330: ``set economy mode``
- 0x2331: ``set party mode``

dirk-schulz, 2022-10-17:
Many thanks, room temperature on 0x0896 is showing know the correct value. Great! 
Regarding the others (burner status, target room temp) I only want the read this value.
But e.g. I found reference to read it on 0x2307, 0x2306 or 0x2544, and I receive values like 537,6°C or -24xx.0°C, which might be to high as target room temp. ;-)
At least I would be highly interested in reading burner status (on/off and performance/level/Leistung) or partymode and energy mode.
Burner status I tried on 0x551e, Party mode on 0x2303 and energy on 0x2302, but on/off is not matching to the display information.
Any idea on that? Many thanks in advance.
Do you have other interesting data points, which you can suggest for the Vitodens 200?
BTW: currently I read the values with P300 protocol. Do you do it with your vitodens 200 the same?

dirk-schulz, 2022-10-17:
OMG! I'm bite straight into the edge of my table!
I just figured out, that viessmann temp datapoints are not all float values! The target room temp is only 1Byte and not 2Bytes like the other temp values. It's correctly documented in the address list in the wiki openv!
After changing the request from 2 to 1 Byte, I got now on 0x2306. 
I will check on partymode and so, if also here the data length is my issue.
Many thanks in the round to get me in the right direction.

corsac-s, 2022-10-18:
> Burner status I tried on 0x551e, Party mode on 0x2303 and energy on 0x2302, but on/off is not matching to the display information. Any idea on that?

I also had discrepancies between display information and vcontrold reported information, not sure why either.

> Do you have other interesting data points, which you can suggest for the Vitodens 200? BTW: currently I read the values with P300 protocol. Do you do it with your vitodens 200 the same?

I also use P300 protocol. Attached is my tentative `vito.xml`. German bits are totally untested, English ones are somehow tested.
[vito.txt](https://github.com/openv/openv/files/9807357/vito.txt)

dirk-schulz, 2022-10-18:
Many thanks for your vito.txt file. It works now for me perfectly. Many thanks for your support!