openv/openv #556: V200KW2 Vorlauftemperaturen soll/ist

GnomiBerlin, 2022-08-26:
Hallo, 
habe die Vitotronik V200KW2 
Details:
DATA TX: 1 C7 F8 4 4 
DATA RX: 20 
DATA RX: 98 
DATA RX: 0 
DATA RX: 1 

Weiss jemand, ob die Vorlauftemperaturen abfragbar sind?
Habe bisher nur sinnlose Werte bei möglichen Adressen erhalten.

Danke

DominikWild, 2022-11-15:
Sind an der Vitotronic Sensoren für die Temperaturen angeschlossen? Falls nicht, sind diese auch nicht abfragbar.

Funktioniert bei dir die Kesselsolltemperatur? Bei meiner V200KW1 bekomme ich immer nur -0,1°C ausgelesen,.

GnomiBerlin, 2022-11-17:
Ich habe erstmal zu meiner Frage oben jetzt die Kesseltemperatur als solche verwendet. Ich will ja die Kenngrößen, die die Heizung zum Regeln verwendet. Kessel-Soll-Temperatur und Kessel-Temperatur.
DPTemp getTempKist("gettempkist", "synchronize", 0x0802);            //Kesseltemperatur-Ist
DPTemp getTempKsoll("gettempksoll", "synchronize", 0x05502);            //Kesseltemperatur-Soll

Damit ist auch die Frage mit "ja" beantwortet und dies sind meine Adressen dafür.
Im Bild sind die Werte, wie diese sich verhalten. Als Erläuterung: ich habe nur morgens und abends Warmwasser und den Rest nur Heizung mit Nachtabsenkung. Orange = soll, blau = Ist
[image]

Ich hoffe, das hilft.

DominikWild, 2022-11-17:
Danke für die Rückmeldung. Unter der Adresse 0x5502 lese ich tatsächlich den Sollwert der Kesseltemperatur meiner V200KW1 aus, obwohl die Adresse in der Übersicht nur bei V200KO1B,KO2B steht. 

Hast mir sehr weitergeholfen!