openv/openv #561: Anleitung für 0.98.11 auf Raspberry funktioniert nicht mehr

andreasloe, 2022-10-11:
Wenn man die Installation vornimmt, so kann man in der Tat vcontrold auf dem Raspberry entpacken. Der Daemon läuft auch, die Angaben zu Startskripten sowie den Diensten sind leider veraltet. Der Dienst wird per se auch nicht installiert, dies muss man unter bullseye selbst vornehmen. Leider ist mir nicht klar, wie das genau geht.

Matten-Matten, 2023-01-18:
Hallo habe leider genau das selbe problem.
Ordner und Datei strukturen wie `~/openv/vcontrold-code/xml-32/xml/vcontrold.xml` existieren nach der Anleitung nicht.

Matten-Matten, 2023-01-19:
Mittlerweile hab ich den Dienst zum laufen gebracht.

Es gibt ein problem bei set befehl für den  betriebsmodus. Habe dafür aber auch schon eine Lösung gefunden.

Sobald ich mehr Zeit habe werde ich mich in Verbindung setzen und Erfahrungen austauschen.

Gruß
matten matten