openv/openv #562: Searching address list of Vitodens 200 WB2C

dirk-schulz, 2022-10-16:
Hello,
Happy to get the first values now from my Vitodens 200 WB2C.
But it seems, that the address tables for 200 here on the side does not fit to my model.
The burner status, partymode, energy saving mode and set room temperature are not showing correct values.
E.g. I'm trying to read set room temperature (soll temp) on Address 0x2306 or 0x2000 or others, but best value I receive is -2453.3 °C or 0?!
Or Partymode on 0x2303, but value does not change, if partymode is turned on.
Current room temp is on 0x0896. Outside temp, boiler temp and warm water temp is running well. 

I would appreciate, if someone with a Vitodens 200 WB2C could share his/her address list with me?
Or has someone a hint for me, how to read out a block of addresses via a esp32 (e.g. Only text from Adress 0x2000 to 0x3000)?

philippoo66, 2023-01-27:
moin Dirk!
Du musst Adresse F8 auslesen (am besten 4 Bytes) und ggf. noch F0 (einByte) und dann anhand der [Liste](https://github.com/philippoo66/ViessData21/blob/master/DataPoints.txt) den Steuerungstyp raussuchen und dann aus dieser [Sammlung](https://github.com/philippoo66/ViessData21/blob/master/DP%20Listen.zip) die passende DP Liste raussuchen. 
Wenn du noch andre Fragen hast (Werte, Umwandlung, Enums) sag Bescheid!
beste Grüsse!
Phil

dirk-schulz, 2023-01-28:
Guten Morgen Phil!
Super, danke Dir. Funktioniert.
Mein Fehler war: Ich hatte übersehen, dass Viessmann die Werte nicht alle als 4Bytes ablegt sondern auch in anderen Größen.
Besten Dank!
Dirk

sirtiger, 2023-02-07:
> moin Dirk!
> Du musst Adresse F8 auslesen (am besten 4 Bytes) und ggf. noch F0 (einByte) und dann anhand der [Liste](https://github.com/philippoo66/ViessData21/blob/master/DataPoints.txt) den [...]

Hallo Phil,

wie kann man denn diese Adressen senden?
Hab schon ein bißchen ausprobiert, aber ich bekomme es nicht hin…

F8 auslesen, wie soll das funktionieren? Den SEND Befehl akzeptiert das vcontrold nicht. Wäre schön, wenn Du das kurz erläutern könntest.

Timo

sirtiger, 2023-02-07:
Moin….

Habe es mit getDevType versucht…

`vctrld>getDevType
ERR: <RECV: read error 11
>FRAMER: read failure
Error in recv, terminating
Error executing getDevType
vctrld>debug on
vctrld>getDevType
DEBUG:Tue Feb  7 21:37:24 2023 : Command: getDevType
DEBUG:Tue Feb  7 21:37:24 2023 : >SENT: 04
DEBUG:Tue Feb  7 21:37:24 2023 : >FRAMER: no preset result
DEBUG:Tue Feb  7 21:37:24 2023 : Waiting for 05
DEBUG:Tue Feb  7 21:37:26 2023 : <RECV: 79 (1200.0 ms)
DEBUG:Tue Feb  7 21:37:26 2023 : <RECV: 01 (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: 05 (2990.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : >SENT: 01
DEBUG:Tue Feb  7 21:37:29 2023 : >SENT: F7
DEBUG:Tue Feb  7 21:37:29 2023 : >SENT: 00
DEBUG:Tue Feb  7 21:37:29 2023 : >SENT: F8
DEBUG:Tue Feb  7 21:37:29 2023 : >SENT: 08
DEBUG:Tue Feb  7 21:37:29 2023 : >FRAMER: no preset result
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 20 (20.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 CB (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 00 (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 2B (10.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 00 (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 00 (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 01 (0.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: len=1 14 (10.0 ms)
DEBUG:Tue Feb  7 21:37:29 2023 : <RECV: received 20 CB 00 2B 00 00 01 14
DEBUG:Tue Feb  7 21:37:29 2023 : 20 CB 00 2B 00 00 01 14 -> UNKNOWN
DEBUG:Tue Feb  7 21:37:29 2023 : UNKNOWN
UNKNOWN
vctrld> `

Das kam dabei raus….

philippoo66, 2023-02-07:
Moin ihr!

Das ist doch genau, was wir wissen wollen... (zumindest F8)

[image]

fu.. auf unknown - das sind die Daten. In der Liste steht dazu

[image]

2B liegt zwischen 28h und 45h. Das 3. Byte an F8 kannste vergessen, in der Liste steht bei allen 20CB da 01. Wahrscheinlich für "eine für alles" (oder einfach nur aus Spass/zum Verwirren - k.A.). Wichtig ist das 4. (und natürlich die ersten beiden) 

Es gibt übrigens eine ergänzte DP Listen [Sammlung](https://github.com/philippoo66/ViessData21/blob/master/DP_Listen_2.zip) . 

Lies vlt auch noch was ich im [readme](https://github.com/philippoo66/ViessData21#readme) zu den DP Listen geschrieben habe...

Grüsse!
Phil

philippoo66, 2023-02-07:
@sirtiger  Timo ich kenn mich mit vcontrold überhaupt nicht aus, ich hab die Kommunikation in Windoofs geschrieben (gibt auch die dll in dem ViessData 2.3 bei mir im Repo, leider bisher undokumentiert und ohne Beispiel), aber ich hab das hier bei vcontrold gefunden:

		<command name="getaddr">
		     <send>SYNC;GETADDR $addr $len;RECV $len $unit</send>
		</command>

wahrscheinlich kannst du da mehr mit anfangen als ich. Die addr wäre jedenlass F8 (hex) und die len 4. unit gibts nich.

philippoo66, 2023-02-07:
ich schreib beizeiten mal ein kleines Tool, was das direkt ausliest und alles im Klartext anzeigt. In ViessData 2.x ist das schon drin

[image]

läuft aufm Raspi dann aber nur mit mono. Aber dafür könnte man (freu) ja auch mal grad das Optolink USB an nen Laptop stöpseln... ;-)

sirtiger, 2023-02-08:
Moin Phil,

erstmal schönen Dank für Deine Antworten.
Das war überaus hilfreich. Echt top..

Danke

sirtiger, 2023-02-09:
Moin Phil,

ich nochmal. Ganz schlau bin ich aus der List nicht geworden. Finde wohl vereinzelnd Adressen, aber wie muß ich die Datei lesen?

`   - Warmwasser-Solltemperatur (51) [Bedien_WW_Solltemperatur~0x6300 (Byte)]`

Aus der DP_VScot_HO1_40

Das ist klar. Steht bei mir auch in der vito.XML für get und set drin.
Der get-Befehl funktioniert auch. Beim set-Befehl bekomme ich einen Fehler.

Timo

 

sirtiger, 2023-02-09:
Und schon wieder…

Habe versucht das Log auszulesen - leider ohne Erfolg.

`iobroker@iobroker:~ $ vclient -h localhost:3002 -c getTempWWsoll
[10802] Thu Feb  9 09:03:07 2023 : timeout wait:vctrld>
[10802] Thu Feb  9 09:03:07 2023 : Error communicating with the server
iobroker@iobroker:~ $ service vcontrold status
● vcontrold.service - LSB: Initscript to start vcontrold deamon
     Loaded: loaded (/etc/init.d/vcontrold; generated)
     Active: active (exited) since Thu 2023-02-09 09:01:07 CET; 2min 17s ago
       Docs: man:systemd-sysv-generator(8)
    Process: 10788 ExecStart=/etc/init.d/vcontrold start (code=exited, status=0/SUCCESS)
        CPU: 299ms

Feb 09 09:01:06 iobroker systemd[1]: Starting LSB: Initscript to start vcontrold deamon...
Feb 09 09:01:07 iobroker systemd[1]: Started LSB: Initscript to start vcontrold deamon.
iobroker@iobroker:~ $ vcontrold status
[10808] Thu Feb  9 09:03:39 2023 : started vcontrold version 0.98.11
iobroker@iobroker:~ $ vclient -h localhost:3002 -c getTempWWsoll
[10811] Thu Feb  9 09:04:22 2023 : timeout wait:vctrld>
[10811] Thu Feb  9 09:04:22 2023 : Error communicating with the server
iobroker@iobroker:~ $`

Komischerweise Plotter Iobroker die Daten mit.
Ich verstehe das nicht mehr…

Timo

philippoo66, 2023-02-09:
Moin Timo!
Die Adresse steht hinter der Schlange '~', und die Länge musst du von dem Datenformat ableiden, was da in Klammern steht. Oft ist das einfach, Byte ist natürlich len = 1. Bei 'get' kann sein, dass eine 'falsche' Länge funtioniert, bei 'set' geht das dann schief.

Wie gesagt, ich hab noch nie in die Raspi Sachen reingeschaut (vcontrold, ioBroker), deswegen krieg ich jetzt schlecht den 'Faden' zu was du schreibst...

Also noch mal zum Verständnis:
mit vcontrold funktioniert das Auslesen, nicht aber das Schreiben? Stimmt die Länge? Kannst du die Telegrammdaten ausgeben? 

Bei den xml's, wo die Daten beschrieben sind, harmoniert die Beschreibung mit der aus der DP_VScot_HO1_40.txt?

bei ioBroker bekommst du eine Fehlermeldung, aber das Auslesen funktioniert?

So wie ich das sehe, funktionieren die vcontrold Sachen mit KW Protokoll, also immer erstmal auf 0x05 von der Steuerung warten ("SYNC") - hast du irgendwelche timout Einstellungen zu knapp gesetzt?

Gib mir mal ein paar mehr Details, was du konkret machst (welche "Befehle", welche Daten gehen hin und her?) damit ich dir folgen kann. 

Vielleicht sagt auch mal jemad, wer vcontrold und ioBroker benutzt, was dazu?! zb @qwert089 loggt per Raspi, vielleicht blickt der das sofort, ansonsten sag mir bitte genau, was du tust (vlt mit Links, wo ich das finde), damit ich gezielt in den Sachen schauen kann...

Grüsse!

sirtiger, 2023-02-09:
Moin Phil,

> Moin Timo! Die Adresse steht hinter der Schlange '~', und die Daten-Länge musst du von dem Datenformat ableiden, was da in Klammern steht. Oft ist das einfach, Byte ist natürlich len = 1. Bei 'get' kann sein, dass eine 'falsche' Länge funtioniert, bei 'set' geht das dann schief.
> 
Ok... Das lasse ich mal meinen Kopf verarbeiten.

> Wie gesagt, ich hab noch nie in die Raspi Sachen reingeschaut (vcontrold, ioBroker), deswegen krieg ich jetzt schlecht den 'Faden' zu was du schreibst...
> 
Ok, ist ja nicht so schlimm, geht ja eigentlich nur um das vcontrold per Konsole...

> Also noch mal zum Verständnis: mit vcontrold funktioniert das Auslesen, nicht aber das Schreiben? Stimmt die Länge? Kannst du die Telegrammdaten ausgeben?
> 
Das Auslesen funktioniert mit dem iobroker. Meine Datenpunkte werden befüllt. Das Setzen über den DP in iobroker ist nicht möglich. Daraufhin habe ich es über die Konsole versucht und da bekomme ich auch nur die get-Befehle zum Laufen (wenn es mal funktioniert, oft Probleme mit timeouts). 

`iobroker@iobroker:~ $ vcontrold
[12113] Thu Feb  9 13:01:05 2023 : started vcontrold version 0.98.11
iobroker@iobroker:~ $ service vcontrold status
● vcontrold.service - LSB: Initscript to start vcontrold deamon
     Loaded: loaded (/etc/init.d/vcontrold; generated)
     Active: active (exited) since Thu 2023-02-09 09:01:07 CET; 4h 0min ago
       Docs: man:systemd-sysv-generator(8)
    Process: 10788 ExecStart=/etc/init.d/vcontrold start (code=exited, status=0/SUCCESS)
        CPU: 299ms

Feb 09 09:01:06 iobroker systemd[1]: Starting LSB: Initscript to start vcontrold deamon...
Feb 09 09:01:07 iobroker systemd[1]: Started LSB: Initscript to start vcontrold deamon.
iobroker@iobroker:~ $ vclient -h localhost:3002 -c getTempWWsoll
[12122] Thu Feb  9 13:01:49 2023 : timeout wait:vctrld>
[12122] Thu Feb  9 13:01:49 2023 : Error communicating with the server
iobroker@iobroker:~ $ service vcontrold restart
==== AUTHENTICATING FOR org.freedesktop.systemd1.manage-units ===
Authentication is required to restart 'vcontrold.service'.
Authenticating as: ,,, (iobroker)
Password:
==== AUTHENTICATION COMPLETE ===
iobroker@iobroker:~ $ vclient -h localhost:3002 -c getTempWWsoll
[12149] Thu Feb  9 13:02:54 2023 : timeout wait:vctrld>
[12149] Thu Feb  9 13:02:54 2023 : Error communicating with the server
iobroker@iobroker:~ $`

geht mal wieder nicht. iobroker plottet mit... (Die Ausgabe stellt er nicht als 'Code' dar!!! Sorry)
Telegrammdaten???? Da kann ich nichts mit anfangen?

> Bei den xml's, wo die Daten beschrieben sind, harmoniert die Beschreibung mit der aus der DP_VScot_HO1_40.txt?
> 
vito.xml und DP_VScot_HO1_40.txt harmonieren, so weit ich das beurteilen kann.

> bei ioBroker bekommst du eine Fehlermeldung, aber das Auslesen funktioniert?
> 
Im iobroker bekomme ich keine Fehlermeldung. Beim set-DP keine Änderung des get-DPs

> So wie ich das sehe, funktionieren die vcontrold Sachen mit KW Protokoll, also immer erstmal auf 0x05 von der Steuerung warten ("SYNC") - hast du irgendwelche timeout Einstellungen zu knapp gesetzt?
> 
Das Protokoll steht auf 'KW'. Wie warte ich auf das 0x05? Da kenne ich mich noch nicht wirklich mit aus... Entweder gebe ich in der Konsole 'vclient -h localhost:3002 -c getTempWWsoll ein oder mache es direkt über die 'commands'. 
Wo kann man die timeout-Parameter einstellen? In der vcontrold?

> Gib mir mal ein paar mehr Details, was du konkret machst (welche "Befehle", welche Daten gehen hin und her? Welche xml's kommen zum Zug?) damit ich dir folgen kann.
z.B. diese-> vito.xml
`
<command name="getTempWWist" protocmd="getaddr">
      <addr>0804</addr>
      <len>2</len>
      <unit>UT</unit>
      <description>Ermittle die Warmwassertemperatur in Grad C</description>
    </command>
    <command name="getTempWWsoll" protocmd="getaddr">
      <addr>6300</addr>
      <len>1</len>
      <unit>UTI</unit>
      <description>Ermittle die Warmwassersolltemperatur in Grad C</description>
    </command>
    <command name="setTempWWsoll" protocmd="setaddr">
      <addr>6300</addr>
      <len>2</len>
      <unit>UTI</unit>
      <description>Setze die Warmwassersolltemperatur in Grad C</description>
    </command>
`

> 
> Vielleicht sagt auch mal jemad, wer vcontrold/ioBroker benutzt, was dazu?! zb @qwert089 loggt per Raspi, vielleicht blickt der das sofort, ansonsten sag mir bitte genau, was du tust (vlt mit Links, wo ich das finde), damit ich gezielt in den Sachen schauen kann...
> 
> Grüsse!
> 
> ps. vielleicht beschreibst du auch noch mal kurz, was das Ziel deines Vorhabens ist. Agierst du überhaupt auf einem Raspi o.ä. oder bist du auf Windoofs?
>
Auf einem Raspberry mit iobroker... Es geht um das kontinuierliche Mitplotten der Daten und um das Ansteuern eines Heizstabes, der über den PV-Überschuß meinen Speicher versorgt.

Komisch ist es, daß der iobroker damit arbeitet, die Konsole aber nicht funktioniert. Ich glaube, ich werde das vcontrold nochmal neu aufsetzen oder kann ich es mit der neuen Version aktualisieren?

Wenn ja, wie funktioniert das? Oder vcontrold vom System löschen, wie würde das funktionieren?

Schon einmal schönen Dank für Deine Hilfe....

Timo

sirtiger, 2023-02-09:
Moin...

Das timeout habe ich auf 7000 gesetzt...  Werde das mal in der vito.xml anpassen.
Leider funktioniert gerade gar nichts mehr....

Ich versuche das mal wieder zu fixen...

Bis dann...

qwert089, 2023-02-09:
Hallo,

ich habe meine Raspberry nach diese Anleitung 
http://blog.bubux.de/viessmann-heizung-auslesen/
aufgebaut.
Etwas andere DP und ein anderes Datenbank Schema.
Das GUI habe ich auch von dieser Seite und angepasst. Da kann man ein zip herunterladen.

VG

philippoo66, 2023-02-09:
noch mal zu oben:
> 0804 2 UT Ermittle die Warmwassertemperatur in Grad C

ich interpretier das so: 0804 ist die Addr in hex (das ist ein ADC Wandlerwert, also schlüssig). 2 ist die Länge in Byte - das passt bei den Wandlerwerten. Das UT ist irgendwie ne unit, die vcd benutzt zur Darstellung, das hat mit dem Protokoll nix zu tun, nur dass beim Wandlerwert das Komma verschoben werden muss.

kopier hier bitte mal die vcd Kommandos hin, die du in der Konsole benutzt, und nen Link, welches vcd du benutzt (ich hab da diverse Forks gefunden, die alle ein wenig anders sind)
 
danke @qwert089 !

sirtiger, 2023-02-09:
> Hallo,
> 
> ich habe meine Raspberry nach diese Anleitung http://blog.bubux.de/viessmann-heizung-auslesen/ aufgebaut. Etwas andere DP und ein anderes Datenbank Schema. Das GUI habe ich auch von dieser Seite und angepasst. Da kann man ein zip herunterladen.
> 
> VG

Danke für den Link. Ich werde es mir morgen mal in Ruhe angucken.

sirtiger, 2023-02-09:
> noch mal zu oben:
> 
> > 0804 2 UT Ermittle die Warmwassertemperatur in Grad C
> 
> ich interpretier das so: 0804 ist die Addr in hex (das ist ein ADC Wandlerwert, also schlüssig). 2 ist die Länge in Byte - das passt bei den Wandlerwerten. Das UT ist irgendwie ne unit, die vc'd benutzt zur Darstellung, das hat mit dem Protokoll nix zu tun, nur dass beim Wandlerwert das Komma verschoben werden muss.
> 
> kopier hier bitte mal die vc'd Kommandos hin, die du in der Konsole benutzt, und nen Link, welches vc'd du benutzt (ich hab da diverse Forks gefunden, die alle ein wenig anders sind)
> 
> und danke @qwert089 !

Ich habe die vito.XML mal angepasst (1 BYTE). Per Iobroker kann ich denn DP jetzt auf einen Temp-Wert setzen und es funktioniert. Top.

Jetzt ist das Problem mit der Konsole noch vorhanden. Da geht nichts mehr.
Ich muß mir das morgen nochmal in Ruhe angucken, weil ich gleich zum Training muß. 
Erstmal Danke für Deine Hilfe.

Timo

philippoo66, 2023-02-09:
so, ich hab mir mal vcontrold und ioBroker etwas näher angeguckt. ioBroker ist ja ein ganz schönes 'Monster', da werd ich mich nicht weiter einlesen, aber meine Vermutung, dass es vc'd 'benutzt' (als Adapter) passt wohl.

Insofern weiss ich nicht, ob es klug wäre, vc'd neu aufzusetzen, wenn Lesen und Schreiben mit ioBroker funktioniert, da dann ja vc'd auch funktionieren muss.

Die beiden kommunizieren wohl über den Netzwerk Port 3002. So wie ich verstanden habe, geht die Konsole den gleichen Weg - richtig? Dann wäre es nicht weiter verwunderlich, dass Konsolenbefehle nicht gehen, wenn ioBroker läuft - der Port ist dann ja schon 'belegt'. Oder schliesst ioBroker den Port immer wieder zwischendurch?

Ansonsten kann man alles wichtige zu vc'd großteils den beiden xml's entnehmen. Tatsächlich war/ist da das Schreiben von WW soll mit falscher Länge angegeben, aber das hast du ja korrigiert. Auch die Einheiten-Geschichte von oben steht da.

Ich kann also jetzt hoffentlich weitere Fragen etwas 'sicherer' beantworten... ;-)

sirtiger, 2023-02-12:
Hallo Phil,

ich habe es hinbekommen. 
Tatsächlich haben sich der Viessmann Adapter und die CLI gegenseitig auf den Fuß getreten. 
Die Adressen werde ich mir vorher mal anschauen, bevor ich sie verwende, ob die Länge auch passt.

Schönen Dank für Deine Hilfe. Echt top.

Timo

philippoo66, 2023-02-13:
hey Timo,
das freut mich!
könntest du vlt [hier](https://github.com/openv/vcontrold/issues/123#issuecomment-1426991034) noch sagen, wie man im CLI einen Wert _schreibt_? der Mann sucht danach und ich bin auch ratlos...
danke & Grüsse!
Phil