openv/openv #563: setBetriebArtM2 Betriebsart setzen mit Telnet

Tonitom, 2022-11-07:
Bitte um Hilfe,
ich verstehe nicht, wie ich mit Telnet die Betriebsart setzen kann "setBetriebArtM2 H+WW".
Wenn ich den Befehl setBetriebArtM2 absetze bekomme ich "ERR: command unknown" zurück. Der "get" Befehl funktioniert.

Muss ich die vito.xml anpassen? Etwa so? 

    <command name="setBetriebArtM2 H+WW" protocmd="setaddr">
      <addr>3323</addr>
      <len>1</len>
      <unit>BA</unit>
      <description>Setze Betriebsart H2</description>
      <device ID="20CB"/>
    </command>

Hat zumindest nicht funktioniert.

Die DeviceID hab ich mit der WindowsApp "voIdent" auslesen können

Ich hab ursprünglich in einem anderen Issue gepostet, der wurde aber nicht nach vorne gespült, daher ein neuer Versuch.

Tonitom

Goersch, 2022-11-08:
<vito>
  <devices>
	<device ID="20CB" name="VScotH01" protocol="P300"/>
	<device ID="2053" name="GWG_VBEM" protocol="GWG"/>
  </devices>
  <commands> 
        <command name='getDevType' protocmd='getaddr'>
                <addr>00F8</addr>
                <len>2</len>
                <unit>DT</unit>
                <description>Ermittle Device Typ der Anlage</description>
                <device ID="2053" protocmd='getvaddr'>
                    <addr>F8</addr>
                    <len>4</len>
                    <error>05 05 05 05</error>
                </device>
        </command>
        <command name='getTempA' protocmd='getaddr'>
	     <addr>0800</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <error>05 05</error>
	     <description>Ermittle die Aussentemeratur in Grad C</description>
             <device ID="2053">
		<addr>6F</addr>
	        <unit>UT1</unit>
		<len>1</len>
	     </device>			
	</command>
	<command name='getTempWWist' protocmd='getaddr'>
	     <addr>0804</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <error>05 05</error>
	     <description>Ermittle die Warmwassertemperatur in Grad C</description>
             <device ID="2053">
		<addr>42</addr>
	        <unit>UT1U</unit>
		<len>1</len>
	     </device>			
	</command>
	<command name='getTempKist' protocmd='getaddr'>
	     <addr>0802</addr>
	     <len>2</len>
	     <unit>UT</unit>
	     <error>05 05</error>
	     <description>Ermittle die Kesseltemperatur in Grad C</description>
             <device ID="2053">
		<addr>70</addr>
	        <unit>UT1</unit>
		<len>1</len>
	     </device>			
	</command>
        <command name='getTempRaumNorSollM2' protocmd='getaddr'>
             <addr>3306</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Ermittle die Raumsolltemperatur normal M2 in Grad C</description>
             <device ID="2053"/>
        </command>
        <command name='getTempRaumRedSollM2' protocmd='getaddr'>
             <addr>3307</addr>
             <len>1</len>
             <unit>ST</unit>
             <description>Ermittle die Raumsolltemperatur reduziert M2 in Grad C</description>
             <device ID="2053"/>
        </command>      
        <command name='getPumpeStatusM2' protocmd='getaddr'>
             <addr>3906</addr>
             <len>1</len>
[... 624 lines cut ...]
        </command>
        <command name="getkmaddr">
          <send>SYNC;GETKMDDR $addr $hexlen 04;RECV $len $unit</send>
          <retry>3</retry>
          <recvTimeout>4000</recvTimeout>
        </command>
        <command name="getkmtestaddr">
          <retry>3</retry>
          <recvTimeout>4000</recvTimeout>
          <send>SYNC;GETKMADDR;SEND BYTES;SEND 01 04;RECV 1</send>
        </command>
        <command name="init">
          <send>SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100;SEND 16 00 00;PAUSE 100</send>
          <retry>3</retry>
        </command>
        <command name="setaddr">
          <send>SYNC;SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
        </command>
      </commands>
    </protocol>
  </protocols>
  <extern xmlns:xi="http://www.w3.org/2003/XInclude">
    <xi:include href="vito.xml" parse="xml"/>
  </extern>
</V-Control>

Goersch, 2022-11-08:
tja, wenn github mal mein ZIP akzeptieren würde

Goersch, 2022-11-08:
https://www.dropbox.com/s/crubxi223ctszqd/vcontrold.xml?dl=0
https://www.dropbox.com/s/7wdh23xeqru63cn/vito.xml?dl=0

Tonitom, 2022-11-08:
Super, danke.

Werd ich gleich testen.

Am 08.11.2022 um 20:57 schrieb Georg Nowak ***@***.******@***.***>>:

EEProm

Vinshi01, 2025-05-07:
Habe das Problem das er bei mir die Betriebsart liest ich sie aber nicht setzen kann. Ich habe eine 20CB