openv/openv #566: Viess-Data 2.06

wespe48, 2022-11-13:
Hallo,
ist jemandem bekannt, dass Viess-Data 2.06 sporadisch die Vitotronic 200 resettet und die Heizung (z.B. Vitocrossal 300) aussteigen läßt?
Egal mit welchem Notebook, Netbook, PiPo usw. die Schnittstelle zur Vitotronic hergestellt wird. 
Wäre schön, wenn jemand antworten würde!

philippoo66, 2023-02-09:
Moin!
ich benutze ViessData seit über 4 Jahren in Wintern nahezu kontinuiertlich und hatte keinen (0) Ausstieg oder Reset der Steuerung. Das was @speters bzgl Datenpunkte geschrieben hat, könnte schon die mögliche Ursache sein. Guck mal in mein Repo [ViessData21](https://github.com/philippoo66/ViessData21/tree/master) , da findest du DP-Listen für über 170 Geräte.
Grüsse!
Phil

wespe48, 2023-03-02:
Danke Phil, leider ist in der DP-Liste nicht mein Vitocrossal 300 enthalten!GrüsseWerner Gesendet von Mail für Windows Von: PhilGesendet: Donnerstag, 9. Februar 2023 02:08An: openv/openvCc: wespe48; AuthorBetreff: Re: [openv/openv] Viess-Data 2.06 (Issue #566) Moin!ich benutze ViessData seit über 4 Jahren in Wintern nahezu kontinuiertlich und hatte keinen (0) Ausstieg oder Reset der Steuerung. Das was @speters bzgl Datenpunkte geschrieben hat, könnte schon die mögliche Ursache sein. Guck mal in mein Repo ViessData21 , da findest du DP-Listen für über 170 Geräte.Grüsse!Phil

philippoo66, 2023-03-03:
> leider ist in der DP-Liste nicht mein Vitocrossal 300 enthalten!

Hast du mal F8 (min 4 Bytes) / F0 (1 Byte) ausgelesen? Was steht drin?

wespe48, 2024-05-31:
Hallo philippoo66,
Ich habe letztes Jahr mit der Aufzeichnung aufgehört, da ich von Programmierung  keine Ahnung habe.
Vor 2 Monaten habe ich wieder einmal in Github gestöbert. Hat mich dann wieder inspiriert, aufgrund der verfügbaren Viessdata-Files, die Aufzeichnungen wieder aufzunehmen. Ihr habt ja tolle Arbeit geleistet, mit der Weiterentwicklung der Viessdata-Anwendungssoftware, denn bis heute ist (nach Viessdata 2.06) kein einziges mal die Vitotronic, mit Vitocrossal 300 ,ausgestiegen. Der Erfolg ist Euch zugeschrieben!! Leider passiert aber folgendes: Die Aufzeichnung (Graph) bricht sporadisch ab und ich muss sie dann manuell wieder triggern. Das passiert z.B. nachts um 4Uhr oder auch tagsüber z.B. mittags um 13Uhr. Daraufhin habe ich alle zur Verfügung stehenden Viessdata-Programme der Reihe nach ausprobiert: Ist bei allen das gleiche Phänomen!  Ist aber schon mal 2 Wochen durchgelaufen! Was kann das sein?
Gruß Wespe48

wespe48, 2024-06-01:
Hallo philippoo66
Kommando zurück! Tut mir leid, aber es muss meine Hardwareschnittstelle sein. Soeben habe ich den eingefrorenen Graphen wieder starten wollen, da kam die Meldung "com3-Fehler". Also, alles in bester Ordnung. Tolle Leistung. Aber könntest du mir bitte näher erläutern, wie ich zu der Adresse F8 komme? Ist das eine Adresse, die man im Debug-Modus auslesen kann? Die Adresse F8 in der Vitotronic hat mit Temperatureinstellung zu tun. Wäre nur zur Info für mich interressant. Vielen Dank.
Viele Grüße wespe48

philippoo66, 2024-06-01:
Moin Werner @wespe48 !

in F8 steht die Geräte-Identifikation, mit deren Hilfe du die passende Datenpunkte-Liste findest. Du kannst die Adresse und die Länge (8 Bytes) in die Liste eintragen und dann `[Read DP]`  drücken (ich denke das gab es in der 2.06 auch schon).

[image]

Ich hab bei mir einen extra Eintrag in die XML gemacht, du kannst aber auch jede beliebige Zeile in der Tabelle 'missbrauchen' ;-)

Wenn das mit 2.06 nicht gehen sollte, kannst du dieses kleine Tool benutzen: https://github.com/openv/openv/wiki/voIdent

> habe ich alle zur Verfügung stehenden Viessdata-Programme ...

Kennst du eigentlich schon meinen neuen [Optolink Switch](https://github.com/philippoo66/optolink-splitter)?! Damit kannst du einfach einen kleinen Raspi an die Heizung hängen und der schreibt bei Bedarf kontinuierlich die Daten in eine csv für Viessdata. Du kannst dann Viessdata im Wohnzimmersessel bei Bedarf aufmachen und die csv laden. Viessdata funktioniert dabei wie gewohnt, nur dass es eine TCP-IP Verbindung zum Raspi aufbaut, und eigentlich der Raspi das Optolink macht. Deswegen geht es vom Wohnzimmersessel (oder eben überall wo W/LAN) ist. 

Der Raspi braucht dabei 24/7 weniger Strom als das Laptop, das soonst vor der Heizung steht, und du kannst auch Hausautomatisierung über MQTT anbinden und das Vitoconnect parallel benutzen (wer es denn will).

Bei dem Optolink-Splitter könnte man auch einbauen, dass wenn eine Zeit lang nix mehr gelesen werden kann, er den Raspi neu startet, und danach sollte dann die Kommunikation eigentlich wieder gehen. 

[image]

Grüsse!
Phil

wespe48, 2024-06-01:
Hallo Phil,

vielen Dank für die ausführliche Erklärung. Werde ich mir die nächste Tage genauer ansehen. Jetzt steht erst mal die Hochzeit meines Enkels im Vordergrund. 
Also nochmals dankeschön...

Grüße!
Werner

wespe48, 2024-06-16:
Hallo Phil,
das ist mein Eintrag unter Adresse F8:
20 CB 03 51.
Übrigens ist die Aufzeichnung (Graph) diese Woche 2x eingefroren. Habe derzeit die Version 2.4.0.2 in Betrieb. Der oben vermeintliche com3-Fehler war wahrscheinlich nicht die Ursache.
Ich habe auch das Erklärvideo von deiner neuen Optolink-Schnittstelle angesehen.
Ich habe noch die ältere Vitoconnect100 und wie komme ich an den Rest, Raspi usw.? Muss dazu sagen, dass ich aber keinerlei Programmiererfahrung habe!
LG Werner

philippoo66, 2024-06-16:
Moin Werner!

zu 20 CB 03 51 passt das hier:

[image]

20CB vorne, 01 als eins für alle (3. Byte) und dann ist deine 51 zwischen 48 und 59 (4. Byte). Die Datei hab ich mal angehängt.

Eigentlich sollte Viessdata durchlaufen, wenn das Betriebssystem und die Kommunikation es lässt. Hat es bei mir jedenfalls immer so lange getan, bis irgend ein Windows Update den Rechner neu gestartet hatte... Gab es eine Fehlermeldung in der Box unten auf dem Statistik-Tab?

Wenn sowohl Vers. 2.06 als auch Vers. 2.4 irgendwann 'einfrieren' ist irgendwas mit deinem System nicht zuverlässig. Wenn du den originalen Optolink Kopf verwendest, tippe ich da eher auf das Windows System. Die Heizung selber läuft eigentich stabil, da hab ich noch nie Probleme mit der Kommunikation gesehen. Mein Kommunikation hat Timeout- und Wiederholungs-Mechanismen eingebaut, selbst wenn mehrfach was schief geht, geht es irgendwann mit dem nächsten Datennpunkt weiter.

>  wie komme ich an den Rest, Raspi usw.?

bei Raspi reicht ein 2er lang aus. Wenn der WLAN können soll besser einen 3er nehmen mit WLAN on Board. Die gibt es z.B. auf ebay kleinanzeigen, ich habe letztens einen 2er für 10€ erworben, das war aber Glück, gewöhnlich werden 20 für nen 2er und 30 für nen 3er 'aufgerufen', wobei man den Verhandlungsspielraum nutzen sollte, weil für gut 40 gibt's den 3er schon neu mit Garantie.

Diesen CP2102 TTL-USB Adapter gibt es 'überall', z.B. [hier](https://www.roboter-bausatz.de/p/usb-zu-ttl-uart-modul-cp2102?srsltid=AfmBOooeGqAhgjDHoGbNU7v7yvPxRIj69vnOudKp9j-vcqvZ_HjOvh024JU) oder [hier](https://www.amazon.de/dp/B0B18JKYBF). Den brauchst du aber nur, wenn du das Vitoconnect parallel betreiben willst.

Programmierkenntnisse sind nicht nötig, aber man sollte einen Raspi zumindest aufsetzen können und sich trauen, da auch ein paar Kleinigkeiten damit zu machen, was aber auch kein Hexenwerk ist und mannigfaltig beschrieben und ich kann dich da auch unterstützen.

Grüsse!
Phil

[DP_VScotHO1_72.txt](https://github.com/user-attachments/files/15857358/DP_VScotHO1_72.txt)

wespe48, 2024-08-02:
Moin Phil,
bei der Auswertung meiner Speichertemperaturkurve habe ich festgestellt, dass an der Cursorposition falsche Werte angezeigt werden.
[image]
[image]
Du siehst an den Graphen, dass der Spitzenwert niedriger angezeigt wird, als in der fallenden Kurve, wenige Millimeter daneben, wo der tatsächliche Wert dann angezeigt wird. Vielleicht kannst du dir das mal ansehen. Vielen Dank!

Grüße
Werner

wespe48, 2024-08-02:
Sorry, 2x das gleiche Bild....
[image]

Grüße
Werner