openv/openv #575: Abgastemperatur Vitodens 200 20CB

schoerg, 2022-12-20:
Weiß jemand wofür die Abgastemperatur gut ist?

Auf Adresse `0x0808` liefert die bei mir Werte zurück. Sind immer niedriger als Warmwasser und Vorlauf auf `0x0804` und `0x2900`. Allerdings liefert die Adresse auch Daten wenn die Anlage nichts macht (Brennerstatus auf 0, Brennerpower auch 0). Auch auf dem LCD im Menü wird es als Abgastemperatur bezeichnet.

Schlau werde ich dabei nicht - ist es eventuell ein Rücklaufsensor?

philippoo66, 2023-01-27:
Moin! 
Die Abgrastemperatur ist dafür gut, dass du siehst, wie warm das 'Gas' (Luft ist auch ein Gas) ist, das durch den Schornstein geht. Weil du ein Wärmetauscher im Kessel hast, erwärmt (wenn der Brenner aus ist) warmes Wasser da drin die Luft im Kessel und das steigt duch das Abgasrohr auf. 
Deswegen solltest du auch bsw. Schwerkraftzirkulation im Speicherladekreis unterbinden, weil sonst der Speicher abgekühlt wird und die Wärme duch den Schornstein geht...
beste Grüsse!
Phil

schoerg, 2023-01-27:
Danke. Ja - was mich aber interessiert ob das wirklich die Gastemperatur ist, oder die Rücklauftemperatur. Der Wert wird manchmal auch für Rücklauf verwendet.

philippoo66, 2023-01-28:
_>> Der Wert wird manchmal auch für Rücklauf verwendet._
das wird dadran liegen, dass 808h der "Wandlerwert" von ADC 'Temperatur Sensor 15' ist. Bei manchen Anlagentypen ist es die Abgastemperatur, bei anderen Anlagen hängt da wohlmöglich ein anderer Sensor drauf...