openv/openv #577: Vito.xml für Vitotronic WO1B mit Vitocal 350-A

vbbasti21, 2022-12-27:
Hat jemand dazu evtl. schon eine vito.xml datei. Habe in einem andere  Thread schon was gefunden was halbwegs geht, allerdings wäre ich sehr an den Parameter interessiert die man setzen kann...

philippoo66, 2023-01-27:
Moin Basti!
moin Basti!
lies mal Adresse F8 aus (am besten 4 Bytes) und ggf. noch F0 (ein Byte) und guck. ob du in der [Liste](https://github.com/philippoo66/ViessData21/blob/master/DataPoints.txt) den Steuerungstyp findest.  Wenn, dann aus dieser [Sammlung](https://github.com/philippoo66/ViessData21/blob/master/DP%20Listen.zip) die passende DP Liste raussuchen.
Wenn du noch andre Fragen hast (Werte, Umwandlung, Enums, Optolink Comm, ...) sag Bescheid!
beste Grüsse!
Phil

vbbasti21, 2023-11-26:
Danke dir ganz herzlich :-)