openv/openv #583: WärmePumpe 222-G | CD70 | Gerätecode 2000 44 | KW2

andyyyy2306, 2023-02-09:
Hallo !
Ich lese mit hilfe dieses Forums seit gut 10 Jahren meine Wärmepumpe (WärmePumpe 222-G | CD70 | Gerätecode 2000 44 | KW2) erfolgreich aus.
Anfangs über einen PC, mittlerweile über einen RaspberryPi.
Die Daten werden in meinem Smarthome System "Domoticz" gespeichert.
Ich habe mir anfang des Jahres eine PV Anlage gebaut und würde nun gerne die Funktion „Warmwasser max.“ oder „einmalige Warmwasserbereitung“ nutzen. Hat Irgendjemand noch mehr Adressen zu meiner Wärmepumpe ?
Ich habe euch meine aktuellen vito.xml und vcontrold.xml angehängt.
vcontold.xml
`<?xml version="1.0"?>
<V-Control xmlns:vcontrol="http://www.openv.de/vcontrol">
  <unix>
     <config>
	<serial>
		<tty>/dev/ttyUSB0</tty>
	</serial>
	<net>
		<port>3002</port>
		<allow ip='127.0.0.1'/>
		<allow ip='192.168.1.0/24'/>
	</net>
	<logging>
		<file>/tmp/vcontrold.log</file>
		<syslog>n</syslog>
		<debug>n</debug>
	</logging>
	<device ID="2000"/>
     </config>
  </unix>
  <units>
		<unit name='Temperatur'>
			<abbrev>T1U</abbrev>
			<calc get='V' set='V'/>
			<type>uchar</type>
			<entity>C</entity>
		</unit>
		<unit name='Temperatur'>
			<abbrev>TD</abbrev>
			<calc get='V/10' set='V*10'/>
			<type>short</type>
			<entity>C</entity>
		</unit>
		<unit name='Temperatur'>
			<abbrev>T1S</abbrev>
			<calc get='V/2' set='V*2'/>
			<type>uchar</type>
			<entity>C</entity>
		</unit>
		<unit name='Umschaltventil Stellung'>
			<abbrev>USV</abbrev>
			<type>enum</type>
			<enum bytes='00' text='UNDEFINIERT'/>
			<enum bytes='01' text='Heizung'/>
			<enum bytes='02' text='Mittelstellung'/>
			<enum bytes='03' text='Warmwasser'/>
			<enum text='?'/>
		</unit>
		<unit name='ReturnStatus'>
			<abbrev>RT</abbrev>
			<type>enum</type>
			<enum bytes='00' text='0'/>
			<enum bytes='01' text='1'/>
			<enum text='keine Antwort'/>
		</unit>
		<unit name='Status'>
			<abbrev>ST</abbrev>
			<calc get='V' set='V'/>
			<type>char</type>
			<entity></entity>
		</unit>
		     <unit name='DevType'>
            <abbrev>DT</abbrev>
            <type>short</type>
        </unit>
		<unit name='Systemzeit'>
			<abbrev>TI</abbrev>
			<type>systime</type>
       		</unit>
        <unit name='SetReturnStatus'>
			<abbrev>SR</abbrev>
          	<type>enum</type>
            <enum bytes='00' text='OK' />
            <enum bytes='05' text='SYNC (NOT OK)' />
            <enum text='NOT OK' />
 		</unit>
		<unit name='Aktuelle Betriebsart'>
			<abbrev>ABA</abbrev>
			<type>enum</type>
			<enum bytes='00' text='Abschaltbetr. (Dauernd)'/>
			<enum bytes='01' text='Red. Betrieb (Schaltuhr)'/>
			<enum bytes='02' text='Normalbetrieb (Schaltuhr)'/>
			<enum bytes='03' text='Normalbetrieb (Dauernd)'/>
			<enum text='?'/>
		</unit>
		<unit name='Betriebsart'>
			<abbrev>BA</abbrev>
			<type>enum</type>
			<enum bytes='00' text='Abschaltbetrieb'/>
			<enum bytes='01' text='Nur Warmwasser'/>
			<enum bytes='02' text='Heizen und Warmwasser'/>
			<enum bytes='03' text='Dauernd red. Betrieb'/>
			<enum bytes='04' text='Dauernd Normalbetrieb'/>
			<enum text='?'/>
		</unit>	
		<unit name='Ferienbetrieb'>
			<abbrev>BFB</abbrev>
			<type>uchar</type>
			<icalc get="(B7 &amp; (0x01&lt;&lt;BP))&gt;&gt; BP"/>
			<enum bytes='00' text='inaktiv'/>
			<enum bytes='01' text='aktiv'/>
			<enum text='?'/>
		</unit>
	</units>	
  <protocols>
		<protocol name='KW2'>
			<macros>
				<macro name='SYNC'>
					<command>SEND 04;WAIT 05</command>
				</macro>
				<macro name='GETADDR'>
					<command>SEND 01 F7</command>
				</macro>
				<macro name='SETADDR'>
					<command>SEND 01 F4</command>
				</macro>
			</macros>
			<commands>
				<command name="getaddr">
					<retry>3</retry>
					<recvTimeout>4000</recvTimeout>
					<send>SYNC;GETADDR $addr $hexlen;RECV $len $unit</send>
				</command>
				<command name="setaddr">
					<send>SYNC;SETADDR $addr $hexlen;SEND BYTES $unit;RECV 1 SR</send>
				</command>
			</commands>
		</protocol> 
  </protocols>
  <extern xmlns:xi="http://www.w3.org/2003/XInclude">
     <xi:include href="vito.xml" parse="xml"/>
  </extern>
</V-Control>
`
vito.xml
`<vito>
  <devices>
	<device ID="2000" name="Vitocal" protocol="KW2"/>
  </devices>
  <commands> 
		<command name='getDummy' protocmd='getaddr'>
			<addr>0800</addr>
			<len>2</len>
			<unit>TD</unit>
			<description>Dummylesen</description>
		</command>
			<command name="getDevType" protocmd="getaddr">
				<addr>00F8</addr>
				<len>2</len>
				<description>Information - Allgemein: Anlagentyp</description>
			</command>
			<command name="getDevType2" protocmd="getaddr">
				<addr>00F9</addr>
				<len>2</len>
				<description>Information - Allgemein: Anlagentyp Subtype</description>
			</command>
		<command name='getTempA' protocmd='getaddr'>
			<addr>0800</addr>
			<len>2</len>
			<unit>TD</unit>
			<description>Aussentemperatur</description>
		</command>
		<command name='getTempPVL' protocmd='getaddr'>
	    		 <addr>0801</addr>
	   		  <len>2</len>
	   		  <unit>TD</unit>
	    		 <description>Primaerkreis Vortemperatur in Grad C</description>
		</command>
		<command name='getTempSVL' protocmd='getaddr'>
	     		<addr>0808</addr>
	     		<len>2</len>
	    		 <unit>TD</unit>
	     		<description>Sekundaerkreis Vorlauftemperatur in Grad C</description>
		</command>
		<command name='getTempSRL' protocmd='getaddr'>
	    		 <addr>0807</addr>
	    		<len>2</len>
	     		<unit>TD</unit>
 			<description>Sekundaerkreis Ruecklauftemperatur in Grad C</description>
		</command>
		<command name='getTempWWU' protocmd='getaddr'>
	     		<addr>0810</addr>
	     		<len>2</len>
	     		<unit>TD</unit>
	     		<description>Warmwassertemperatur unten in Grad C</description>
		</command>
		<command name='getTempWWO' protocmd='getaddr'>
	    		 <addr>080c</addr>
	     		<len>2</len>
	     		<unit>TD</unit>
	     		<description>Warmwassertemperatur oben in Grad C</description>
		</command>
		<command name='getTempHKS' protocmd='getaddr'>
	     		<addr>0898</addr>
	     		<len>2</len>
	     		<unit>TD</unit>
	     		<description>Solltemperatur Heizkreis in Grad C</description>
		</command>
        	<command name='getPumpeStatusWp' protocmd='getaddr'>
             		<addr>088a</addr>
             		<len>1</len>
             		<unit>RT</unit>
             		<description>Ermittle den Status der Waermepumpe</description>
        	</command>
        	<command name='getPumpeStatusPp' protocmd='getaddr'>
             		<addr>08bf</addr>
             		<len>1</len>
             		<unit>RT</unit>
             		<description>Ermittle den Status der Primaerpumpe</description>
       		</command>        
		<command name='getPumpeStatusWW' protocmd='getaddr'>
            		 <addr>08b0</addr>
            		 <len>1</len>
             		<unit>RT</unit>
             		<description>Ermittle den Status der Warmwasserbereitung</description>
       		</command>
        	<command name='getPumpeStatusWWU' protocmd='getaddr'>
            		 <addr>08b3</addr>
             		<len>1</len>
             		<unit>RT</unit>
             		<description>Ermittle den Status der Umwaelzpumpe Warmwasser</description>
        	</command>
						<command name='getWWmax' protocmd='getaddr'>
					<addr>????</addr>
					<len>1</len>
					<unit>RT</unit>
					<description>Ermittle ob Einmalladung WW</description>
			</command>
			<command name='setWWmax' protocmd='setaddr'>
					<addr>????</addr>
					<len>1</len>
					<unit>RT</unit>
					<description>aktiviere Einmalladung WW</description>
			</command>
  </commands> 
</vito>
`

Danke und LG

Issr4k, 2023-02-21:
die Einmalige Warmwasserbereitung läuft bei mir mit "B0D0".
Alternativ kannst du noch die B020 ausprobieren.
Schreibe mal mit vcontrold "0" und "1" rein und beobachte auf dem Display deiner Wärmepumpe unter Warmwasser ob ein Häkchen im Viereck hinter "Einmalige Wasserberitung" gesetzt wird.
**vito.xml**
 <command name="set1xWW" protocmd="setaddrValue">
        <addr>B0D0</addr>
        <len>1</len>
        <unit>WW</unit>
        <description>Bedienung 1xWW-Bereitung (0=aus 2=ein)</description>
    </command>
**vcontrold.xml**
    <unit name="WWBereitung">
      <abbrev>WW</abbrev>
      <type>enum</type>
      <enum bytes="00" text="0" />
      <enum bytes="02" text="1" />
      <enum text="UNKNOWN" />
    </unit>

andyyyy2306, 2023-02-25:
Hallo Issr4k.
Danke, ich kann zwar schreiben und bekomme ein OK von der Heizung zurück, jedoch tut sich bei den beiden Adressen nix..
Alle Adressen die ich bis jetzt benutze liegen im Bereich 08xx