openv/openv #584: Sorry ich bekomme Vcontrold nicht mehr installiert oder lauffähig

Triton-KS, 2023-02-12:
Servus zusammen,

scheibar bekomme ich das Brett vorm Kopf nicht mehr weg nachdem ich die funktionierende SD-Karte überschrieben habe.

Nach dieser Anleitung denke ich, ich habe nun merhfach versuch Vcontrold auf einen Raspberry pi 3 oder pi4 zu installieren. aber irgendwie bekomme ich den Dienst nicht mehr lauffähig.

Es das Raspberry PI OS egal? Also Bullseye oder muss es eine frührere Version sein?
muss Bluetooth ausgeschaltet sein ?

Als Fehlermeldung bekomme ich:

```
sudo /etc/init.d/vcontrol status
● vcontrol.service - LSB: Initscript to start vcontrold deamon
   Loaded: loaded (/etc/init.d/vcontrol; generated)
   Active: failed (Result: exit-code) since Sun 2023-02-12 22:55:16 CET; 9min ago
     Docs: man:systemd-sysv-generator(8)
  Process: 827 ExecStart=/etc/init.d/vcontrol start (code=exited, status=203/EXEC)

Feb 12 22:55:16 vcontroldtest systemd[1]: Starting LSB: Initscript to start vcontrold deamon...
Feb 12 22:55:16 vcontroldtest systemd[827]: vcontrol.service: Failed to execute command: Exec format error
Feb 12 22:55:16 vcontroldtest systemd[827]: vcontrol.service: Failed at step EXEC spawning /etc/init.d/vcontrol:…t error
Feb 12 22:55:16 vcontroldtest systemd[1]: vcontrol.service: Control process exited, code=exited, status=203/EXEC
Feb 12 22:55:16 vcontroldtest systemd[1]: vcontrol.service: Failed with result 'exit-code'.
Feb 12 22:55:16 vcontroldtest systemd[1]: Failed to start LSB: Initscript to start vcontrold deamon.
Hint: Some lines were ellipsized, use -l to show in full.
```

Sorry ich verstehe es einfach gerade nicht oder erkenne den Fehler nicht.

Danke für die Hilfe

Issr4k, 2023-02-21:
Hi Mit deinem Log kann ich wenig anfangen.

Kann dir aber meine Installations Notizen anbieten.
Die sind auch aus dem Internet zusammengeklickt, aber funktionieren auf meine Pi4:
Hoffe es hilft dir!

apt-get install git libxml2-dev cmake telnet
sudo mkdir vcontrold
sudo git clone https://github.com/openv/vcontrold.git vcontrold
cd vcontrold
sudo mkdir build
cd build
sudo cmake -DMANPAGES=OFF ..
sudo make
sudo make install
cd ..
sudo mkdir /etc/vcontrold
sudo cp xml/300/* /etc/vcontrold
 
sudo nano /etc/vcontrold/vcontrold.xml (anpassen oder backupfile verwenden)

Wechseln in den build Ordner (wo vcontrold liegt)
dort:
lsusb (ausführen) das Ergebnis könnte dann so aussehen:

Bus 001 Device 005: ID 1a86:7523 QinHeng Electronics CH340 serial converter	<---- Optolink-Schnittstelle
Bus 001 Device 004: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

im Anschluss -> dmesg | grep "now attached"

Ergebnis z.B. so : 

[ 2156.032950] usb 1-1.2: ch341-uart converter now attached to ttyUSB0		<- ttyUSB0 in vcontrold.xml ggf. anpassen

USB TTL converter verbinden und dmesg beobachten:
[ 5137.627482] pl2303 1-7.1:1.0: pl2303 converter detected
[ 5137.629615] usb 1-7.1: pl2303 converter now attached to ttyUSB0

Damit das Gerät eindeutig identifiziert werden kann, nutzen wir den absoluten Pfad vom USB Port:
Befehl: ls -al /dev/serial/by-path
Antwort: lrwxrwxrwx 1 root root 13 28. Jan 09:34 platform-3f980000.usb-usb-0:1.2:1.0-port0 -> ../../ttyUSB0
Der Name wäre dann: 
/dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0-port0

Das in /etc/vcontrold/vcontrold.xml eintragen:
 <serial>
        <tty>/dev/serial/by-path/platform-3f980000.usb-usb-0:1.2:1.0-port0</tty>
    </serial>

Zum Schluss noch:
sudo nano /etc/vcontrold/vito.xml (anpassen oder backupfile verwenden)

Hoizi007, 2023-11-08:
Hi habe auch das Problem dass ich nix zum laufen bringe

könntest du evtl. deine Anleitung noch etwas erweitern von Anfang bis Ende? bei aktuellen Debian? bin scheinbar unfähig dafür das zu kapieren 

lg.