openv/openv #590: Schreiben von Schaltzeiten

cuthulino, 2023-03-21:
Hallo,
ich habe eine Vitocal 200-S mit einer WO1C Steuerung. Die ID ist 204D.
So sieht meine vito.xml aus:
```
    <command name="getTimer3Mo" protocmd="getaddr">
      <addr>9202</addr>
      <len>3</len>
      <description>Schaltzeit Montag 3 HK1</description>
    </command>
    <command name="setTimer3Mo" protocmd="setaddr">
      <addr>9202</addr>
      <len>3</len>
      <description>Schaltzeit Montag 3 HK1</description>
    </command>
```

Ich habe bisher alles gut hinbekommen mit vcontrold, außer die Schaltzeiten.
Ich kann sie ohne Probleme auslesen. Bspw.:
```
vclient -c getTimer3Mo
getTimer3Mo:
A0 C0 01 
```
Das entspricht 20:00 - 24:00 im reduzierten Betrieb.

Versuche ich das ganze allerdings zu schreiben, bekomme ich einen Fehler und komme nicht mehr weiter:
```
vclient -c "unit off","debug on","setTimer3Mo A1 C0 01","unit on"            
unit off:

debug on:

setTimer3Mo A1 C0 01:
DEBUG:Tue Mar 21 21:19:11 2023 : Command: setTimer3Mo A1 C0 01
DEBUG:Tue Mar 21 21:19:11 2023 : Process 1340 tries to aquire lock
DEBUG:Tue Mar 21 21:19:11 2023 : Process 1340 got lock
DEBUG:Tue Mar 21 21:19:11 2023 : >FRAMER: open device /dev/ttyVito ProtocolID 41
DEBUG:Tue Mar 21 21:19:11 2023 : Configuring serial interface /dev/ttyVito
DEBUG:Tue Mar 21 21:19:11 2023 : >SENT: 04
DEBUG:Tue Mar 21 21:19:13 2023 : <RECV: len=1 05 (2000.0 ms)
DEBUG:Tue Mar 21 21:19:13 2023 : <RECV: received 05
DEBUG:Tue Mar 21 21:19:13 2023 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_LINK_STATUS(05) (was FFFF)
DEBUG:Tue Mar 21 21:19:13 2023 : >FRAMER: closed
DEBUG:Tue Mar 21 21:19:13 2023 : >SENT: 16
DEBUG:Tue Mar 21 21:19:13 2023 : >SENT: 00
DEBUG:Tue Mar 21 21:19:13 2023 : >SENT: 00
DEBUG:Tue Mar 21 21:19:13 2023 : <RECV: len=1 06 (10.0 ms)
DEBUG:Tue Mar 21 21:19:13 2023 : <RECV: received 06
DEBUG:Tue Mar 21 21:19:13 2023 : >FRAMER: framer_reset_actaddr framer_current_addr = FRAMER_NO_ADDR (was FE05)
DEBUG:Tue Mar 21 21:19:13 2023 : >FRAMER: opened
DEBUG:Tue Mar 21 21:19:13 2023 : Error in length of the hex string (3) != send length of the command (1), terminating
DEBUG:Tue Mar 21 21:19:13 2023 : Error executing setTimer3Mo A1 C0 01
ERR: Error in length of the hex string (3) != send length of the command (1), terminating
Error executing setTimer3Mo A1 C0 01
unit on:
DEBUG:Tue Mar 21 21:19:13 2023 : Command: unit on
```

Ich habe bei get und set kein unit angegeben, weil der Datentyp bei mir nicht mit denen der xml übereinstimmt und ich es per Bytes schreiben und lesen wollte. 
Die Zeiten liegen bei mir, wie [hier](https://github.com/openv/openv/wiki/Adressen-W%C3%A4rmepumpen-Vitotronic-WO1C), in dem Adressbereich ab 0x9200 und sind jeweils 3 Byte lang.
B0 -> Von
B1 -> Bis
B2 -> Betriebsart 

Was mache ich hier falsch?
Es werden nicht die 3 Bytes sondern nur 1 gesendet!?

daniele2209, 2023-10-17:
Hallo @cuthulino 
ich habe genau das gleiche Problem.
Hast Du es hinbekommen Schaltzeiten zu setzen?

Viele Grüsse

cuthulino, 2023-10-17:
Das würde ich eindeutig mit Jain beantworten.

Ich habe als Workaround mit dem Typ enum gearbeitet und mir so die nötigen Zeiten definiert.

`
<enum bytes="00 01 00" text="0:00 - 0:10 (Standby)" />
`

daniele2209, 2023-10-18:
Hi, Ja und Nein.
Kannst Du mir deine vctrontrold.xml zukommen lassen?