openv/openv #594: H01B control screen now blank

sgilder1, 2023-04-21:
Hi,

I am using vcontrold successfully, but now have a problem where the control unit on my Vitodens V200W (H01B) where the screen is blank and I can no longer control the boiler from the screen.  The boiler operates, and I can continue to get and set attributes through vcontrold.  Is this something that could have been caused by the use of vcontrold? Is there anyway to completely reset the control unit (I have tried disconnecting power and reconnecting, but with no success).

sgilder1, 2023-11-18:
Hi @AntoninSlovacek. Unfortunately, I didn't. In the end I needed to replace the whole unit, and I haven't had the courage to reconnect to the raspberrypi running vcontrold in case the same thing happens again.

AntoninSlovacek, 2023-11-18:
Thats scary!