openv/openv #597: vitotronic 200 in Openhab einbinden

helmar74, 2023-07-07:
Hallo,

ich bin neu in diesem Thema und würde gerne meine vitotronic 200 in Openhab einbinden.
Ich würde eine Lösung mit ESP8266 bzw. ESP32 am besten finden.
Wenn diese noch MQTT "sprechen" würde dann wäre das schon super.
(Analog zu Tasmota)
Daher meine Frage in die Runde:
Hat jemand ein ähnliches Setup am Laufen und könnte ich Infos über Hardware und Software bekommen?

Danke und schönen Gruss

Helmar

j0ta29, 2023-07-07:
Hallo Helmar,
ich habe selbst kein Setup mit Openhab und MQTT. Meine Therme lese ich über einen ESP8266 an der Optolink-Schnittstelle aus und funke die Daten per WiFi zu Home Assistant (die beiden sprechen ein eigenes Protokoll). Das Ganze funktioniert über die ESPHome-Plattform, die aber auch MQTT unterstützt.
Hier ist mein Setup beschrieben: https://github.com/openv/openv/wiki/ESPHome-Optolink
Vielleicht hilft Dir das ja einen kleinen Schritt weiter zu kommen.
Viele Grüße
Jürgen

helmar74, 2023-07-07:
Servus Jürgen,

danke für die Infos, schaue ich mir bei Gelegenheit an.

Gruss

Helmar

helmar74, 2023-07-08:
> Hallo Helmar, ich habe selbst kein Setup mit Openhab und MQTT. Meine Therme lese ich über einen ESP8266 an der Optolink-Schnittstelle aus und funke die Daten per WiFi zu Home Assistant (die beiden [...]

In der Beschreibung steht:
"
Da der Code noch nicht Teil des offizielle Release von ESPHome ist, kann es zeitweise zu Fehler beim Compilieren der Firmware kommen. 
"

Ist die Komponente mittlerweile Teil des offiziellen Releases?

j0ta29, 2023-07-09:
Hallo Helmar,
leider dauert es teilweise sehr lange, bis die Jungs von NabuCasa Pull Requests von anderen Autoren in das offizielle Release mit aufnehmen. Wie lange das noch dauert, kann ich leider nicht sagen. Compile-Fehler in ESPHome sind aber eher die Ausnahme. Das passiert dann, wenn sich die API von ESPHome Core nicht abwärtskompatibel ändert.
Gruß
Jürgen

helmar74, 2023-07-13:
Servus nochmal,

nur zur Info:
Ich habe das Ganze nun über einen "alten" Raspi 3B und vcontrold mit einen Template Skript gelöst.
Sieht aktuell so in Openhab aus:
[image]

[image]

[image]

[image]

closed 2023-07-13 by helmar74